Koenkk/zigbee2mqtt #9981: Aqara E1 / QBKG39LM Switch binding

kiall, 2021-12-03:
## What happened

Binding the buttons on an "Aqara E1 2 gang switch (without neutral)" / QBKG39LM is not possible, as the `genOnOff` etc output clusters aren't detected.

## What did you expect to happen

The various output clusters to be detected, assuming they exist :)

It's possible the clusters need to be re-scanned after changing the device into decoupled mode?

It's also possible these clusters just don't exist on this device and it doesn't support binding at all!

As an aside, I see in the [documentation page](https://www.zigbee2mqtt.io/devices/QBKG39LM.html) for these devices a note for the required firmware versions, I've the latest for the CC26X2R1 - however this doesn't contain the changes introduced recently in the listed firmware. 

## How to reproduce it (minimal and precise)

1. Pair a QBKG39LM with Z2M
2. Set the `operation_mode` of the left and right buttons to decoupled mode
3. Attempt to bind either button to a light & observe the only available bind target clusters are `Ota` and `genTime`.

[image]

## Debug info

Zigbee2MQTT version: 1.22.1
Adapter hardware: CC26X2R1 
Adapter firmware version: zStack3x0 20210708

closed 2022-01-10 by github-actions[bot]

Anonym-tsk, 2022-01-31:
Same issue, please reopen

SergioRius, 2022-08-18:
Same issue with [WXKG07LM](https://www.zigbee2mqtt.io/devices/WXKG07LM.html#xiaomi-wxkg07lm), please consider reopening the issue.

rishab123s, 2022-10-09:
I have the same issue with QBKG39LM Switch binding, please reopen the issue.

danlex-corp, 2022-11-17:
I have the same issue with **QBKG41LM** Switch binding, please reopen the issue.

lsteelandt, 2023-03-12:
This issue remains on Zigbee2Mqtt version 1.30.2.
Using an Sonos Coordinator (EZSP v8 with 6.10.3.0 build 297)

Used Aqara switches : [QBKG39LM](https://www.zigbee2mqtt.io/devices/QBKG39LM.html#xiaomi-qbkg39lm) and [QBKG38LM](https://www.zigbee2mqtt.io/devices/QBKG38LM.html#xiaomi-qbkg38lm)

GigaVik, 2023-10-23:
This issue remains on Zigbee2Mqtt version 1.33.1.

GigaVik, 2023-11-30:
Please open. This issue remains on Zigbee2Mqtt version 1.33.2.

Koenkk, 2023-11-30:
AFAIK this device doesn't send out standard ZCL commands so this is not possible.

aleks-mariusz, 2024-11-07:
> AFAIK this device doesn't send out standard ZCL commands so this is not possible.

can you elaborate how to determine this ? and what the implications are for binding?  or more info?

i ran into this after setting up some new switches all over the house, seems pretty lame to not be able to bind aqara switches to aqara light bulbs 🤦

Koenkk, 2024-11-09:
> can you elaborate how to determine this ?

This device does not have a `genOnOff` cluster (as you can see in the OP). This means it cannot send on/off commands.

aleks-mariusz, 2024-11-10:
> > can you elaborate how to determine this ?
> 
> This device does not have a `genOnOff` cluster (as you can see in the OP). This means it cannot send on/off commands.

@Koenkk thanks for the explanation.  but i have to wonder tho, can ANYTHING be done about this? i mean I am guessing z2m cannot add support somehow to introduce this functionality then, right?

what about aqara, what would be needed from them? a fw update? 

i don't understand why they would intentionally not include support for this.. do you suppose it could be purposeful, or just an oversight?

i'm just really baffled why my aqara wall switch cannot control my aqara bulb ffs.. is this type of info available somehow to check before you buy a product? i mean i would have thought twice about this model if i knew something basic like this wouldn't be currently possible. and i really don't wanna replace all my switches because of this but i'm going to have to think twice about getting aqara products if this is intentionally crippled this way.

Koenkk, 2024-11-11:
> what about aqara, what would be needed from them? a fw update?

A fw update from them indeed, nothing can be done from the z2m side.

wlfbck, 2025-09-13:
Ah, found this issue while trying to figure why i cannot bind them. A shame. I suggest emailing support@aqara.com so they do something about this.