Koenkk/zigbee2mqtt #9983: IKEA E1524/E1810 remote control does not pair anymore with new version

pfischbach, 2021-12-03:
## What happened
Updated to latest version, now IKEA E1524 does not pair anymore 
Used 3 different remotes new batteries no success. Remotes pair with IKEA Hub
[image]
[https://pastebin.com/8s2bHKdX](url)
## What did you expect to happen
remote pairs correct like before

## How to reproduce it (minimal and precise)
Bring remote into pairing mode, and wait for pairing log.

## Debug info
Zigbee2MQTT version: Zigbee2MQTT version 1.22.1-dev (commit #dae66fd)
Adapter hardware: CC2531
Adapter firmware version:
Coordinator firmware version: '{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20211116,"transportrev":2},"

00joshi, 2021-12-06:
I had some issues with pairing but after force deleting, and rebooting zigbee2mqtt, i managed to pair the device.

### Debug info
Zigbee2MQTT Version
    1.22.1 commit: 48b790cb
Coordinator Typ
    zStack3x0
Coordinator Version
    20201026

pfischbach, 2021-12-06:
Thanks for the feedback. 
Yes with the older coordinator it worked for me too. But had other issues with Eurotronics Thermostats
Just tried again. delete, restart and re-pairing without success Remote does not pair

Megacheezy, 2021-12-16:
had this issue as well after applying update, i managed to get past it by touchlinking the remote to an already paired lightbulb. then the remote popped up in Z2M.

lafin, 2021-12-18:
I recently had the same problem. I removed everything except configuration.yaml, then restarted zigbee2mqtt and tried to connect to the coordinator again and everything went well after that.

closed 2022-01-25 by github-actions[bot]