Koenkk/zigbee2mqtt #9984: How to fix USB device link

zibous, 2021-12-03:
Can'nt set USB device link

## What did you expect to happen
After a certain time, the USB link keeps changing and the ZigbeeMQTT can then no longer work 

## How to reproduce it (minimal and precise)
???
I've tried the following, but it doesn't work: 
```
root@nuc:/opt# lsusb
Bus 002 Device 002: ID 152d:0539 JMicron Technology Corp. / JMicron USA Technology Corp. JMS539/567 SuperSpeed SATA II/III 3.0G/6.0G Bridge
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 007: ID 8087:0a2a Intel Corp. Bluetooth wireless interface
Bus 001 Device 006: ID 05e3:0610 Genesys Logic, Inc. Hub
Bus 001 Device 073: ID 1cf1:0030 Dresden Elektronik ZigBee gateway [ConBee II]
Bus 001 Device 004: ID 03eb:204b Atmel Corp. LUFA USB to Serial Adapter Project
Bus 001 Device 003: ID 25a7:fa10 Areson Technology Corp 2.4G Wireless Receiver
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

root@nuc:/opt# nano /etc/udev/rules.d/99-usb-serial.rules
# -------------------------------------------------------
# 1cf1:0030 Dresden Elektronik ZigBee gateway [ConBee II]
# -------------------------------------------------------
#idVendor           0x1cf1 Dresden Elektronik
#idProduct          0x0030 ZigBee gateway [ConBee II]
#iProduct                2 ConBee II
#iSerial                 3 DE2230427
# ------------------------------------------------------
SUBSYSTEM=="tty", ATTRS{idVendor}=="1cf1", ATTRS{idProduct}=="0030", SYMLINK+="ttyACM2"

# ----------------------------------------------------------
# 03eb:204b Atmel Corp. LUFA USB to Serial Adapter Project
# -----------------------------------------------------------
#idVendor           0x03eb Atmel Corp.
#idProduct          0x204b LUFA USB to Serial Adapter Project
#iProduct                2 CUL868
#iSerial                 0
#-------------------------------------------------------------
SUBSYSTEM=="tty", ATTRS{idVendor}=="03eb", ATTRS{idProduct}=="204b", SYMLINK+="ttyACM0"

root@nuc:/opt#  udevadm control --reload-rules

root@nuc:/opt# ls -l /dev/serial/by-id
total 0
lrwxrwxrwx 1 root root 13 Dec  3 19:16 usb-SHA_CUL868-if00 -> ../../ttyACM0
lrwxrwxrwx 1 root root 13 Dec  3 19:18 usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2230427-if00 -> ../../ttyACM3

```
Why does the first symbol link work but not the second?
**What am I doing wrong or what can I do so that the symbol link for the ConBee II always remains ttyACM2.**

Or it would be easier if I could use the '/dev/serial/by-path/pci-0000:00:14.0-usb-0:4:1.0', but it doesn't work 

## Debug info
Zigbee2MQTT version: 1.22.1-dev commit: c47b9c7c
Adapter hardware: ConBee2/RaspBee2
Adapter firmware version: 0.6.46

```
1.22.1-dev commit: c47b9c7c
Coordinator type
ConBee2/RaspBee2
Coordinator revision
0x26660700
Frontend version
0.6.46
Stats
Total 40
By device type
EndDevice: 35
Router: 5
By power source
Battery: 32
Mains (single phase): 5
undefined: 3
By vendor
LUMI: 17
_TZE200_yw7cahqs: 9
eWeLink: 4
undefined: 3
IKEA of Sweden: 3
OSRAM: 2
3A Smart Home DE: 1
_TYST11_yw7cahqs: 1
By model
TS0601: 9
lumi.weather: 7
lumi.sensor_wleak.aq1: 3
lumi.sensor_ht: 3
TRADFRI remote control: 2
TH01: 2
Plug 01: 2
TRADFRI Signal Repeater: 2
lumi.sensor_magnet.aq2: 2
66666: 1
MS01: 1
lumi.sensor_motion.aq2: 1
lumi.sens: 1
lumi.sensor_cube.aqgl01: 1
lumi.sensor_motion: 1
LXT56-LS27LX1.7: 1
w7cahqs: 1
```
Thanks.

hellcry37, 2021-12-04:
Just my 2 cents, I don't think this has anything to do with a bug on zigbee2mqtt, has no business being here. Also when ppl would understand deCONZ Conbee is trash?

zibous, 2021-12-04:
Maybe, but but now overnight the USB link has changed (ttyACM3 --> ttyACM1) again:

```

ls /dev/serial/by-id/*
lrwxrwxrwx 1 root root 13 Dec  3 20:54 /dev/serial/by-id/usb-SHA_CUL868-if00 -> ../../ttyACM0
lrwxrwxrwx 1 root root 13 Dec  4 08:17 /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2230427-if00 -> ../../ttyACM1

ls -al /dev | grep ttyAC
crw-rw----  1 root dialout 166,     0 Dec  3 20:54 ttyACM0
crw-rw----  1 root dialout 166,     1 Dec  4 08:22 ttyACM1
lrwxrwxrwx  1 root root             7 Dec  4 08:17 ttyACM2 -> ttyACM1
lrwxrwxrwx  1 root root             7 Dec  4 08:17 zigbee-serial -> ttyACM1

homeassistant: true
permit_join: false
mqtt:
  base_topic: conbee2mqtt
  server: mqtt://mbsbrocker.home:1883
  user: *****
  password: *****
  include_device_information: true
serial:
  port: /dev/zigbee-serial
frontend:
  port: 8070
```
Since the "Device Links" are ok, I would like to use them, but  if I use the `devlink link:` `/dev/zigbee-serial`, then that does not work because zigbee-herdsman does not recognize it. 

`port: /dev/ttyACM1` is working.

```
udevadm test -a add $(udevadm info -q path -n /dev/ttyACM1)
This program is for debugging only, it does not run any program
specified by a RUN key. It may show incorrect results, because
some values may be different, or not available at a simulation run.

Load module index
Parsed configuration file /usr/lib/systemd/network/99-default.link
Parsed configuration file /usr/lib/systemd/network/73-usb-net-by-mac.link
Created link configuration context.
Reading rules file: /usr/lib/udev/rules.d/01-md-raid-creating.rules
Reading rules file: /usr/lib/udev/rules.d/39-usbmuxd.rules
Reading rules file: /usr/lib/udev/rules.d/40-vm-hotadd.rules
Reading rules file: /usr/lib/udev/rules.d/50-apport.rules
Reading rules file: /usr/lib/udev/rules.d/50-firmware.rules
Reading rules file: /usr/lib/udev/rules.d/50-udev-default.rules
Reading rules file: /usr/lib/udev/rules.d/55-dm.rules
Reading rules file: /usr/lib/udev/rules.d/55-scsi-sg3_id.rules
Reading rules file: /usr/lib/udev/rules.d/56-dm-mpath.rules
Reading rules file: /usr/lib/udev/rules.d/56-dm-parts.rules
Reading rules file: /usr/lib/udev/rules.d/56-lvm.rules
Reading rules file: /usr/lib/udev/rules.d/58-scsi-sg3_symlink.rules
Reading rules file: /usr/lib/udev/rules.d/60-autosuspend.rules
Reading rules file: /usr/lib/udev/rules.d/60-block.rules
Reading rules file: /usr/lib/udev/rules.d/60-cdrom_id.rules
Reading rules file: /usr/lib/udev/rules.d/60-crda.rules
Reading rules file: /usr/lib/udev/rules.d/60-drm.rules
Reading rules file: /usr/lib/udev/rules.d/60-evdev.rules
Reading rules file: /usr/lib/udev/rules.d/60-fido-id.rules
Reading rules file: /usr/lib/udev/rules.d/60-input-id.rules
Reading rules file: /usr/lib/udev/rules.d/60-multipath.rules
Reading rules file: /usr/lib/udev/rules.d/60-open-vm-tools.rules
Reading rules file: /usr/lib/udev/rules.d/60-persistent-alsa.rules
Reading rules file: /usr/lib/udev/rules.d/60-persistent-input.rules
Reading rules file: /usr/lib/udev/rules.d/60-persistent-storage-dm.rules
Reading rules file: /usr/lib/udev/rules.d/60-persistent-storage-tape.rules
Reading rules file: /usr/lib/udev/rules.d/60-persistent-storage.rules
Reading rules file: /usr/lib/udev/rules.d/60-persistent-v4l.rules
Reading rules file: /usr/lib/udev/rules.d/60-sensor.rules
[... 67 lines cut ...]

DEVLINKS=/dev/zigbee-serial /dev/serial/by-path/pci-0000:00:14.0-usb-0:4:1.0 /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2230427-if00 /dev/ttyACM2

ID_VENDOR=dresden_elektronik_ingenieurtechnik_GmbH
ID_VENDOR_ENC=dresden\x20elektronik\x20ingenieurtechnik\x20GmbH
ID_MODEL=ConBee_II
ID_MODEL_ENC=ConBee\x20II
ID_REVISION=0100
ID_SERIAL=dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2230427
ID_SERIAL_SHORT=DE2230427
ID_TYPE=generic
ID_USB_INTERFACES=:020201:0a0000:
ID_USB_INTERFACE_NUM=00
ID_USB_DRIVER=cdc_acm
ID_MM_CANDIDATE=1
TAGS=:systemd:
USEC_INITIALIZED=2312511529199
Unload module index
Unloaded link configuration context.
```

Is it possible that the zigbee-herdsman does not support DEVLINKS?
Would be happy if I could get a tip. how I can prevent switching. 

Thanks.

Danit2, 2021-12-04:
Did you see this?
https://www.zigbee2mqtt.io/guide/configuration/adapter-settings.html#adapter-settings

You must use `/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2230427-if00` in your config.

zibous, 2021-12-04:
@Danit2 

Thanks, but do not work.

Error: **Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,....**

```

homeassistant: true
permit_join: false
mqtt:
  base_topic: conbee2mqtt
  server: mqtt://mbs.siebler.home:1883
  user: smarthome
  password: seOnly4Me
  include_device_information: true

serial:
  port: /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2230427-if00
```

```
ls /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2230427-if00
lrwxrwxrwx 1 root root 13 Dec  4 09:15 /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2230427-if00 -> ../../ttyACM1
```

```
1891.143624] cdc_acm 1-4:1.0: ttyACM1: USB ACM device
[2321894.820938] usb 1-4: USB disconnect, device number 49
[2321895.166934] usb 1-4: new full-speed USB device number 50 using xhci_hcd
[2321895.317047] usb 1-4: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
[2321895.317053] usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[2321895.317057] usb 1-4: Product: ConBee II
[2321895.317060] usb 1-4: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[2321895.317063] usb 1-4: SerialNumber: DE2230427
```

```
● zigbee2mqtt.service - zigbee2mqtt
     Loaded: loaded (/etc/systemd/system/zigbee2mqtt.service; enabled; vendor preset: enabled)
     Active: active (running) since Sat 2021-12-04 10:46:57 CET; 4s ago
   Main PID: 301160 (node)
      Tasks: 8 (limit: 9347)
     Memory: 86.5M
     CGroup: /system.slice/zigbee2mqtt.service
             ├─301160 /usr/local/bin/node /opt/zigbee2mqtt/index.js
             └─301189 udevadm info -e

Dec 04 10:46:57 nuc2020 systemd[1]: Started zigbee2mqtt.
Dec 04 10:47:01 nuc2020 node[301160]: Zigbee2MQTT:debug 2021-12-04 10:47:01: Loaded state from file /opt/zigbee2mqtt/data/state.json
Dec 04 10:47:01 nuc2020 node[301160]: Zigbee2MQTT:info  2021-12-04 10:47:01: Logging to console only'
Dec 04 10:47:01 nuc2020 node[301160]: Zigbee2MQTT:info  2021-12-04 10:47:01: Starting Zigbee2MQTT version 1.22.1-dev (commit #c47b9c7c)
Dec 04 10:47:01 nuc2020 node[301160]: Zigbee2MQTT:info  2021-12-04 10:47:01: Starting zigbee-herdsman (0.13.176)

Dec  4 10:52:26 nuc2020 node[302549]: #033[31mZigbee2MQTT:error#033[39m 2021-12-04 10:52:26: Failed to start zigbee                                                       │
│Dec  4 10:52:26 nuc2020 node[302549]: #033[31mZigbee2MQTT:error#033[39m 2021-12-04 10:52:26: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-s│
│Dec  4 10:52:26 nuc2020 node[302549]: #033[31mZigbee2MQTT:error#033[39m 2021-12-04 10:52:26: Exiting...                                                                   │
│Dec  4 10:52:26 nuc2020 node[302549]: #033[31mZigbee2MQTT:error#033[39m 2021-12-04 10:52:26: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000m│
│Dec  4 10:52:26 nuc2020 node[302549]:     at ZStackAdapter.start (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:102:27)      │
│Dec  4 10:52:26 nuc2020 node[302549]:     at Controller.start (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/controller.ts:123:29)                         │
│Dec  4 10:52:26 nuc2020 node[302549]:     at Zigbee.start (/opt/zigbee2mqtt/lib/zigbee.ts:58:27)                                                                          │
│Dec  4 10:52:26 nuc2020 node[302549]:     at Controller.start (/opt/zigbee2mqtt/lib/controller.ts:100:27)                                                                 │
│Dec  4 10:52:26 nuc2020 node[302549]:     at start (/opt/zigbee2mqtt/index.js:97:5)                                                                                       │
│Dec  4 10:52:26 nuc2020 systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE                                                               │
│Dec  4 10:52:26 nuc2020 systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.                                                                                  │
│Dec  4 10:52:26 nuc2020 systemd[1]: zigbee2mqtt.service: Scheduled restart job, restart counter is at 13.                                                                 │
│Dec  4 10:52:26 nuc2020 systemd[1]: Stopped zigbee2mqtt.                                                                                                                  │
│Dec  4 10:52:26 nuc2020 systemd[1]: Started zigbee2mqtt.                                                                                                                  │
│Dec  4 10:52:29 nuc2020 influxd[868]: [httpd] 127.0.0.1 - - [04/Dec/2021:10:52:29 +0100] "POST /write?db=telegraf HTTP/1.1" 204 0 "-" "Telegraf/1.15.2+ds1-0ubuntu1 Go/1.1│
│Dec  4 10:52:30 nuc2020 node[302652]: #033[34mZigbee2MQTT:debug#033[39m 2021-12-04 10:52:30: Loaded state from file /opt/zigbee2mqtt/data/state.json                      │
│Dec  4 10:52:30 nuc2020 node[302652]: #033[32mZigbee2MQTT:info #033[39m 2021-12-04 10:52:30: Logging to console only'                                                     │
│Dec  4 10:52:30 nuc2020 node[302652]: #033[32mZigbee2MQTT:info #033[39m 2021-12-04 10:52:30: Starting Zigbee2MQTT version 1.22.1-dev (commit #c47b9c7c)                   │
│Dec  4 10:52:30 nuc2020 node[302652]: #033[32mZigbee2MQTT:info #033[39m 2021-12-04 10:52:30: Starting zigbee-herdsman (0.13.176)                                          │
│Dec  4 10:52:30 nuc2020 node[302652]: #033[34mZigbee2MQTT:debug#033[39m 2021-12-04 10:52:30: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":n│

```

zibous, 2021-12-04:
Now it look like, that it is working:

```
serial:
  port: >-
    /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2230427-if00
  adapter: deconz
```

closed 2021-12-04 by zibous