Koenkk/zigbee2mqtt #9987: TuYa TS011F_plug_3 no set recover state after power outage

ster1um, 2021-12-03:
this plug are supposed in dev branch but not work all.
It’s impossible to set recover state after power off.

Log 

```
Debug Received Zigbee message from '0xa4c138329b21d2b9', type 'attributeReport', cluster 'manuSpecificTuya_3', data '{"53264":2}' from endpoint 1 with groupID 0
Debug No converter available for 'TS011F_plug_3' with cluster 'manuSpecificTuya_3' and type 'attributeReport' and data '{"53264":2}'
```

this is the variant of the tuya plug
[image]

Inferi0r, 2021-12-11:
Agreed. I have 10 of the same new Plugs. power_outage_memory can be changed to 'restore' of 'on', but result will be 'off' for all devices.

ster1um, 2021-12-12:
> Agreed. I have 10 of the same new Plugs. power_outage_memory can be changed to 'restore' of 'on', but result will be 'off' for all devices.

Yes, same behaviour.

taimaiduc, 2021-12-21:
I have the same problem.

ster1um, 2021-12-22:
I set optimistic to false but don’t change nothing!

sergey-1976, 2022-01-17:
I have 3 of the same new Plugs. (TS011F_plug_3 _TZ3000_typdpbpg)
I set "power_outage_memory", but result always is 'off'.

sergey-1976, 2022-02-17:
You need to update the firmware via the Tuya Hub. After that, everything works fine.

closed 2022-03-27 by github-actions[bot]

carloda, 2022-05-24:
Can you see the consumption values ​​with the NAS-WR01B model?