Koenkk/zigbee2mqtt #10005: Philips Hue Ensis (white) - not supported

moonwhaler, 2021-12-04:
Hello,

I just got a brand new Philips Hue Ensis, but the device was reported as "not supported". I am using the latest version of Zigbee2MQTT (1.22.1 commit: c2b5229f).

The device was logged as "929003053301_01" and "929003053301_02".

My guess is, that the device ids are not inside the "philips.js" yet and should be added.

Thank you.

danieledwardgeorgehitchcock, 2021-12-05:
Hi,

mid you could post your database.db entries for this device, and also the [colour temperature](https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_3-adding-converter-s-for-your-device) values, I can add this for you.

Koenkk, 2021-12-05:
I expect it is the same as the black version, added it now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-01-13 by github-actions[bot]