Koenkk/zigbee2mqtt #10008: Ikea STYRBAR  Remote E2002 sends each message twice

nezteew, 2021-12-04:
## What happened
I bought a new Ikea STYRBAR remote control N2
This one: https://www.zigbee2mqtt.io/devices/E2001_E2002.html
The pairing was successfull.
I see the messages in the zigbee log when a button is pressed, but every message is send twice.
```
Dez 04 22:16:37 ioBroker-RasPi npm[5051]: Zigbee2MQTT:info  2021-12-04 22:16:37: MQTT publish: topic 'zigbee2mqtt/tradfri-remote', payload '{"action":"off","action_group":901,"linkquality":47}'
Dez 04 22:16:37 ioBroker-RasPi npm[5051]: Zigbee2MQTT:info  2021-12-04 22:16:37: MQTT publish: topic 'zigbee2mqtt/tradfri-remote', payload '{"action":"off","linkquality":47}'
```

(sometimes one of the two messages has a the action_group field, but not always).

## What did you expect to happen
Only one message is send when a button is pressed

## How to reproduce it (minimal and precise)
Press a button and look into the log

## Debug info
Zigbee2MQTT version: 1.22.0 commit: df0543d4
Adapter hardware: CC2531
Adapter firmware version: 20201127

II was not able to attach the herdsman log as requested, I did not really understand the instruction in https://www.zigbee2mqtt.io/guide/usage/debug.html#enabling-logging

00joshi, 2021-12-06:
I can replicate the issue and provide the herdsman log:

[styrbar.log](https://github.com/Koenkk/zigbee2mqtt/files/7659985/styrbar.log)

Wikibear, 2021-12-22:
Can confirm this issue.

[image]

Koenkk, 2021-12-22:
Did an attempt to fix this issue (at least fixed it from my IKEA E1812). To apply the fix.

- Update to the latest dev branch  (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)
- Repair the device (this is needed to reset the device bindings)
- Please let me know if it works

forHacs, 2021-12-22:
It definitely heals `on`/`off` actions, however `arrow_right_click`/`arrow_left_click` are still doubled.

Zigbee2MQTT version 1.22.1-dev commit: 53003b3

[styrbar.log](https://github.com/Koenkk/zigbee2mqtt/files/7765516/styrbar.log)

Wikibear, 2021-12-22:
Guys please confirm that everything is twice.

Wikibear, 2021-12-23:
Question: IkeaShortButton E1812 send tripple messages. Must i open a new issue or is this the same issue?

Koenkk, 2021-12-23:
@forHacs  fixed (update to the latest dev branch again in a few hours from now)

@Wikibear is this on the latest dev branch? If yes post the herdsman debug log of this 
See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

Wikibear, 2021-12-23:
1.22.1-dev commit: 03f24a63
Updated again and now this version is in use. Seems that Shurtcut button work now. Styrbar ON OFF works now. Styrbar left arrow or right arrow isn't:

`  zigbee-herdsman:deconz:driver query aps data indication +3ms
  zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 216 +1ms
  zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr. 216 srcAddr: 0x3151 destAddr: 0x0 profile id: 0x104 cluster id: 0x5 lqi: 255 +23ms
  zigbee-herdsman:deconz:frameParser response payload: 5,124,17,18,7,1,1,13,0 +0ms
  zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":true,"direction":0,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":18,"manufacturerCode":4476,"commandIdentifier":7},"Payload":{"value":257,"value2":13},"Command":{"ID":7,"parameters":[{"name":"value","type":33},{"name":"value2","type":33}],"name":"tradfriArrowSingle"}},"address":12625,"endpoint":1,"linkquality":255,"groupID":null,"wasBroadcast":false,"destinationEndpoint":1}' +125ms
  zigbee-herdsman:controller:endpoint DefaultResponse 0x0c4314fffef691db/1 5(7, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) +6s
  zigbee-herdsman:deconz:adapter zclFrame.payload: +6s
  zigbee-herdsman:deconz:adapter { cmdId: 7, statusCode: 0 } +1ms
  zigbee-herdsman:deconz:adapter no response expected +0ms
  zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1 +31ms
  zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0x3151 EP:1 SeqNr. 217 request id: 49 +79ms
  zigbee-herdsman:deconz:frameParser DATA_REQUEST RESPONSE - request id: 49 status: 0 +101ms
  zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1 +11ms
  zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10100110 +5ms
  zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 1 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1 +4ms
  zigbee-herdsman:deconz:driver query aps data confirm +1ms
  zigbee-herdsman:deconz:driver DATA_CONFIRM - sending data state request - SeqNr. 218 +7ms
  zigbee-herdsman:deconz:frameParser DATA_CONFIRM RESPONSE - destAddr: 0x3151 request id: 49 confirm status: 0 +17ms
  zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1 +10ms
  zigbee-herdsman:deconz:adapter sendZclFrameToEndpoint - message send with transSeq Nr.: 18 +113ms
`

forHacs, 2021-12-23:
> fixed (update to the latest dev branch again in a few hours from now)

Thanks a lot! Works like a charm on 1.22.1-dev commit: 2a8a98f

forHacs, 2021-12-23:
@Wikibear update one more time, you are one commit behind of the arrows fix.

Wikibear, 2021-12-26:
I think that it works now. Thanks for support.

closed 2021-12-27 by Koenkk