Koenkk/zigbee2mqtt #10009: Lidl Silvercrest radiator thermostat 368308_2010 should be supported but isn't working

JasperE84, 2021-12-04:
## Debug info
Zigbee2MQTT version: 1.22.1
Adapter hardware: zzh!
Adapter firmware version: 
`{'meta': {'maintrel': 1, 'majorrel': 2, 'minorrel': 7, 'product': 1, 'revision': 20201026, 'transportrev': 2}, 'type': 'zStack3x0'}`

## Issue
After pairing a newly purchased [368308_2010 (docs link)](https://www.zigbee2mqtt.io/devices/368308_2010.html) unit, the unit can't be controlled. 
Changing thermostat temperature in hass results in errors in the log.

The payload in the z2m/hass logs doesn't seem to make much sense, which maybe explains the errors in hass. Invalid tuya dpid's maybe? 

Let me know if I can do something to help. Unfortunately I don't own a tuya bridge, planning to return my thermostat unit soon unless there's a fix.

## The hass log:
```
2021-12-04 22:33:46 WARNING (MainThread) [homeassistant.components.mqtt.number] Payload '{"auto_lock":"AUTO","away_mode":null,"away_preset_days":0,"away_preset_temperature":{"0":6,"1":34,"10":96,"11":42,"12":96,"13":34,"14":96,"15":42,"16":96,"17":34,"2":24,"3":42,"4":36,"5":34,"6":68,"7":42,"8":92,"9":34},"battery_low":{"0":2,"1":34,"10":96,"11":42,"12":96,"13":34,"14":96,"15":42,"16":96,"17":34,"2":24,"3":42,"4":36,"5":34,"6":68,"7":42,"8":92,"9":34},"binary_one":null,"binary_two":null,"boost_time":34,"child_lock":null,"comfort_temperature":false,"current_heating_setpoint":"16.0","current_heating_setpoint_auto":null,"detectwindow_temperature":null,"detectwindow_timeminute":null,"eco_temperature":null,"holidays":[{"hour":5,"minute":34,"temperature":24},{"hour":42,"minute":36,"temperature":34},{"hour":4,"minute":42,"temperature":92},{"hour":34,"minute":96,"temperature":42},{"hour":32,"minute":34,"temperature":96},{"hour":42,"minute":96,"temperature":34}],"linkquality":105,"local_temperature_calibration":null,"max_temperature":{"0":19,"1":1,"2":1,"3":0,"4":0,"5":34,"6":0,"7":0},"min_temperature":34,"position":{"0":1,"1":34,"10":96,"11":42,"12":96,"13":34,"14":96,"15":42,"16":96,"17":34,"2":24,"3":42,"4":36,"5":34,"6":68,"7":42,"8":92,"9":34},"voltage":null,"window_detection":"OFF","window_detection_params":{},"workdays":[{"hour":4,"minute":34,"temperature":24},{"hour":42,"minute":36,"temperature":34},{"hour":4,"minute":42,"temperature":92},{"hour":34,"minute":96,"temperature":42},{"hour":32,"minute":34,"temperature":96},{"hour":42,"minute":96,"temperature":34}]}' is not a Number
2021-12-04 22:33:46 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'local_temperature' when rendering '{{ value_json.local_temperature }}'
2021-12-04 22:33:46 ERROR (MainThread) [homeassistant.components.mqtt.climate] Could not parse temperature from
2021-12-04 22:33:46 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'system_mode' when rendering '{{ value_json.system_mode }}'
2021-12-04 22:33:46 ERROR (MainThread) [homeassistant.components.mqtt.climate] Invalid modes mode:
2021-12-04 22:33:46 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'preset' when rendering '{{ value_json.preset }}'
```

## The Z2M log
```
Zigbee2MQTT:debug 2021-12-04 22:33:45: Received Zigbee message from 'thermostaat_zolder_2', type 'commandSetDataResponse', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,32],"type":"Buffer"},"datatype":2,"dp":16,"fn":0,"status":0,"transid":16}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2021-12-04 22:33:45: MQTT publish: topic 'zigbee2mqtt/thermostaat_zolder_2', payload '
{
  "auto_lock": "AUTO",
  "away_mode": null,
  "away_preset_days": 0,
  "away_preset_temperature": {
    "0": 6,
    "1": 34,
    "2": 24,
    "3": 42,
    "4": 36,
    "5": 34,
    "6": 68,
    "7": 42,
    "8": 92,
    "9": 34,
    "10": 96,
    "11": 42,
    "12": 96,
    "13": 34,
    "14": 96,
    "15": 42,
    "16": 96,
    "17": 34
  },
  "battery_low": {
    "0": 2,
    "1": 34,
    "2": 24,
    "3": 42,
    "4": 36,
    "5": 34,
    "6": 68,
    "7": 42,
    "8": 92,
    "9": 34,
    "10": 96,
    "11": 42,
    "12": 96,
    "13": 34,
    "14": 96,
    "15": 42,
    "16": 96,
    "17": 34
  },
  "binary_one": null,
  "binary_two": null,
  "boost_time": 34,
  "child_lock": null,
  "comfort_temperature": false,
  "current_heating_setpoint": "16.0",
  "current_heating_setpoint_auto": null,
  "detectwindow_temperature": null,
  "detectwindow_timeminute": null,
  "eco_temperature": null,
  "holidays": [
    {
      "hour": 5,
      "minute": 34,
      "temperature": 24
    },
    {
      "hour": 42,
      "minute": 36,
      "temperature": 34
    },
    {
      "hour": 4,
      "minute": 42,
      "temperature": 92
    },
    {
      "hour": 34,
      "minute": 96,
      "temperature": 42
    },
    {
      "hour": 32,
      "minute": 34,
      "temperature": 96
    },
    {
      "hour": 42,
      "minute": 96,
      "temperature": 34
    }
  ],
  "linkquality": 105,
  "local_temperature_calibration": null,
  "max_temperature": {
    "0": 19,
    "1": 1,
    "2": 1,
    "3": 0,
    "4": 0,
    "5": 34,
    "6": 0,
    "7": 0
  },
  "min_temperature": 34,
  "position": {
    "0": 1,
    "1": 34,
    "2": 24,
    "3": 42,
    "4": 36,
    "5": 34,
    "6": 68,
    "7": 42,
    "8": 92,
    "9": 34,
    "10": 96,
    "11": 42,
    "12": 96,
    "13": 34,
    "14": 96,
    "15": 42,
    "16": 96,
    "17": 34
  },
  "voltage": null,
  "window_detection": "OFF",
  "window_detection_params": {},
  "workdays": [
    {
      "hour": 4,
      "minute": 34,
      "temperature": 24
    },
    {
      "hour": 42,
      "minute": 36,
      "temperature": 34
    },
    {
      "hour": 4,
      "minute": 42,
      "temperature": 92
    },
    {
      "hour": 34,
      "minute": 96,
      "temperature": 42
    },
    {
      "hour": 32,
      "minute": 34,
      "temperature": 96
    },
    {
      "hour": 42,
      "minute": 96,
      "temperature": 34
    }
  ]
}
'
Zigbee2MQTT:debug 2021-12-04 22:37:12: Received Zigbee message from 'thermostaat_zolder_2', type 'attributeReport', cluster 'genBasic', data '{"appVersion":66}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-12-04 22:38:10: Received Zigbee message from 'thermostaat_zolder_2', type 'attributeReport', cluster 'genBasic', data '{"appVersion":66}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-12-04 22:39:52: Received Zigbee message from 'thermostaat_zolder_2', type 'attributeReport', cluster 'genBasic', data '{"appVersion":66}' from endpoint 1 with groupID 0

```

## The thermostat card as it looks like in hass
[image]

gmarchart, 2021-12-06:
Hello, this is already reported - the z2m edge support the valve without external converter, now. 

#9509 
#8934

JasperE84, 2021-12-06:
Excellent! Changing the docker to :latest-dev (didn't know about that tag yet) indeed gives me a working solution.

For references sake, here's a payload as its now cached:

```
    {
        "current_heating_setpoint": "0.0",
        "local_temperature": "20.0",
        "system_mode": "off",
        "away_mode": "OFF",
        "preset": "manual",
        "child_lock": "UNLOCK",
        "voltage": 1540,
        "comfort_temperature": "15.0",
        "eco_temperature": "17.0",
        "away_preset_year": 21,
        "away_preset_month": 12,
        "away_preset_day": 4,
        "away_preset_hour": 21,
        "away_preset_minute": 8,
        "away_preset_temperature": "17.0",
        "away_preset_days": 1,
        "local_temperature_calibration": "0.0",
        "current_heating_setpoint_auto": "17.0",
        "binary_two": "OFF",
        "detectwindow_temperature": "15.0",
        "detectwindow_timeminute": 0,
        "monday_temp_1": "17.0",
        "monday_hour_1": "06",
        "monday_minute_1": "00",
        "monday_temp_2": "21.0",
        "monday_hour_2": "09",
        "monday_minute_2": "00",
        "monday_temp_3": "17.0",
        "monday_hour_3": "17",
        "monday_minute_3": "00",
        "monday_temp_4": "21.0",
        "monday_hour_4": "23",
        "monday_minute_4": "00",
        "monday_temp_5": "17.0",
        "monday_hour_5": "24",
        "monday_minute_5": "00",
        "monday_temp_6": "21.0",
        "monday_hour_6": "24",
        "monday_minute_6": "00",
        "monday_temp_7": "17.0",
        "monday_hour_7": "24",
        "monday_minute_7": "00",
        "monday_temp_8": "21.0",
        "monday_hour_8": "24",
        "monday_minute_8": "00",
        "monday_temp_9": "17.0",
        "tuesday_temp_1": "17.0",
        "tuesday_hour_1": "06",
        "tuesday_minute_1": "00",
        "tuesday_temp_2": "21.0",
        "tuesday_hour_2": "09",
        "tuesday_minute_2": "00",
        "tuesday_temp_3": "17.0",
        "tuesday_hour_3": "17",
        "tuesday_minute_3": "00",
        "tuesday_temp_4": "21.0",
        "tuesday_hour_4": "23",
        "tuesday_minute_4": "00",
        "tuesday_temp_5": "17.0",
        "tuesday_hour_5": "24",
        "tuesday_minute_5": "00",
        "tuesday_temp_6": "21.0",
        "tuesday_hour_6": "24",
        "tuesday_minute_6": "00",
[... 111 lines cut ...]
        "sunday_hour_2": "09",
        "sunday_minute_2": "00",
        "sunday_temp_3": "17.0",
        "sunday_hour_3": "17",
        "sunday_minute_3": "00",
        "sunday_temp_4": "21.0",
        "sunday_hour_4": "23",
        "sunday_minute_4": "00",
        "sunday_temp_5": "17.0",
        "sunday_hour_5": "24",
        "sunday_minute_5": "00",
        "sunday_temp_6": "21.0",
        "sunday_hour_6": "24",
        "sunday_minute_6": "00",
        "sunday_temp_7": "17.0",
        "sunday_hour_7": "24",
        "sunday_minute_7": "00",
        "sunday_temp_8": "21.0",
        "sunday_hour_8": "24",
        "sunday_minute_8": "00",
        "sunday_temp_9": "17.0",
        "binary_one": "OFF"
    }

```

closed 2021-12-06 by JasperE84

matzedo, 2021-12-15:
Hi @JasperE84 , can you let us know how you set the radiator into pairing mode, as this is neither described in the manufactures userguide, nor on the [linked device page](https://www.zigbee2mqtt.io/devices/368308_2010.html)? Thanks in advance.

gmarchart, 2021-12-15:
> Hi @JasperE84 , can you let us know how you set the radiator into pairing mode, as this is neither described in the manufactures userguide, nor on the [linked device page](https://www.zigbee2mqtt.io/devices/368308_2010.html)? Thanks in advance.

On side 26 of the manual -> press button 9 Rotarybutton above the unit until you see pair in the display of valve