Koenkk/zigbee2mqtt #9991: ZNCZ04LM specific setting 'temperature_precision' has no effect

degreal, 2021-12-04:
## What happened
Specific option 'temperature_precision: 2' has no effect. Still reporting temperature with 1 degree precision.
## What did you expect to happen
Temperature reading of '24.12' degrees instead of '24' degrees.

## How to reproduce it (minimal and precise)
Open device tab 'Settings (specific)'
Slide value from 0 to 2
Wait for next temperature update
[image]

## Debug info
Zigbee2MQTT version: 1.22.1-1
Adapter hardware: CC2652P
Adapter firmware version: zStack3x0 20210708

Logs of changing precision setting and waiting for temperature update
https://www.toptal.com/developers/hastebin/ebapecucol.apache

Koenkk, 2021-12-04:
The calibration is applied as soon as the device reports a new temperature.

degreal, 2021-12-04:
> The calibration is applied as soon as the device reports a new temperature.

Yes. I can confirm that temperature_calibration works but temperature_precision does not. Or am misundestanding the temperature_precision setting?

Koenkk, 2021-12-05:
@degreal it could be that this device does not support more precise power measurements.

closed 2022-01-13 by github-actions[bot]