Koenkk/zigbee2mqtt #9993: SONOFF ZIGBEE MINI - Ability to change modes (PULSE - EDGE - Follow)

sareju, 2021-12-04:
SONOFF introduced the 3 different modes to setup the ZIGBEE mini and these settings should be available via MQTT to make the devices usable. Thanks a lot!

francisp2, 2021-12-06:
Do you need a firmware update for that ? And where to find the firmware ?

sareju, 2021-12-06:
Hi,

   first of all let me thanks a lot for the answer! I am in an Italian Home assistant forum and I am seeing many people having the same question.
   At the end I even ordered the SONOFF Zigbee bridge as last resort fix... but having the ability to do the setting via Zignee2mqtt would be great!!!
   I will try to do some search on Wednesday and eventually I think we could ask to ITEAD to collaborate on this subject as they will sell man more of their boxes that are quite good quality and interesting price.

Thanks a lot,

Rosario Savoia

Mobile (& WA): +39 334 6814068

closed 2022-01-13 by github-actions[bot]

NewRaiden, 2022-01-21:
Still no answers for that? Did you find a solution?

NewRaiden, 2022-01-21:
> Hi, first of all let me thanks a lot for the answer! I am in an Italian Home assistant forum and I am seeing many people having the same question. At the end I even ordered the SONOFF Zigbee bridge [...]

Trovata una soluzione?

francisp2, 2022-01-21:
How can one find a solution if no one can find the firmware update to enable it ?

NewRaiden, 2022-01-21:
> francisp2

I saw from a youtuber that his device was by default in impulse mode (it was 2020). He did not update the firmware cause He said that was unable to pair it with the SONOFF Bridge and his "ewelink" app. 

Video here: https://www.youtube.com/watch?v=e-GMPQkENkA

That was one of the first reviews for that device... Maybe that was a test version; but maybe the pulse mode does not require a firmware update.

francisp2, 2022-01-21:
I tried to watch that video, but I can't make heads or tails from the (translated) English subtitles.

francisp2, 2022-01-21:
Here they still show a momentary switch is not possible: 
https://itead.cc/product/sonoff-zbmini-zigbee-smart-switch/

NewRaiden, 2022-01-21:
> I tried to watch that video, but I can't make heads or tails from the (translated) English subtitles.

Essentially He shows the differences between Zigbee and Wifi version (the video was from 2020). 
He than says that He was unable to connect It using the SONOFF Official Bridge so He used the Tasmota bridge; He shows that the default mode was the pulse one infact the button worked perfectly.

Giancky79, 2023-05-27:
I found a solution but is hardware

https://community.home-assistant.io/t/zbmini-in-zha-enable-edge-mode/407170/10