Koenkk/zigbee2mqtt #10019: 'NHPB/SHUTTER/1' (aka Schneider Wiser Shutter) support with set timer position

fariouche, 2021-12-05:
Hello,
This is about the support of Schneider Shutter controller.
https://www.se.com/ww/en/product/S520567W/odace---switch---roller-blind---for-wiser---white/

Unfortunately I still setting up my zigbee2mqtt installation, and the shutters are still not connected to the zigbee2mqtt yet (I'm still using the Schneider hub), and when I've done the tests with Z2M I forgot to dump the endpoints...

But I know that the following worked:

const exposes = require('zigbee-herdsman-converters/libs/exposes');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
cont tz = require('zigbee-herdsman-converters/converters/toZigbee');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

module.exports = [
{
    zigbeeModel: ['NHPB/SHUTTER/1'],
    model: 'NHPB/SHUTTER/1',
    vendor: 'Schneider Electric',
    description: 'Schneider Wiser Shutter Controller',
    fromZigbee: [fz.cover_position_tilt, fz.command_cover_close, fz.command_cover_open, fz.command_cover_stop],
    toZigbee: [tz.cover_state, tz.cover_position_tilt],
    exposes: [e.cover_position()],
    meta: {coverInverted: true},
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1) || device.getEndpoint(5);
        await reporting.bind(endpoint, coordinatorEndpoint, ['closuresWindowCovering']);
        await reporting.currentPositionLiftPercentage(endpoint);
    },
}];

I do not remember where I've found it exactly
Only issue is the famous problem of the 2min timer that prevents the accurate reporting of the position.
But it fortunately I own a Wiser Home touch hub, and I do have a wireshark capture that sets the shutter control timer.

The command sent by the hub to set the timer to 26 seconds is this one:
00 05 02 01 04 01 01 e3 14 5e 10 2f 02 00 e0 21 1a 00

From wireshark decoding, it is a write attributes command (0x02), attribute field 0xe000, 16 bits UInt (0x21), value 0x001a
I will try to add it to the code (but I need some time as I do not know where to add it in zigbee-herdsman code yet)

And last, this device supports OTA, but I've found nothing about the firmwares URLs. There is currently no firmware update available, so not sure it will be interresting for me to invest time searching the wifi TCP dump from the hub to retrieve the URLs...

toto9262, 2021-12-05:
Hello, I have the same shutter. This is my config, if  it helps you

 const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['NHPB/SHUTTER/1'],
    model: 'S520567',
    vendor: 'Schneider Electric',
    description: 'Volets',

        fromZigbee: [fz.cover_position_tilt, fz.command_cover_close, fz.command_cover_open, fz.command_cover_stop],
        toZigbee: [tz.cover_position_tilt, tz.cover_state, tz.command_cover_stop],
        exposes: [e.cover_position()],
        meta: {coverInverted: true},
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1) || device.getEndpoint(5);
            await reporting.bind(endpoint, coordinatorEndpoint, ['closuresWindowCovering']);
            await reporting.currentPositionLiftPercentage(endpoint);
   },
};

module.exports = definition;

fariouche, 2021-12-06:
Yes, same as mine.
I hope that it will be easy to integrate the set timer, so that we can use the set position/get position in %

fariouche, 2021-12-28:
@Koenkk thank you, I see that the device is now added.
I'm currently looking at adding the lift duration so that percentage can work...
From what I see, should I do the same as the ubisys j1 device and add a schneider_configure_xxx section with only an attribute open_to_closed_s? (it does not have all the other configurations I think)
I still have to figure out where the actual command is sent to the device... :-)

Koenkk, 2021-12-29:
@fariouche the Ubisys (`ubisys_configure_j1`) are specific for the Ubisys brand, so they won't work with schneider electric. You can check if such feature is supported when using it with the original gateway, if it is supported reverse engineer the send commands by sniffing: https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html#with-cc2531

fariouche, 2021-12-29:
Yes, this is already done and I put it in the initial comment. I have the command sent, decoded it... Just need to to send it now using z2m but I don't know yet how to implement it, so I was searching for a reference code to copy and replace with the correct command :)

Koenkk, 2021-12-29:
Can you share your `pcapng` file + network key when sending this command with the original gateway? Also provide the packet No. of this command (first column in wireshark)

fariouche, 2021-12-29:
[wiser_shutter_set_timer.pcapng.zip](https://github.com/Koenkk/zigbee2mqtt/files/7789957/wiser_shutter_set_timer.pcapng.zip)

The frame is number 415.
The key is "bcc6f2b284f520079b82a9e17ea8a15e"

There is a lot of packets because I have also 6 radiator valves from same manufacturer and they are very chatty

Koenkk, 2021-12-29:
Try with the following external converter:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const tzLocal = {
    lift_duration: {
        key: ['lift_duration'],
        convertSet: async (entity, key, value, meta) => {
            await entity.write(0x0102, {0xe000: {value, type: 0x21}}, {manufacturerCode: 0x105e});
            return {state: {lift_duration: value}};
        },
    }
}

const definition = {
    zigbeeModel: ['NHPB/SHUTTER/1'],
    model: 'S520567',
    vendor: 'Schneider Electric',
    description: 'Roller shutter',
    fromZigbee: [fz.cover_position_tilt],
    toZigbee: [tz.cover_position_tilt, tz.cover_state],
    exposes: [e.cover_position(), exposes.numeric('lift_duration', ea.STATE_SET).withUnit('seconds').withValueMin(0).withValueMax(300).withDescription('Duration of lift')],
    meta: {coverInverted: true},
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(5);
        await reporting.bind(endpoint, coordinatorEndpoint, ['closuresWindowCovering']);
        await reporting.currentPositionLiftPercentage(endpoint);
    },
};

module.exports = definition;

```

fariouche, 2021-12-30:
Thank you very much! Much quicker than me.
Sorry, took me some time to figure out why domoticz was crashing after the new update...

So, it is recognized... then when I go to "expose" tab and change the lift_duration, I get an error "no converter available for lift_duration" (I'm using Z2M 1.22.0)

fariouche, 2021-12-30:
ok I added tzLocal.lift_duration to toZigbee, and now it works...

Tried 50% and then pressed the "double blue arrows" and it went down for some time and stopped to the reported 50% position. The position is also updated accordingly. great!
Thank you!

fariouche, 2021-12-30:
Now, I try to add a second shutter.... and this time I got a crash when changing the duration:

MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":false},"status":"ok","transaction":"yt87l-9"}'
Zigbee2MQTT:error 2021-12-30 20:19:25: Publish 'set' 'lift_duration' to '0xd0cf5efffe493b88' failed: 'Error: Write 0xd0cf5efffe493b88/21 closuresWindowCovering({"57344":{"value":26,"type":33}}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4190,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
Error: DatabaseEntry with ID '2' already exists
    at Database.insert (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/database.ts:35:19)
    at Device.undelete (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/device.ts:326:25)
    at Controller.onDeviceJoined (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/controller.ts:520:20)

fariouche, 2021-12-30:
Sorry, me again: it seems that after the shutter just joined, I cannot set the lift_duration with the error above (Unsupported attribute)
If I try again it will eventually crash like above.
If I restart z2m it works.

Koenkk, 2021-12-31:
whoops forgot to add it to toZigbee indeed. Added this now, changes will be available in tomorrows release.

Assuming this can be closed now

closed 2021-12-31 by Koenkk

KKox-26, 2022-01-04:
Thanks, @fariouche  and @Koenkk its works perfectly with the external converter.

bonjour81, 2022-01-19:
Hi! Shall I understand that the S520567 now supports lift_duration /set command in a similar way it's described for the other schneider module https://www.zigbee2mqtt.io/devices/CCT5015-0001.html 
? 
It seems not yet described in the device page of the  S520567

KKox-26, 2022-01-20:
Hi,

Yes exactly is you use the external converter only, maybe that's why it is not in the device page for now.

bonjour81, 2022-01-21:
I assume it may be integrated later (without need of external converter) in z2m ?

graphicsxp, 2022-02-19:
Hello. I just purchased this device and it is recognized by z2m. It shows up in the list of devices.
however if I try to open/close, nothing happens. No logs either.

Is this because this device is not really supported, although it is listed as supported on z2m website ? I'vre read about this external converter above, is it still required  ?

b0z02003, 2022-03-04:
> whoops forgot to add it to toZigbee indeed. Added this now, changes will be available in tomorrows release.
> 
> Assuming this can be closed now

Hello, the above patch adds support for lift_duration to 'PUCK/SHUTTER/1', is there any plan to include a similar patch for the previously mentioned 'NHPB/SHUTTER/1' without the need for an external converter ?

ya-kay, 2022-04-14:
Thanks a lot! 
It also works for the model "1GANG/SHUTTER/1" Model: MEG5113-0300/MEG5165-0000.
I did the same adjustment like you did on "PUCK/SHUTTER/1".
(Maybe the support of lift_duration can be added on https://www.zigbee2mqtt.io/devices/MEG5113-0300_MEG5165-0000.html)

phil239, 2022-05-16:
> Thanks a lot!
> It also works for the model "1GANG/SHUTTER/1" Model: MEG5113-0300/MEG5165-0000.
> I did the same adjustment like you did on "PUCK/SHUTTER/1".
> (Maybe the support of lift_duration can be added on https://www.zigbee2mqtt.io/devices/MEG5113-0300_MEG5165-0000.html)

Thanks for your work!
I also own several MEG5113-0300 and would now like to get the external converter working. I already added it successfully to my config but when I try to adjust lift_duration on the GUI, I get the following error: "No converter available for 'lift_duration' (xx)"

Any ideas?

axellebot, 2022-08-23:
It seems like those manufacturer model also match this zigbee model :
- S530567
- NU350820

Francky3084, 2024-07-25:
Hi,
Still surprised that in July 2024, it stills not basicly and managed correctly without external converter.
Is there a reason for that ?

KKox-26, 2024-07-25:
> Hi, Still surprised that in July 2024, it stills not basicly and managed correctly without external converter. Is there a reason for that ?

Model S520567 is supporter without external converter for a long time.

Francky3084, 2024-07-25:
> > Hi, Still surprised that in July 2024, it stills not basicly and managed correctly without external converter. Is there a reason for that ?
> 
> Model S520567 is supporter without external converter for a long time.

On my side, on a very fresh install of Zigbee2MQTT, my S520567 is recongnized out of the box ... but I can't interract correctly with it. (Error popups and so on)
Until converter is in place, works like a charm.

So, i don't know what's going on but on my side, still needed.