Koenkk/zigbee2mqtt #10020: IKEA E1766 not binding to groups, groupID is null

MrHollowPS, 2021-12-05:
## What happened
I'm trying to Bind an IKEA E1766 (firmware 2.3.079) remote to a group but the groupID is still null when running commands

## What did you expect to happen
I expect the remote to be able to control the group items and be bound to it.

## How to reproduce it (minimal and precise)
Create a group, bind the remote

## Debug info
Zigbee2MQTT version: 1.22.1-1
Adapter hardware: Conbee II
Adapter firmware version: Latest Firmware (deCONZ_ConBeeII_0x26720700.bin.GCF)

`Debug Received MQTT message on 'zigbee2mqtt/bridge/request/group/add' with data '{"friendly_name":"Blinds","transaction":"t4a8f-8"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/response/group/add', payload '{"data":{"friendly_name":"Blinds","id":1},"status":"ok","transaction":"t4a8f-8"}'
Debug Received MQTT message on 'zigbee2mqtt/bridge/request/device/bind' with data '{"from": "Living Room/Remote", "to": "Blinds"}'
Debug binding cluster 'genOnOff' from 'Living Room/Remote' to 'Blinds'
Debug Received Zigbee message from 'Living Room/Remote', type 'commandUpOpen', cluster 'closuresWindowCovering', data '{}' from endpoint 1 with groupID null
Info MQTT publish: topic 'zigbee2mqtt/Living Room/Remote', payload '{"action":"open","battery":74,"linkquality":255,"update":{"state":"idle"},"update_available":null}'
Info MQTT publish: topic 'zigbee2mqtt/Living Room/Remote', payload '{"action":"","battery":74,"linkquality":255,"update":{"state":"idle"},"update_available":null}'
Info MQTT publish: topic 'zigbee2mqtt/Living Room/Remote/action', payload 'open'
Info Successfully bound cluster 'genOnOff' from 'Living Room/Remote' to 'Blinds'
Debug binding cluster 'genLevelCtrl' from 'Living Room/Remote' to 'Blinds'
Info Successfully bound cluster 'genLevelCtrl' from 'Living Room/Remote' to 'Blinds'
Debug binding cluster 'closuresWindowCovering' from 'Living Room/Remote' to 'Blinds'
Info Successfully bound cluster 'closuresWindowCovering' from 'Living Room/Remote' to 'Blinds'
Info MQTT publish: topic 'zigbee2mqtt/bridge/response/device/bind', payload '{"data":{"clusters":["genOnOff","genLevelCtrl","closuresWindowCovering"],"failed":[],"from":"Living Room/Remote","to":"Blinds"},"status":"ok"}'
Debug Received Zigbee message from 'Living Room/Remote', type 'commandDownClose', cluster 'closuresWindowCovering', data '{}' from endpoint 1 with groupID null
Info MQTT publish: topic 'zigbee2mqtt/Living Room/Remote', payload '{"action":"close","battery":74,"linkquality":255,"update":{"state":"idle"},"update_available":null}'
Info MQTT publish: topic 'zigbee2mqtt/Living Room/Remote', payload '{"action":"","battery":74,"linkquality":255,"update":{"state":"idle"},"update_available":null}'
Info MQTT publish: topic 'zigbee2mqtt/Living Room/Remote/action', payload 'close'`

Koenkk, 2021-12-05:
Seems ikea removed group binding functionality in their latests firmwares, see https://github.com/Koenkk/zigbee2mqtt/issues/7716

MrHollowPS, 2021-12-05:
That explains why it I was getting tons of errors when playing with the bind/unbind commands. I had to manually remove the bindings from database file... Well thanks for the update.

closed 2021-12-05 by Koenkk