Koenkk/zigbee2mqtt #10027: Agara E1 error after update to 1.22

rbalaev, 2021-12-05:
**After the update to 1.22.1 from 1.21 I have the following several errors in my HA log:**

`Logger: homeassistant.helpers.template
Source: helpers/template.py:566
First occurred: 01:42:55 (421 occurrences)
Last logged: 21:12:21

Error parsing value: access to attribute 'update' of 'dict' object is unsafe. (value: {"action":"single","last_seen":"2021-12-05T18:04:21.295Z","linkquality":96,"operation_mode":"control_relay","power_outage_memory":true,"state":"OFF","temperature":28,"update_available":null}, template: {{ value_json['update']['state'] == "available" }})
Error parsing value: access to attribute 'update' of 'dict' object is unsafe. (value: {"action":"","last_seen":"2021-12-05T18:04:21.295Z","linkquality":96,"operation_mode":"control_relay","power_outage_memory":true,"state":"OFF","temperature":28,"update_available":null}, template: {{ value_json['update']['state'] == "available" }})
Error parsing value: access to attribute 'update' of 'dict' object is unsafe. (value: {"action":null,"last_seen":"2021-12-05T18:06:26.781Z","linkquality":90,"operation_mode":"control_relay","power_outage_memory":true,"state":"OFF","temperature":28,"update_available":null}, template: {{ value_json['update']['state'] == "available" }})
Error parsing value: access to attribute 'update' of 'dict' object is unsafe. (value: {"action":null,"last_seen":"2021-12-05T18:11:25.427Z","linkquality":84,"operation_mode_left":"decoupled","operation_mode_right":"decoupled","power_outage_memory":true,"state":"ON","state_left":"ON","state_right":"OFF","temperature":31,"update_available":null}, template: {{ value_json['update']['state'] == "available" }})
Error parsing value: access to attribute 'update' of 'dict' object is unsafe. (value: {"action":null,"last_seen":"2021-12-05T18:12:21.449Z","linkquality":87,"operation_mode":"control_relay","power_outage_memory":true,"state":"OFF","temperature":28,"update_available":null}, template: {{ value_json['update']['state'] == "available" }})`

**Device**
- QBKG41LM
- QBKG39LM
- QBKG40LM

Zigbee2MQTT version: 1.22.1-1
Adapter hardware: Egony Stick V4 (Ebyte ver.)

Koenkk, 2021-12-06:
Duplicate of #9939 and should already be fixed in the dev branch.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2021-12-06 by Koenkk

rbalaev, 2021-12-06:
> Duplicate of #9939 and should already be fixed in the dev branch.
> 
> Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

I am not using the development branch, when will it be in release?

Koenkk, 2021-12-06:
1 January