Koenkk/zigbee2mqtt #10033: Xiaomi Aqara Opple (WXCJKG13LM) "loses its status" after apprx. 1 hour (each button press is seen as "action": "null")

odelma, 2021-12-05:
## What happened
Updated zigbee2mqtt to latest beta yesterday, currently on 1.22.1-dev (zStack3x0 20201026). Experienced some very strange behavior like these Aqara switches going to command mode, one device group started controlling  a switch that did not belong to it etc. After some unpainring-repairing and recreation of the misbehaving group, I was happy that eveything seemed to work. But this joy was too soon. I have 6 WXCJKG13LM's and after a while none of them worked. I then noticed that they were sending action: null (and no separate message to action topic). Also voltage is showing null and battery 0. Once I press athe activation button in the back, they start working again (only to be going back to not working after about an hour). When they are revived with the activation button I can see:

MQTT publish: topic 'zigbee2mqtt/Kylpyhuoneen katkaisin2', payload '{"action":"","battery":100,"linkquality":93,"operation_mode":"event","state":"OFF","temperature":25,"voltage":3073}'
Info 
MQTT publish: topic 'zigbee2mqtt/Kylpyhuoneen katkaisin2/action', payload 'button_2_single'

After about and hour just: 
MQTT publish: topic 'zigbee2mqtt/Kylpyhuoneen katkaisin2', payload '{"action":"null","battery":0,"linkquality":93,"operation_mode":"event","state":"OFF","temperature":25,"voltage":null}'

## What did you expect to happen
I expect them to would continue working without the need to press the wakeup button every hour as they did before the udpate.

## How to reproduce it (minimal and precise)
Update to latest beta and watch what happens with WXCJKG13LM.

## Debug info
Zigbee2MQTT version:  1.22.1-dev.
Adapter hardware: Electrolama zzh!
Adapter firmware version: zStack3x0 20201026

This is how the devices that are "lost" are showing (after about of 1 hour of not using them):
[image]

[image]

salopette, 2021-12-06:
https://github.com/Koenkk/zigbee2mqtt/issues/9952#issuecomment-986985249

odelma, 2021-12-07:
Seems to be fixed by https://github.com/Koenkk/zigbee-herdsman-converters/commit/edea8525dcf5653e4eea8cd1655c25a9156c8e0a (ref. https://github.com/Koenkk/zigbee2mqtt/issues/9952)

[image]

closed 2021-12-07 by odelma