Koenkk/zigbee2mqtt #10036: thermostat TS0601 and manufacturer code _TZE200_hue3yfsn,

Hewelius, 2021-12-06:
Dear colleagues programmers,
I'm trying to add to my system Tuya heater thermostat TS0601 and manufacturer code _TZE200_hue3yfsn. Generally it looks to be supported. I have checked in tuya.js. But I'm still getting info from z2m that it is not supported. I have been trying to add local converter but other problem appears: e.window_open is not a function.
Sorry for screenshot in polish.

[image]

_Originally posted by @Hewelius in https://github.com/Koenkk/zigbee2mqtt/discussions/10031_

fardem, 2021-12-06:
Hi,

do you have updated the zigmee2mqtt? In the newest version must be you're device are included.

look here and search for you device "TZE200_hue3yfsn"

https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/tuya.js

Hewelius, 2021-12-06:
Hi, 
I was pretty sure, that this is up to date, because of set "automatic updates". And finally the version of z2m was outdated.

Thank you a lot.

closed 2021-12-06 by Hewelius

Chaosinme666, 2022-02-24:
@Hewelius I got the same problem as well. Z2M update doesn't help at all. Do you have any solution?