Koenkk/zigbee2mqtt #10043: Turning on availability globally depletes IKEA E1524/E1810 remote battery

jum, 2021-12-06:
## What happened

I have recently turned on availability on my ZigBee network globally, with no exceptions.

## What did you expect to happen

Just to be able to monitor the state of bulbs still being turned on by mechanical switches.

## How to reproduce it (minimal and precise)

Just turn on availability and a few weeks afterwards all IKEA remotes have no battery power. Just changed three remote batteries the second time in two months. I have now excluded all the remotes from availability and will see how that turns out.

## Debug info
Zigbee2MQTT version:
Adapter hardware: CC2531, CC2530, CC26X2R1, CC1352P-2, Conbee II
Adapter firmware version:

Zigbee2MQTT:info  2021-12-04 10:59:47: Starting Zigbee2MQTT version 1.22.1 (commit #c2b5229f)
Zigbee2MQTT:info  2021-12-04 10:59:47: Starting zigbee-herdsman (0.13.176)
Zigbee2MQTT:info  2021-12-04 10:59:58: zigbee-herdsman started (resumed)
Zigbee2MQTT:info  2021-12-04 10:59:58: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20210708,"transportrev":2},"type":"zStack3x0"}'
Zigbee2MQTT:info  2021-12-04 10:59:58: Currently 40 devices are joined:
Zigbee2MQTT:info  2021-12-04 10:59:58: Schlafzimmer/Remote (0x842e14fffe0f6e3d): E1524/E1810 - IKEA TRADFRI remote control (EndDevice)

Koenkk, 2021-12-06:
Availability feature has no impact on the battery of end devices, the way it works is that it sets a timer of 25 hours (at the z2m side), if a message is received the timer is reset, if the timer is trigged the device is marked as unavailable.

jum, 2021-12-06:
Well, OK. Any other idea why these devices empty their batteries in no time? Anything I can do to debug this?

Koenkk, 2021-12-07:
Note sure what is the reason for it, try OTA upgrading your device I also found https://github.com/Koenkk/zigbee2mqtt/issues/3444

closed 2022-01-15 by github-actions[bot]

syyyr, 2022-04-28:
Hi, my IKEA remote also quickly deleted battery. I tried the solution written here and it worked https://github.com/Koenkk/zigbee2mqtt/issues/3444#issuecomment-1052994262. Five days in and the battery is still fine.

jum, 2022-05-26:
I found some strange things while doing this:

Changing batteries required some major voodoo for me. With my Slaesh stick as coordinator I could not get any of the remotes with depleted batteries online, even with full debug of Herdsman nothing would show up. I had to connect my TI experimental stick, then I would see from the debug output that interview failed due to a timeout while configuring the device (could not set the group). I had to force remove the remotes, then pairing them again. After that I removed the IKEA default group they where in and put them in the proper group. Reconfiguring after that was fine. I then switched the coordinator back to the Sleash stick, all is fine so far. I will monitor the batteries in the next days.