Koenkk/zigbee2mqtt #10044: LEDVANCE AC26940/AC31266 throws error

msmthng, 2021-12-06:
## What happened
The device was interviewed successfully but it will be displayed as OSRAM | AB3257001NJ

When I try to send ON or OFF state after a while I get following error:

Publish 'set' 'state' to 'xDevicex' failed: 'Error: Command 0x7cb03eaa0a0c7ba0/3 genOnOff.off({}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 17328 - 3 - 58 - 6 - 11 after 10000ms)'

## What did you expect to happen

I would expect that the device is working since it is listed under supported devices.

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version: 1.22.0
Adapter hardware: cc2538
Adapter firmware version: "product":1,"revision":20210120,"transportrev":2},"type":"zStack3x0"

sjorge, 2021-12-06:
`Timeout - 17328 - 3 - 58 - 6 - 11 after 10000ms` could mean it has a hard time reaching the device, try moving it closer to another router or the coordinator

Doxlike, 2021-12-07:
Hi,

I think I have the same problem. But I got the error since I got these ledvance plugs on any device in my network

Maybe they create this issues for the whole network

Br

Doxlike, 2021-12-11:
> ````

````@msmthng I changed the ledvance plugs from router to enddevice. My network seems to be fast again and the errors are gone (no network route found). Can you please try the same so we will know if this workaround solves the issue.

For this:
- please stop zigbee2mqtt and open the database.db file. 
- Search for your ledvance plugs and change "Router" to "EndDevice"
- start zigbee2mqtt again and check if it's working

Kind regards
Dox

Doxlike, 2021-12-11:
> @msmthng I changed the ledvance plugs from router to enddevice. My network seems to be fast again and the errors are gone (no network route found). Can you please try the same so we will know if [...]

Forget about, it's slow again :(

Doxlike, 2021-12-12:
Okay, problems reappear not as worse as before but sometimes lights or plugs are not turning on.

closed 2022-01-19 by github-actions[bot]