Koenkk/zigbee2mqtt #10055: Error parsing  value: access to attribute 'update' of 'dict' object is unsafe

joshuaspence, 2021-12-06:
## What happened
After updating to the latest version of Zigbee2MQTT, the following error appears in my Home Assistant logs:

```
ERROR (MainThread) [homeassistant.helpers.template] Error parsing value: access to attribute 'update' of 'dict' object is unsafe. (value: {"brightness":100,"color":{"h":32,"hue":32,"s":81,"saturation":81,"x":0.4574,"y":0.41},"color_mode":"color_temp","color_temp":366,"last_seen":"2021-12-04T04:54:08.021Z","linkquality":87,"power_on_behavior":"previous","state":"OFF","update_available":null}, template: {{ value_json['update']['state'] == "available" }})
```

It seems to have been caused by #9306.

## What did you expect to happen
No error logs.

## How to reproduce it (minimal and precise)
I'm not exactly sure, but happy to dig a bit more if needed.

## Debug info
Zigbee2MQTT version: 1.22.1-1

Koenkk, 2021-12-07:
Duplicate of #9939 and should already be fixed in the dev branch.

closed 2021-12-07 by Koenkk

joshuaspence, 2021-12-07:
Thanks. I didnt look at that other one as the title seemed unrelated