Koenkk/zigbee2mqtt #10058: New Tuya device - Water/Gas Valve _TYZB01_4tlksk8a

VoyteckPL, 2021-12-07:
Clearly indicate what you tried and where you got stuck.
-->
This device is recognised as this:

https://www.zigbee2mqtt.io/devices/TS0001.html#tuya-ts0001

The other problem with this device is when the power is cut off and back on the device turns off automatically. There is "power_on_behavior" missing. Maybe it could be added ?

## Information about the device + link

https://pl.aliexpress.com/item/1005002633228786.html?spm=a2g0s.9042311.0.0.27425c0fUPGr6d

## data/database.db entry of the device

{"id":19,"type":"Router","ieeeAddr":"0x00124b000fd8b36b","nwkAddr":7439,"manufId":0,"manufName":"_TYZB01_4tlksk8a","modelId":"TS0001","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,3,6],"outClusterList":[3,6,4],"clusters":{"genBasic":{"attributes":{"modelId":"TS0001","manufacturerName":"_TYZB01_4tlksk8a","zclVersion":3,"appVersion":32,"hwVersion":0,"dateCode":"20060831        "}},"genOnOff":{"attributes":{"onOff":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00212effff05270b","endpointID":1}],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":32,"hwVersion":0,"dateCode":"20060831        ","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1638818542059,"useImplicitCheckin":true}

VoyteckPL, 2021-12-17:
Any chance on update with this one ?

VoyteckPL, 2021-12-29:
@Koenkk - any chance on fixing this one please?

Koenkk, 2021-12-29:
Fixed.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2021-12-29 by Koenkk

VoyteckPL, 2021-12-30:
@Koenkk Thanks a lot. Is POWER ON BEHAVIOR possible to add somehow?

Koenkk, 2021-12-30:
@VoyteckPL for that you have to check if it is possible using the original gateway, if it is follow: https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html#requirements-and-caveats

VoyteckPL, 2021-12-30:
@Koenkk - I don't have original gateway so no chance? :)

Koenkk, 2021-12-30:
No, I need to know what to send

VoyteckPL, 2021-12-30:
@Koenkk - what if I "borrow" this Tuya Gateway 
[https://allegro.pl/oferta/centralka-bramka-zigbee-wifi-tuya-smart-life-9054486387?utm_feed=aa34192d-eee2-4419-9a9a-de66b9dfae24&utm_content=supercena&utm_source=google&utm_medium=cpc&utm_campaign=_dio_pla_ss_catch-all&ev_adgr=dio_catch-all&ev_campaign_id=15562800182&gclid=CjwKCAiAzrWOBhBjEiwAq85QZ9S3HFgwNAXHSyQo2DD7hPY901zpZkxZDByajGi6f-pTn_2Xd5dvlhoCy8YQAvD_BwE](url)

And do the operations from the manual you provided?
Will that work?

Koenkk, 2021-12-30:
Found a possible way, let me know if it works (you should see power_on_behavior in the frontend under device -> exposes)

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

VoyteckPL, 2021-12-30:
@Koenkk - fantastic!

VoyteckPL, 2021-12-30:
@Koenkk I think there is a problem:

`Zigbee2MQTT:error 2021-12-30 12:17:15: Publish 'set' 'power_on_behavior' to 'Garaż Zawór' failed: 'Error: Write 0x00124b000fd8b36b/1 genOnOff({"moesStartUpOnOff":2}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
Zigbee2MQTT:error 2021-12-30 12:18:06: Publish 'set' 'power_on_behavior' to 'Garaż Zawór' failed: 'Error: Write 0x00124b000fd8b36b/1 genOnOff({"moesStartUpOnOff":2}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
Zigbee2MQTT:error 2021-12-30 12:18:19: Publish 'set' 'power_on_behavior' to 'Garaż Zawór' failed: 'Error: Write 0x00124b000fd8b36b/1 genOnOff({"moesStartUpOnOff":2}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
Zigbee2MQTT:error 2021-12-30 12:18:27: Publish 'set' 'power_on_behavior' to 'Garaż Zawór' failed: 'Error: Write 0x00124b000fd8b36b/1 genOnOff({"moesStartUpOnOff":2}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
Zigbee2MQTT:error 2021-12-30 12:18:40: Publish 'set' 'power_on_behavior' to 'Garaż Zawór' failed: 'Error: Write 0x00124b000fd8b36b/1 genOnOff({"moesStartUpOnOff":2}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
Zigbee2MQTT:error 2021-12-30 12:20:39: Publish 'set' 'power_on_behavior' to 'Garaż Zawór' failed: 'Error: Write 0x00124b000fd8b36b/1 genOnOff({"moesStartUpOnOff":1}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'`

VoyteckPL, 2021-12-30:
[image]
I can see the options but it doesn't save the state.

Koenkk, 2021-12-30:
Then it is not supported via the standard TuYa way, the only way to find out if this is possible is by using the original tuya gateway.

VoyteckPL, 2021-12-30:
I will borrow this gateway (is it standard tuya gateway?) and seek for the codes.

VoyteckPL, 2022-01-01:
@Koenkk I got a video from seller from aliexpress and there are options - relay status. I guess I need tuya gateway to get codes anyway? 
[image]

Koenkk, 2022-01-01:
> I guess I need tuya gateway to get codes anyway?

yes

NewFolk, 2022-01-12:
> I will borrow this gateway (is it standard tuya gateway?) and seek for the codes.
 Hi, **VoyteckPL**
How are you doing with the recognition of the codes?

VoyteckPL, 2022-01-12:
I don't have it yet. I will try to organise it this week and hopefully it will work.

VoyteckPL, 2022-01-12:
I will have it this week - confirmed.

VoyteckPL, 2022-01-13:
@Koenkk I already have original tuya gateway.
Not sure why but I don't see setting posted by aliexpress seller. There is only timer in Tuya app. However, sending the screenshot made by your guide.

[image]

VoyteckPL, 2022-01-13:
@Koenkk another interesting thing is that I found date codes for different device:

QS-Zigbee-D02-TRIAC-2C-LN
https://www.zigbee2mqtt.io/devices/QS-Zigbee-D02-TRIAC-2C-LN.html#lonsonho-qs-zigbee-d02-triac-2c-ln

I found an option to set minimum and maximum brightness (brightness range for a specific bulb).
For light 1:

[image]

For light 2

[image]

And some other (maybe useful?) data:

[image]

NewFolk, 2022-01-13:
@VoyteckPL Do you see same setting in tuya app as alliexpress seller? May be you should update something :(
I think that you have different to seller valve version. Mine is _TZ3000_iedbgyxt.