Koenkk/zigbee2mqtt #10059: Datek HLU2909K reports wrong voltage, current and power

large, 2021-12-07:
## What happened
After joining the HLU2929K ( https://www.zigbee2mqtt.io/devices/HLU2909K.html ) to the network, none of the reporting endpoints where setup.

After reading the Clusters for this device I ended up with this reporting type:
[image]

Afterwards the readings for this unit is 0,01 decimals wrong:
[image]

## What did you expect to happen
On this example a single led-bulb is connected, this is was I expect of values
Voltage = 23104V -> 231,04V
Current = 22A -> 0,22A 
Power = 5W -> 0,05W

I also expected this device to have a energy meter of kWh, is that supported?

## How to reproduce it (minimal and precise)
Add a HLU2909K to your z2m system and check if reporting is set or not.
Add reports and see if the decimal error is still there

## Debug info
Zigbee2MQTT version: 1.18.1 commit: 1.18.1
Adapter hardware: CC2531
Adapter firmware version: 20211115, zStack12
Frontend version: 0.3.76

Koenkk, 2021-12-07:
Before reporting an issue, make sure to use the latest version first. 
If you are still on 1.18.1 check first point of https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.19.0

large, 2021-12-08:
Thank you for the feedback @Koenkk. I never saw the update because of the repo was old :)
After upgrading to 1.19.0 the CC2531 crashed & burned, never recovered.
Overwhelmed by all the data it crashes all the time.

I'll get back to you when I receive the zzh CC2652R and see if the issue remains.

large, 2021-12-09:
Problem seems fixed in 1.19.0, now the Datek needs is a kWh counter

closed 2021-12-09 by large