Koenkk/zigbee2mqtt #10062: No endpoint for the OTA update available for the Aqara Bulb ZNLDP12LM

robertugo2, 2021-12-07:
## What happened

There are no endpoints available for Aqara Bulb [ZNLDP12LM](https://www.zigbee2mqtt.io/devices/ZNLDP12LM.html#xiaomi-znldp12lm). Thus that OTA update is not working as well. In the "Clusters" view, there are no endpoints:
[image]

Repairing or z2m is not working. Reconfigure button is not working causing error, that device cannot be reconfigured.

I have many of those bulbs and I have a problem only with this one. Rest of functionality (on/off control, brightness, color temp) is working fine.

Repairing logs are as follows:

```
Received Zigbee message from 'light_lazienki', type 'attributeReport', cluster 'genLevelCtrl', data '{"61440":3591680,"currentLevel":255}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/light_lazienki', payload '{"brightness":255,"color_mode":"xy","color_temp":153,"linkquality":7,"power_outage_memory":null,"state":"OFF","update_available":null}'
Debug Device 'light_lazienki' announced itself
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"light_lazienki","ieee_address":"0x00158d0003e7f557"},"type":"device_announce"}'
Debug Received Zigbee message from 'light_lazienki', type 'attributeReport', cluster 'genBasic', data '{"dateCode":"11-22-2018"}' from endpoint 1 with groupID 0
Debug Received Zigbee message from 'light_lazienki', type 'read', cluster 'genTime', data '["time"]' from endpoint 1 with groupID 0
Debug Received Zigbee message from 'light_lazienki', type 'attributeReport', cluster 'genLevelCtrl', data '{"61440":3591680,"currentLevel":255}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/light_lazienki', payload '{"brightness":255,"color_mode":"xy","color_temp":153,"linkquality":2,"power_outage_memory":null,"state":"OFF","update_available":null}'
Debug Device 'light_lazienki' announced itself
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"light_lazienki","ieee_address":"0x00158d0003e7f557"},"type":"device_announce"}'
Debug Received Zigbee message from 'light_lazienki', type 'attributeReport', cluster 'genBasic', data '{"dateCode":"11-22-2018"}' from endpoint 1 with groupID 0
Debug Received Zigbee message from 'light_lazienki', type 'read', cluster 'genTime', data '["time"]' from endpoint 1 with groupID 0
Debug Received Zigbee message from 'light_lazienki', type 'attributeReport', cluster 'genLevelCtrl', data '{"61440":3591680,"currentLevel":255}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/light_lazienki', payload '{"brightness":255,"color_mode":"xy","color_temp":153,"linkquality":13,"power_outage_memory":null,"state":"OFF","update_available":null}'
Debug Device 'light_lazienki' announced itself
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"light_lazienki","ieee_address":"0x00158d0003e7f557"},"type":"device_announce"}'
Debug Received Zigbee message from 'light_lazienki', type 'attributeReport', cluster 'genBasic', data '{"dateCode":"11-22-2018"}' from endpoint 1 with groupID 0
Debug Received Zigbee message from 'light_lazienki', type 'read', cluster 'genTime', data '["time"]' from endpoint 1 with groupID 0
Debug Received Zigbee message from 'light_lazienki', type 'attributeReport', cluster 'genLevelCtrl', data '{"61440":3591680,"currentLevel":255}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/light_lazienki', payload '{"brightness":255,"color_mode":"xy","color_temp":153,"linkquality":13,"power_outage_memory":null,"state":"OFF","update_available":null}'
Debug Device 'light_lazienki' announced itself
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"light_lazienki","ieee_address":"0x00158d0003e7f557"},"type":"device_announce"}'
Debug Received Zigbee message from 'light_lazienki', type 'attributeReport', cluster 'genBasic', data '{"dateCode":"11-22-2018"}' from endpoint 1 with groupID 0
Debug Received Zigbee message from 'light_lazienki', type 'read', cluster 'genTime', data '["time"]' from endpoint 1 with groupID 0
Debug Received Zigbee message from 'light_lazienki', type 'attributeReport', cluster 'genLevelCtrl', data '{"61440":3591680,"currentLevel":255}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/light_lazienki', payload '{"brightness":255,"color_mode":"xy","color_temp":153,"linkquality":13,"power_outage_memory":null,"state":"OFF","update_available":null}'
Debug Device 'light_lazienki' announced itself
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"light_lazienki","ieee_address":"0x00158d0003e7f557"},"type":"device_announce"}'
Debug Received Zigbee message from 'light_lazienki', type 'attributeReport', cluster 'genBasic', data '{"dateCode":"11-22-2018"}' from endpoint 1 with groupID 0
Debug Received Zigbee message from 'light_lazienki', type 'read', cluster 'genTime', data '["time"]' from endpoint 1 with groupID 0
Debug Received Zigbee message from 'light_lazienki', type 'attributeReport', cluster 'genBasic', data '{"65281":{"10":0,"100":1,"101":255,"102":153,"3":38,"5":1155,"7":"0x0000000000000000","8":279,"9":2816}}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/light_lazienki', payload '{"brightness":255,"color_mode":"xy","color_temp":153,"linkquality":23,"power_outage_memory":null,"state":"ON","update_available":null}'
Debug Received Zigbee message from 'light_lazienki', type 'attributeReport', cluster 'genBasic', data '{"65281":{"10":0,"100":1,"101":255,"102":153,"3":38,"5":1155,"7":"0x0000000000000000","8":279,"9":2816}}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/light_lazienki', payload '{"brightness":255,"color_mode":"xy","color_temp":153,"linkquality":21,"power_outage_memory":null,"state":"ON","update_available":null}'
Warning Device 'light_lazienki' left the network
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"light_lazienki","ieee_address":"0x00158d0003e7f557"},"type":"device_leave"}'
Debug Retrieving state of 'light_lazienki' after reconnect
Info Device 'light_lazienki' joined
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"light_lazienki","ieee_address":"0x00158d0003e7f557"},"type":"device_joined"}'
Debug Device 'light_lazienki' announced itself
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"light_lazienki","ieee_address":"0x00158d0003e7f557"},"type":"device_announce"}'
Debug Received Zigbee message from 'light_lazienki', type 'attributeReport', cluster 'genBasic', data '{"appVersion":23,"modelId":"lumi.light.aqcn02"}' from endpoint 1 with groupID 0
Debug Received Zigbee message from 'light_lazienki', type 'read', cluster 'genTime', data '["time"]' from endpoint 1 with groupID 0
Debug Received Zigbee message from 'light_lazienki', type 'attributeReport', cluster 'genBasic', data '{"65281":{"10":5943,"100":1,"101":255,"102":153,"3":38,"5":1156,"7":"0x0000000000000000","8":279,"9":2816}}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/light_lazienki', payload '{"brightness":255,"color_temp":153,"linkquality":42,"power_outage_memory":null,"state":"ON","update_available":null}'
Debug Received Zigbee message from 'light_lazienki', type 'attributeReport', cluster 'genBasic', data '{"65281":{"10":5943,"100":1,"101":255,"102":153,"3":38,"5":1156,"7":"0x0000000000000000","8":279,"9":2816}}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/light_lazienki', payload '{"brightness":255,"color_temp":153,"linkquality":42,"power_outage_memory":null,"state":"ON","update_available":null}'
Debug Retrieving state of 'light_lazienki' after reconnect
Error Failed to read state of 'light_lazienki' after reconnect (Read 0x00158d0003e7f557/1 genOnOff(["onOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205)))
Debug Received Zigbee message from 'light_lazienki', type 'readResponse', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/light_lazienki', payload '{"brightness":255,"color_temp":153,"linkquality":60,"power_outage_memory":null,"state":"ON","update_available":null}'
[... cut ...]
Info MQTT publish: topic 'zigbee2mqtt/light_lazienki', payload '{"brightness":255,"color_mode":"xy","color_temp":153,"linkquality":60,"power_outage_memory":null,"state":"ON","update_available":null}'
```

## What did you expect to happen

OTA update is working

## How to reproduce it (minimal and precise)

I don't know, how to reproduce that.

## Debug info
Zigbee2MQTT version: 1.22.1 commit: 48b790cb
Adapter hardware: [CC2538-CC2592](https://modkam.ru/2019/08/07/koordinator-zigbee-v3/)
Adapter firmware version: zStack30x 20201010

robertugo2, 2022-01-07:
Issue has been fixed by:
* removing device from the network
* restarting z2m
* joining device again to the network
Possibly some cache prevents from endpoints update?

Koenkk, 2022-01-07:
The restart shouldn't be needed anymore in 1.22.2, can this be closed?

robertugo2, 2022-01-07:
I think yes.

closed 2022-01-07 by robertugo2