Koenkk/zigbee2mqtt #10073: unable to make MCCGQ14LM work properly with new firmware

chuck2kill, 2021-12-07:
Hi,

I followed the discussions about MCCGQ14LM from xiaomi, and was waiting for the new firmware.
I flashed my CC2531 as soon as the firmware was online, but I'm not able to make it work.

I can pair it, but it doesn't send anything.
here is my js file:

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
        zigbeeModel: ['lumi.magnet.acn001'],
        model: 'MCCGQ14LM',
        vendor: 'Xiaomi',
        description: 'Aqara E1 door & window contact sensor',
        fromZigbee: [],
        toZigbee: [],
        exposes: [e.contact(), e.battery(), e.battery_voltage()],
};

module.exports = definition;
```

I'm a kind of a newbie with zigbee2mqtt. I guess i do something wrong, but I m not really sure about the configurations i need to apply.

thanks for reading me

Koenkk, 2021-12-07:
Remove the external converter, you don't need it any more with 1.22.1

chuck2kill, 2021-12-08:
Thank you
It was a bit tricky but I reinstalled everything, and it works well now.
Great.

closed 2021-12-08 by chuck2kill