Koenkk/zigbee2mqtt #10074: Sometimes Zigbee devices respond slow

Doxlike, 2021-12-07:
## What happened
Sometimes my Zigbee devices didnt work (light didnt turn on/off) or the devices are slow. It is not related to a specific device. It can affect to all devices but not at the same time.

## What did you expect to happen
That Zigbee devices respond fast on request as they do usually

## How to reproduce it (minimal and precise)
Turn on/off any lamp/switch/plug

## Debug info
Zigbee2MQTT version: 1.22 Dev
Adapter hardware: Sonoff usb zigbee 3.0 stick
Adapter firmware version: 20210708

Hi Folks, iam running into serious problems. The problem started 2 weeks ago (no changes at all) that sometimes devices in my network had a large respond time. I have round about 20 devices in my network 12 routers (plugs) and the rest are sensors and buttons (battery powered tradfri buttons/Aqara Sensors). I found in the logs the following when the issue exists:

https://pastebin.com/g0cQAk7X

can someone guide me a little bit? Its not an issue of a single plug or button it happen to all of them but not everytime so its completely random.

Another Example: (Adapter was tradfriPhotoWall and it completely failed to turn on)
https://pastebin.com/Bnn7YUCZ 

Last but not least, Thanks for this great piece of software!

fegyosz, 2021-12-14:
Hi,
I also bought cheap ikea routers like tradfri plugs to extend coverage.
Yesterday I removed all Ikea routers because it seems those devices stop working after some time, can be few hours or few days. This makes routing for end devices to stop or become very slow. To me it appears they become a black hole for zigbee messages. Unplug replug resolve the issue immediately, for some time.
I'm going to investigate further if this is a generic issue with Ikea things or it is due to inappropriate support

Doxlike, 2021-12-14:
@fegyosz for me it started with the ledvance ac26940 devices in my network. Before I had only some aqara sensors and many Ikea tradfri stuff (plugs and lights). I disabled the routing function on the ledvance ac26940 and it looked good for a short time period but now it's still slow and I have some message drops again.

awalsum, 2021-12-19:
> @fegyosz for me it started with the ledvance ac26940 devices in my network. Before I had only some aqara sensors and many Ikea tradfri stuff (plugs and lights). I disabled the routing function on [...]

Maybe a dumb question, but how do you disable the routing function? I'm having some trouble with a Tuya TS0121_plug and want to see what happens if I disable the routing function.

Doxlike, 2021-12-19:
> Maybe a dumb question, but how do you disable the routing function? I'm having some trouble with a Tuya TS0121_plug and want to see what happens if I disable the routing function.

I dunno if it's right but in the database file I changed the entry from router to enddevice. Since then I had less problems

awalsum, 2021-12-19:
Ok, thanks. I'll have a look.

regevbr, 2022-01-19:
Still an issue I'm afraid

Palthe, 2022-01-24:
Have the same problem. If no device has been triggered for a while, the next trigger (movement, light switch etc) takes some time.
Do not know how to solve this.

closed 2022-03-03 by github-actions[bot]

parkwart, 2022-11-19:
Had the same issue, but after some testing i found out that my ps4 in Standby Mode caused all the trouble. So my advice would be to disable 2.4 ghz wifi devices / entire WiFi network to find a potential culprit.

geforce28, 2024-02-05:
Hi @ all.
Have this issue in the newest Version with Sonoff Dongle...

Have you fixed this and how ?

Changed for testing to original hue bridge with same channel and it works well.
So nothing to do with wifi issues..