Koenkk/zigbee2mqtt #10076: Philips Hue White Ambience E27 - 806lm (Model: 9290024898)

tmaniac, 2021-12-07:
I recently added 2 new devices to my Zigbee network but they where not instantly recognized.
So i added them via the external converters i found and patched together a converter file.

However, some other folks could bump into the same issue so i post this request to add the device to the DB.
(and i like all my devices to be recognized without custom-work.)

Device information (sorry could not find the specific product page from Philips)
[Product page on Otto.de](https://www.otto.de/p/philips-hue-philips-hue-white-ambiance-e27-dopelpack-2x570lm60w-led-leuchtmittel-e27-2-stueck-warmweiss-cct-farbtemperatursteuerung-von-2200-6500-kelvin-warmweiss-bis-tageslichtweiss-1504119973/#variationId=1504119974)

Brand: Philips (or Signify)
Model/Type: 9290024898

I created the following patched together converter:
const extend = require('zigbee-herdsman-converters/lib/extend');

```
const definition = {
    zigbeeModel: ['LTA004'],
    model: '9290024898',
    vendor: 'Signify Netherlands B.V.',
    description: 'Hue white ambiance E27 806lm with Bluetooth',
    extend: extend.light_onoff_brightness_colortemp(),
};

module.exports = definition;
```

And my device DB entry:
```
{"id":36,"type":"Router","ieeeAddr":"0x001788010b9beff3","nwkAddr":53347,"manufId":4107,"manufName":"Signify Netherlands B.V.","powerSource":"Mains (single phase)","modelId":"LTA004","epList":[11,242],"endpoints":{"11":{"profId":260,"epId":11,"devId":268,"inClusterList":[0,3,4,5,6,8,4096,64515,768],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"LTA004","manufacturerName":"Signify Netherlands B.V.","powerSource":1,"zclVersion":2,"appVersion":2,"stackVersion":1,"hwVersion":1,"dateCode":"20210422","swBuildId":"1.88.2"}},"lightingColorCtrl":{"attributes":{"colorCapabilities":16,"colorTempPhysicalMin":153,"colorTempPhysicalMax":454}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":1,"hwVersion":1,"dateCode":"20210422","swBuildId":"1.88.2","zclVersion":2,"interviewCompleted":true,"meta":{"configured":88764544},"lastSeen":1634931568263}
```

Koenkk, 2021-12-07:
Could you provide the color temp range? https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_3-adding-converter-s-for-your-device

tmaniac, 2021-12-07:
Just executed the command:

`Read result of 'lightingColorCtrl': {"colorTempPhysicalMin":153,"colorTempPhysicalMax":454}`

Koenkk, 2021-12-07:
`LTA004` is already supported in z2m 1.22.1 (https://github.com/Koenkk/zigbee-herdsman-converters/blob/f176bd5524027397705ed71cc0f7305c6505114a/devices/philips.js#L586)

closed 2021-12-07 by Koenkk

tmaniac, 2021-12-07:
you are right, however; the model 9290024898 is not recognized without the converter....