Koenkk/zigbee2mqtt #10078: TS011F not supported

outlying, 2021-12-07:
Hello there,

first thing first, I'm running `Zigbee2mqtt Edge` I installed it a few hours ago in hope that maybe it will solve my problem (I've seen some other issues before creating this one)

I went through this page
https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html

But I'm not getting those "device unsupported" log messages, I don't see the manufacturer so I'm not sure about all those steps in the tutorial, it seems not up to date

I only got this:

```
Device '0x84fd27fffece9903' joined
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x84fd27fffece9903","ieee_address":"0x84fd27fffece9903"},"type":"device_joined"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x84fd27fffece9903"},"type":"device_connected"}'
```

But about tab of the device clearly states that the device is not supported:

[image]

## Information about the device + link
Device is power Plug, Blitzwolf model BW-SHP15

Link to Polish sell service (mostly for pictures):
https://allegro.pl/oferta/inteligentne-gniazdko-zigbee-blitzwolf-3680w-16a-11476042179?utm_source=tdb&utm_medium=afiliacja&utm_content=Skimlinks+PL&tduid=1f548728b5e298285a83ddf2bed21593  

## data/database.db entry of the device
I'm not sure what is it, I will gladly provide it if you could explain

outlying, 2021-12-08:
After another reset and moving out of dev version I got proper logs

```
Warning Device '0x84fd27fffece9903' with Zigbee model 'TS011F' and manufacturer name '_TZ3000_mraovvmm' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
```

Koenkk, 2021-12-08:
`_TZ3000_mraovvmm` is already supported in z2m 1.22.1

outlying, 2021-12-08:
Annoying, it wasn't working before (few tries done) and now it does :-/ well case closed

closed 2021-12-08 by outlying