Koenkk/zigbee2mqtt #10086: Error: Cannot read property 'settings' of undefined

psobiech, 2021-12-08:
## What happened
There are periodic errors in the console, no groups are listed in the frontend. 
Devices work as normal, already bound groups seem to be working. 

It might be an issue with my configuration, but it seems to follow the configuration guide.
I moved groups to a separate file, but the issue persists.

## What did you expect to happen
Groups working, visible on the frontend, no errors in console ;-)

## How to reproduce it (minimal and precise)
No idea, the config was working, maybe its an issue with my configuration.

## Debug info
Zigbee2MQTT version 1.22.1 (commit #c2b5229) - latest docker version
Adapter hardware: slae.sh / CC2652RB
Adapter firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20210708,"transportrev":2},"type":"zStack3x0"}'

```
Error: Cannot read property 'settings' of undefined
Debug TypeError: Cannot read property 'settings' of undefined at /app/lib/extension/groups.ts:126:26 at Array.filter (<anonymous>) at Groups.onStateChange (/app/lib/extension/groups.ts:125:49) at EventEmitter.emit (events.js:400:28) at EventBus.emitStateChange (/app/lib/eventBus.ts:144:22) at State.set (/app/lib/state.ts:86:23) at Controller.publishEntityState (/app/lib/controller.ts:217:37) at publish (/app/lib/extension/receive.ts:150:22) at Receive.onDeviceMessage (/app/lib/extension/receive.ts:170:13)
```

If I console.log(this.zigbee.groups()) i get the following:
```
[
--
    zh: Group { databaseID: 5, groupID: 1, _members: [Set], meta: {} }
  },
  Group {
    zh: Group { databaseID: 78, groupID: 2, _members: [Set], meta: {} }
  },
  Group {
    zh: Group { databaseID: 84, groupID: 3, _members: [Set], meta: {} }
  },
  undefined
]
```

Full configuration: https://pastebin.com/qs72ztTd (--- are separators of files)

Koenkk, 2021-12-08:
Can you provide your data/database.db file here?

psobiech, 2021-12-08:
https://pastebin.com/mgvas1td here it is, I assumed it to be binary file :-)

Koenkk, 2021-12-08:
Stopping z2m, removing all groups with `"groupID":null` and starting z2m should fix it, do you know how to reproduce this issue?

psobiech, 2021-12-08:
Ok, that fixed the issue, thanks!

I have this instance of z2m for quite some time and was updating every now and then, I did not notice which update or change caused the issue, since I only started investigating when I wanted to add another group and it was not working already. 

The only things I did with groups in the meantime is removeAll, then re-add devices once or twice via MQTT, then bind the group with a remote. 
I only had these 3 groups from the beginning of the configuration and was not adding/removing any.

Sorry that I cannot be more specific, I was just not paying attention.. If I am able to reproduce it again, I will add an issue with reproduction steps and reference this issue. 

Thanks again!

closed 2022-01-15 by github-actions[bot]