Koenkk/zigbee2mqtt #10087: Philips 433714 bulb exposes color temperature but shouldn't

bartn, 2021-12-08:
# Bug Report
## What happened
I have several [Philips 433714](https://www.zigbee2mqtt.io/devices/433714.html) lights and via Zigbee2MQTT they expose color temperature (via color_temp and color_temp_startup), however this is uncorrect for these old Philips Hue bulbs. They were one of the first generations bulb and only emit light at a fixed color temperature.
## What did you expect to happen
Bulbs of this type (433714) should not expose color_temp possibilities.
## How to reproduce it (minimal and precise)
Pair 433714 to Zigbee2MQTT.
## Debug Info
Zigbee2MQTT version:  1.21.2-1
Adapter hardware: cc2652P2
Adapter firmware version: x

Koenkk, 2021-12-08:
According to @grasshide it does (https://github.com/Koenkk/zigbee-herdsman-converters/issues/1218)

bartn, 2021-12-08:
Ah curious to hear @grasshide experience. For completeness sake; here's a picture of my 433714 bulb: https://imgur.com/a/lHFO1UM

grasshide, 2021-12-11:
I think @bartn is right. Those lights aren't able to change the color temperature. At the time I submitted #1218 the doku was stating that it is possible to set a startup color temperature. Therefore I thought it should be possible to set it regularly as well.
When I set the request I was very much at the beginning of understanding all of the Zigbee stuff. Eventually I didn't get it to work with node-red, but I wasn't sure if it was my mistake or an HW issue.
I just re-testet and can definitely agree with @bartn that the bulbs aren't capable of changing color.
Sorry for the confusion.

closed 2021-12-14 by Koenkk