Koenkk/zigbee2mqtt #10088: Need support with adding Tuya device: _TYST11_qtbrwrfv

MickazNL, 2021-12-08:
## Information about the device + link
During a black friday sale I bought a couple of Zigbee Smoke detectors. My thaughts were that I'm never the first to use something with Zigbee2MQTT.

Guess I was wrong.

I bought these: [Bol.com](https://www.bol.com/nl/nl/p/alecto-smart-smoke10-smart-zigbee-rookmelder-pushmelding-extra-luid-alarm-85-db/9200000130339846/)

They seem to be Tuya branded so I made an external converter for it following this guide: [Support new Tuya Devices](https://www.zigbee2mqtt.io/advanced/support-new-devices/02_support_new_tuya_devices.html#instructions)

But after step 2 I'm not sure about what to do next. Can someone help me?

## data/database.db entry of the device
```
{
   "id":44,
   "type":"EndDevice",
   "ieeeAddr":"0xec1bbdfffe79bf0d",
   "nwkAddr":48275,
   "manufId":0,
   "manufName":"_TYST11_qtbrwrfv",
   "powerSource":"Battery",
   "modelId":"tbrwrfv",
   "epList":[
      1
   ],
   "endpoints":{
      "1":{
         "profId":260,
         "epId":1,
         "devId":0,
         "inClusterList":[
            0,
            3
         ],
         "outClusterList":[
            3,
            25
         ],
         "clusters":{
            "genBasic":{
               "attributes":{
                  "modelId":"tbrwrfv",
                  "manufacturerName":"_TYST11_qtbrwrfv",
                  "powerSource":3,
                  "zclVersion":3,
                  "appVersion":73,
                  "stackVersion":0,
                  "hwVersion":1,
                  "dateCode":"20180727"
               }
            }
         },
         "binds":{
            
         },
         "configuredReportings":{
            
         },
         "meta":{
            
         }
      }
   },
   "appVersion":73,
   "stackVersion":0,
   "hwVersion":1,
   "dateCode":"20180727",
   "zclVersion":3,
   "interviewCompleted":true,
   "meta":{
      
   },
   "lastSeen":1638976723332,
   "useImplicitCheckin":true
}
```

## Converter
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require("zigbee-herdsman-converters/lib/tuya");

const definition = {
    // Since a lot of Tuya devices use the same modelID, but use different data points
    // it's usually necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'tbrwrfv',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TYST11_qtbrwrfv'
        },
    ],
    model: 'SMART-SMOKE 10',
    vendor: 'Alecto',
    description: 'Alecto Smoke Detector SMART-SMOKE 10',
    supports: 'thermostat, temperature',
    fromZigbee: [
        fz.ignore_basic_report, // Add this if you are getting no converter for 'genBasic'
        fz.tuya_data_point_dump, // This is a debug converter, it will be described in the next part
    ],
    toZigbee: [
        tz.tuya_data_point_test, // Another debug converter
    ],
    onEvent: tuya.setTime, // Add this if you are getting no converter for 'commandSetTimeRequest'
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genBasic']);
    },
    exposes: [
        e.battery(),
        e.smoke(),
        e.battery_low(),
        e.linkquality()
    ],
};

module.exports = definition;
```

## Debug Log
evice '0xec1bbdfffe79bf0d' announced itself
Received Zigbee message from '0xec1bbdfffe79bf0d', type 'commandSetDataResponse', cluster 'manuSpecificTuya', data '{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":1,"fn":0,"status":0,"transid":4}' from endpoint 1 with groupID 0
Received Zigbee message from '0xec1bbdfffe79bf0d', type 'commandSetDataResponse', cluster 'manuSpecificTuya', data '{"data":{"data":[0],"type":"Buffer"},"datatype":5,"dp":11,"fn":0,"status":0,"transid":4}' from endpoint 1 with groupID 0
Received Zigbee message from '0xec1bbdfffe79bf0d', type 'commandSetDataResponse', cluster 'manuSpecificTuya', data '{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":12,"fn":0,"status":0,"transid":4}' from endpoint 1 with groupID 0
Received Zigbee message from '0xec1bbdfffe79bf0d', type 'commandSetDataResponse', cluster 'manuSpecificTuya', data '{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":16,"fn":0,"status":0,"transid":4}' from endpoint 1 with groupID 0
Received Zigbee message from '0xec1bbdfffe79bf0d', type 'commandSetDataResponse', cluster 'manuSpecificTuya', data '{"data":{"data":[2],"type":"Buffer"},"datatype":4,"dp":14,"fn":0,"status":0,"transid":4}' from endpoint 1 with groupID 0
Received Zigbee message from '0xec1bbdfffe79bf0d', type 'commandSetDataResponse', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,100],"type":"Buffer"},"datatype":2,"dp":15,"fn":0,"status":0,"transid":4}' from endpoint 1 with groupID 0
Received Zigbee message from '0xec1bbdfffe79bf0d', type 'commandSetDataResponse', cluster 'manuSpecificTuya', data '{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":8,"fn":0,"status":0,"transid":4}' from endpoint 1 with groupID 0

MickazNL, 2021-12-17:
Can someone point me in the right direction?

TweakIT, 2022-01-02:
Try adding  [fz.tuya_smoke] in  fromZigbee: since it's a tuya device. If you have a Tuya hub you could also try the following instructions: https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html#_9-find-your-data-point

Do you also have a copy of your tuya.dump.txt?

MickazNL, 2022-01-04:
Hi,

Thanks for your help.

I've updated the converter (see original post) and added the fz.tuya_smoke entry to the fromZigbee array.
I also changed the modelID to end with "\u0000"

After those changes I now have a tuya.dump.txt file which consists of the following lines:

`
1641300976799 0xec1bbdfffe79bf0d 1b00 08 01 00 01
1641300977040 0xec1bbdfffe79bf0d 1c00 09 04 00 00
1641300980274 0xec1bbdfffe79bf0d 1b00 0b 05 00 00
`

(I paired the device and triggered the alarm)

I don'thave a Tuya hub to follow those instructions. I'm only using Zigbee2MQTT which is run inside a docker container.

MickazNL, 2022-01-04:
There seems to be some different entries in the tuya.dump.txt file now that the device has been online for longer.

Some seem to be duplicate lines:

1641300976799 0xec1bbdfffe79bf0d 1b00 08 01 00 01
1641300977040 0xec1bbdfffe79bf0d 1c00 09 04 00 00
1641300980274 0xec1bbdfffe79bf0d 1b00 0b 05 00 00
1641302740521 0xec1bbdfffe79bf0d 1c00 08 01 00 01
1641302740725 0xec1bbdfffe79bf0d 1d00 09 04 00 00
1641302761827 0xec1bbdfffe79bf0d 1d00 08 01 00 01
1641302762056 0xec1bbdfffe79bf0d 1e00 09 04 00 00
1641302762874 0xec1bbdfffe79bf0d 1f00 09 04 00 01
1641302766000 0xec1bbdfffe79bf0d 1f00 08 01 00 00
1641302825023 0xec1bbdfffe79bf0d 1f00 08 01 00 01
1641302825238 0xec1bbdfffe79bf0d 2000 09 04 00 00
1641302826556 0xec1bbdfffe79bf0d 2100 09 04 00 01
1641302829697 0xec1bbdfffe79bf0d 2100 08 01 00 00

So far it's just reporting linkquality:

[image]

TweakIT, 2022-01-04:
Looking at the data of the dump, I think the way the model works is the same as the Woox smoke detector. Since you said that you paired the device and triggered an alarm (I assume by using the test funtion), I think `dp` `0b` is for the test. That leaves `08` and `09` for the smoke and battery_low. Maybe you can trigger the alarm using a candle or something else so we can determine which is the right on. 

After this I believe you can just adapt the `tuya_woox_smoke` in `fromZigbee.js` and define the dp's in tuya.js.

MickazNL, 2022-01-04:
Thanks for your input, I've changed the parameters. Didn't think about the test button triggering something different than real smoke.

I will have to setup a test environment haha ..

TweakIT, 2022-01-04:
Yeah, it's a bit strange. But perhaps they added it because these detectors can be linked using their hub. According to the manual, the testing 'alarm' only generates 2-3 signal tones. So they could have implemented the SmokeTest dp to specifically trigger it. It would be nice to also know how to trigger the real alarm remotely so they can be linked without their own hub.

Not really a need for a test environment. Just keep your current converter running and trigger the real alarm (without the test button). We should see which `dp` changed in the dump and can adjust the code for it.

TweakIT, 2022-01-19:
I have bought the smoke detector myself. Using a custom converter I can get it to show some data. But I don't know which datapoints are correct. Triggering the test alarm is setting smoke to true, but it is not changing it back to false after the test is over. Unfortunately, I can't find the tuya.dump.txt (or it is not created). Maybe you have more luck when trying my converter.

The best option would be to have a hub to determine the right datapoints. Maybe somebody with a hub could help. 

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require("zigbee-herdsman-converters/lib/tuya");

const tuyaLocaldataPoints = {
// 8 is test, 9 is battery, 1 is real smoke?
    alectoSmoke: 1, //?
    alectoBattery: 9, 
    alectoSmokeTest: 8,
};

const myFz = {
        tuya_alecto_smoke: {
        cluster: 'manuSpecificTuya',
        type: ['commandDataResponse', 'commandDataReport'],
        convert: (model, msg, publish, options, meta) => {
            const dp = msg.data.dp;
            const value = tuya.getDataValue(msg.data.datatype, msg.data.data);
            switch (dp) {
            case tuyaLocaldataPoints.alectoBattery:
                return {battery_low: value === 0};
            case tuyaLocaldataPoints.alectoSmoke:
                return {smoke: value === 0};
            case tuyaLocaldataPoints.alectoSmokeTest:
                return {smoke: value};
            default:
                meta.logger.warn(`zigbee-herdsman-converters:tuya_smoke: Unrecognized DP #${ dp} with data ${JSON.stringify(msg.data)}`);
            }
        },
    },
};

const definition = {
    // Since a lot of Tuya devices use the same modelID, but use different data points
    // it's usually necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            modelID: 'tbrwrfv\u0000',
            manufacturerName: '_TYST11_qtbrwrfv'
        },
    ],
    model: 'SMART-SMOKE10',
    vendor: 'Alecto',
    description: 'Alecto Smoke Detector SMART-SMOKE 10',
    fromZigbee: [
      myFz.tuya_alecto_smoke, fz.ignore_tuya_set_time,
      fz.ignore_basic_report, // Add this if you are getting no converter for 'genBasic'
      fz.tuya_data_point_dump // This is a debug converter, it will be described in the next part
      ],
    toZigbee: [
      tz.tuya_data_point_test, // Another debug converter
    ],
    onEvent: tuya.onEventsetTime,
    meta: {},
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genBasic']);
    },
    exposes: [e.smoke(), e.battery_low()],
};

module.exports = definition;
```

MickazNL, 2022-01-19:
Thanks for your efforts! Will try your converter. Tbh, the whole process doesn't all make sense to me :'), every hour I seemed to get new data points. But I'll see what I can do.

I'll probably try it out in the weekend as I'm currently a bit busy. 

Will comment my findings!

TweakIT, 2022-01-19:
Yeah I am confused by the datapoints as well. I don't know what it could report more than smoke, battery_low and LQI. That is where a hub could help, I believe. But since these smoke detectors are the least ugly with Zigbee, I am determined to make them work. Maybe I will just order a hub from Ali since they are not that expensive (in comparisons with other detectors). It would also be helpful to determine how they can be linked (the manual says there is support).

MickazNL, 2022-01-19:
Yeah I bought 4 of these for the same reason. I'll be happy to support you with some money if that's holding you back.

dicktump, 2022-01-26:
I also have one of these, but I don't get it working. Now, I'm not an expert in this Zigbee stuff, but I have a tech/Linux background. Still it's a bit vague how I can get it working. If there is anything I can do to help let me know :)

TweakIT, 2022-01-26:
I have bought a hub and found some interesting details. The device is sending a lot more information than I expected! It also confirmed that one of my smoke detectors was broken out of the box. Of course it was the one I used for initial testing...

First, the instructions that can be sent to the device:

Code | Type | Values
-- | -- | --
self_checking | Boolean | "{true,false}"
muffling | Boolean | "{true,false}"

Self_checking will set the boolean to true and will trigger the test alarm on the smoke detector. After this the boolean is set to false again (in the app).

Muffling allows to silence the alarm. Which could be useful for a detector near the kitchen. 

Than the values reported by the device itself:

<div class="ant-table-body" style="box-sizing: border-box; margin: 0px; padding: 0px; transition: opacity 0.3s ease 0s; color: rgba(0, 0, 0, 0.85); font-family: -apple-system, &quot;system-ui&quot;, &quot;Segoe UI&quot;, &quot;Helvetica Neue&quot;, Helvetica, Arial, &quot;PingFang SC&quot;, &quot;Hiragino Sans GB&quot;, &quot;Microsoft YaHei&quot;, sans-serif, &quot;Apple Color Emoji&quot;, &quot;Segoe UI Emoji&quot;, &quot;Segoe UI Symbol&quot;; font-size: 14px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; background-color: rgb(255, 255, 255); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;">

Code | Type | Values
-- | -- | --
smoke_sensor_status | Enum | {   "range": [     "alarm",     "normal"   ] }
smoke_sensor_value | Integer | {   "unit": "",   "min": 0,   "max": 1000,   "scale": 1,   "step": 1 }
self_checking | Boolean | "{true,false}"
checking_result | Enum | {   "range": [     "checking",     "check_success",     "check_failure",     "others"   ] }
lifecycle | Boolean | "{true,false}"
battery_state | Enum | {   "range": [     "low",     "middle",     "high"   ] }
battery_percentage | Integer | {   "unit": "%",   "min": 0,   "max": 100,   "scale": 0,   "step": 1 }
muffling | Boolean | "{true,false}"

</div><br class="Apple-interchange-newline">

After this, I checked each datapoint for each value:
```
const tuyaLocaldataPoints = {
// Silence 16
// Smoke detection state 1
// Detected smoke 2
// Auto-detect 8
// Auto-detect result 9
// Fault alarm 11
// Lifecycle 12
// Battery level state 14
// Battery level 15

    alecto_Smoke_state: 1,
    alecto_Smoke_value: 2,
    alecto_Self_checking: 8,
    alecto_Checking_result: 9,
    alecto_Smoke_test: 11,
    alecto_Lifecycle: 12,
    alecto_Battery_state: 14,
    alecto_Battery_percentage: 15,
    alecto_Silence: 16,
};
```
Now I am only stuck on exposing these values in the right way. I didn't find the time yet how I should do that in Zigbee2MQTT. I created a class for the datapoints:

```
const myFz = {
        tuya_alecto_smoke: {
        cluster: 'manuSpecificTuya',
        type: ['commandDataResponse', 'commandDataReport'],
        convert: (model, msg, publish, options, meta) => {
            const dp = msg.data.dp;
            const value = tuya.getDataValue(msg.data.datatype, msg.data.data);
            switch (dp) {
            case tuyaLocaldataPoints.alecto_Smoke_state:
                return {smoke_state: value};
            case tuyaLocaldataPoints.alecto_Smoke_value:
                return {smoke_value: value};
            case tuyaLocaldataPoints.alecto_Self_checking:
                return {self_checking: value};
            case tuyaLocaldataPoints.alecto_Checking_result:
                return {checking_result: value};
            case tuyaLocaldataPoints.alecto_Smoke_test:
                return {smoke_test: value};
            case tuyaLocaldataPoints.alecto_Lifecycle:
                return {lifecycle: value};
            case tuyaLocaldataPoints.alecto_Battery_percentage:
                return {battery: value};
            case tuyaLocaldataPoints.alecto_Battery_state:
                return {battery_state: value};
            case tuyaLocaldataPoints.alecto_Silence:
                return {silence: value};
            default:
                meta.logger.warn(`zigbee-herdsman-converters:tuya_smoke: Unrecognized DP #${ dp} with data ${JSON.stringify(msg.data)}`);
            }
        },
    },
};
```

And a definition for the device:
```
const definition = {
    // Since a lot of Tuya devices use the same modelID, but use different data points
    // it's usually necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            modelID: 'tbrwrfv\u0000',
            manufacturerName: '_TYST11_qtbrwrfv'
        },
    ],
    model: 'SMART-SMOKE10',
    vendor: 'Alecto',
    description: 'Alecto Smoke Detector SMART-SMOKE 10',
    fromZigbee: [
      myFz.tuya_alecto_smoke, fz.ignore_tuya_set_time,
      fz.ignore_basic_report, // Add this if you are getting no converter for 'genBasic'
      fz.tuya_data_point_dump // This is a debug converter, it will be described in the next part
      ],
    toZigbee: [
      tz.tuya_data_point_test, // Another debug converter
    ],
    onEvent: tuya.onEventsetTime,
    meta: {},
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genBasic']);
    },
    exposes: [],
};

module.exports = definition;
```

Looking at the code of other devices, I think it should be something like:
`exposes.enum("smoke_state", ea.STATE, ["alarm", "normal"] `

for the smoke_state value (in exposes:[]). But I don't now exactly how that is supposed to work.

TweakIT, 2022-01-27:
I added:

            case tuyaLocaldataPoints.alecto_Smoke_state:
                return {smoke_state: {0: "alarm", 1: "normal"}[value]};

to tuya_alecto_smoke:

Now the device is reporting smoke:state normal when pairing. But when testing the smoke detector it is reset to Null. @Koenkk could you perhaps help us in creating a working converter for the datapoints I listed?

Koenkk, 2022-01-27:
>  But when testing the smoke detector it is reset to Null.

This is probably because another value is send, try debugging which via:

```
        case tuyaLocaldataPoints.alecto_Smoke_state:
             console.log("ALECTO SMOKE GOT " + value + "\n");
            return {smoke_state: {0: "alarm", 1: "normal"}[value]};
```

TweakIT, 2022-01-28:
> This is probably because another value is send, try debugging which via:

This pointed me in the right direction. I ignored that debug converter didn't create a tuya.dump.txt (because of the path, I saw you fixed this in the dev branch but not yet in the main?). The problem with this is that the server crashes because of it (which I didn't notice earlier), which means the other values are not received because the server is restarting. I removed the tuya debug converters and it solved the problem! The detectors are now reporting values:

[image]

I never received any data on the smoke_sensor_value, neither did I see a value for it when using the hub. So perhaps it is a dummy DP. The only problem I still have is by adding support to silence and test the detectors. My working converter (you can test it @MickazNL, @dicktump ):

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require("zigbee-herdsman-converters/lib/tuya");

const tuyaLocaldataPoints = {
// Silence 16
// Smoke detection state 1
// Detected smoke 2
// Auto-detect 8
// Auto-detect result 9
// Fault alarm 11
// Lifecycle 12
// Battery level state 14
// Battery level 15

    alecto_Smoke_state: 1, //?
    alecto_Smoke_value: 2,
    alecto_Self_checking: 8,
    alecto_Checking_result: 9,
    alecto_Smoke_test: 11,
    alecto_Lifecycle: 12,
    alecto_Battery_state: 14,
    alecto_Battery_percentage: 15,
    alecto_Silence: 16,
};

const myFz = {
        tuya_alecto_smoke: {
        cluster: 'manuSpecificTuya',
        type: ['commandDataResponse', 'commandDataReport'],
        convert: (model, msg, publish, options, meta) => {
            const dp = msg.data.dp;
            const value = tuya.getDataValue(msg.data.datatype, msg.data.data);
            switch (dp) {
            case tuyaLocaldataPoints.alecto_Smoke_state:
                console.log("ALECTO SMOKE STATE " + value + "\n");
                return {"Smoke state": {0: "Alarm", 1: "Normal"}[value]};
            case tuyaLocaldataPoints.alecto_Smoke_value:
                console.log("ALECTO SMOKE VALUE" + value + "\n");
                return {smoke_value: value};
            case tuyaLocaldataPoints.alecto_Self_checking:
                console.log("ALECTO SELF CHECKING " + value + "\n");
                return {self_checking: value};
            case tuyaLocaldataPoints.alecto_Checking_result:
                console.log("ALECTO CHECKING RESULT " + value + "\n");
                return {checking_result: {0: "Checking", 1: "Check success", 2: "Check failure", 3: "Others"}[value]};
            case tuyaLocaldataPoints.alecto_Smoke_test:
                console.log("ALECTO SMOKE TEST " + value + "\n");
                return {smoke_test: value};
            case tuyaLocaldataPoints.alecto_Lifecycle:
                console.log("ALECTO SMOKE LIFECYCLE " + value + "\n");
                return {lifecycle: value};
            case tuyaLocaldataPoints.alecto_Battery_percentage:
                console.log("ALECTO BATTERY % " + value + "\n");
                return {battery: value};
            case tuyaLocaldataPoints.alecto_Battery_state:
[... 102 lines cut ...]
            }
        },
    },
};

const definition = {
    fingerprint: [
        {
            modelID: 'tbrwrfv\u0000',
            manufacturerName: '_TYST11_qtbrwrfv'
        },
    ],
    model: 'SMART-SMOKE10',
    vendor: 'Alecto',
    description: 'Alecto Smoke Detector SMART-SMOKE 10',
    fromZigbee: [
      myFz.tuya_alecto_smoke],
    toZigbee: [myFz.tuya_alecto_smoke],
    meta: {},
    exposes: [exposes.text("Smoke state", ea.STATE, ["Alarm", "Normal"]), exposes.text('battery_state', ea.STATE, ["Low", "Middle", "High"]), exposes.text("checking_result", ea.STATE, ["checking", "Check success", "Check failure", "Others"]), exposes.numeric('smoke_value', ea.STATE), exposes.numeric('battery', ea.STATE), exposes.binary('lifecycle', ea.STATE), exposes.enum('self_checking', ea.STATE_SET, ["True", "False"]), exposes.enum('silence', ea.STATE_SET, ["True", "False"])]};

module.exports = definition;
```

First I used exposes.binary for these values, but this generated an error when changing ea.STATE to ea.STATE_SET. It also resulted in a crash when using binary and adding the converter to toZigbee: []. This version allows Zigbee2MQTT to start, but is not working as intended. When I try to change the value from false to true, I get: `Error Error: Cannot read property 'includes' of undefined` . I'm probably missing something.

TweakIT, 2022-01-29:
Today I tried implementing the silencing function. This didn't work. Although it is shown in the app as an option when it is paired to the hub, the device doesn't do anything with it. I also refreshed the device values in the Tuya dev environment, which confirms the value is send but not stored on the device. 

According to the manual there is a second method to silence the alarm:
[image]
I tried this on multiple devices but none of them started blinking at the indicated time. Therefore I believe this is not a implemented function, the same as for the smoke value DP. 

If someone could validate the converter and check if silencing does anything for them, please let me know. After this I will formally do the implementation for the device so that is supported for everyone. 

Unfortunately the device doesn't seems to support binding. Therefore I am thinking of an automation in HA that will just trigger the test mode on all other devices when one of them is reporting the alarm modus. I don't believe it is possible to implement this directly into Zigbee2MQTT. But if somebody has some ideas about this, I am happy to hear it.

TweakIT, 2022-02-06:
I updated the converter, since it was broken after the latest main release. If somebody could check if it works for them I will finalize the converter.

MickazNL, 2022-02-06:
Hi @TweakIT 

Thanks for putting in all the work to make the device compatible! 

I tried your updated converter but things do seem to be a bit different for me:
[image]

I'll yet have to try a real smoke test to see if the reporting function works, but the checking function did.

If I read the code correctly it's supposed to return either Alarm or Normal right wheres for me it's returning null?

I've only just powered the device on again, I will have to see if it reports the value after a while maybe?

MickazNL, 2022-02-06:
I made some small changes to your code in regards to the use of spaces and capital letters and now it seems to be working for me. I changed "Smoke state" to smoke_state and now more things seem to report :).

Here's the code as I'm using it right now:

```
 const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require("zigbee-herdsman-converters/lib/tuya");

const tuyaLocaldataPoints = {
// Silence 16
// Smoke detection state 1
// Detected smoke 2
// Auto-detect 8
// Auto-detect result 9
// Fault alarm 11
// Lifecycle 12
// Battery level state 14
// Battery level 15

    alecto_Smoke_state: 1, 
    alecto_Smoke_value: 2,
    alecto_Self_checking: 8,
    alecto_Checking_result: 9,
    alecto_Smoke_test: 11,
    alecto_Lifecycle: 12,
    alecto_Battery_state: 14,
    alecto_Battery_percentage: 15,
    alecto_Silence: 16,
};

const myFz = {
        tuya_alecto_smoke: {
        cluster: 'manuSpecificTuya',
        type: ['commandDataResponse', 'commandDataReport'],
        convert: (model, msg, publish, options, meta) => {
            const dpValue = tuya.firstDpValue(msg, meta, 'tuya_alecto_smoke');
            const dp = dpValue.dp;
            const value = tuya.getDataValue(dpValue);
            switch (dp) {
            case tuyaLocaldataPoints.alecto_Smoke_state:
                console.log("ALECTO SMOKE STATE " + value + "\n");
                return {smoke_state: {0: "Alarm", 1: "Normal"}[value]};
            case tuyaLocaldataPoints.alecto_Smoke_value:
                console.log("ALECTO SMOKE VALUE" + value + "\n");
                return {smoke_value: value};
            case tuyaLocaldataPoints.alecto_Self_checking:
                console.log("ALECTO SELF CHECKING " + value + "\n");
                return {self_checking: {true: "True", false: "False"}[value]};
            case tuyaLocaldataPoints.alecto_Checking_result:
                console.log("ALECTO CHECKING RESULT " + value + "\n");
                return {checking_result: {0: "Checking", 1: "Check success", 2: "Check failure", 3: "Others"}[value]};
            case tuyaLocaldataPoints.alecto_Smoke_test:
                console.log("ALECTO SMOKE TEST " + value + "\n");
                return {smoke_test: value};
            case tuyaLocaldataPoints.alecto_Lifecycle:
                console.log("ALECTO SMOKE LIFECYCLE " + value + "\n");
                return {lifecycle: value};
            case tuyaLocaldataPoints.alecto_Battery_percentage:
                console.log("ALECTO BATTERY % " + value + "\n");
                return {battery: value};
            case tuyaLocaldataPoints.alecto_Battery_state:
                console.log("ALECTO BATTERY STATE " + value + "\n");
                return {battery_state: {0: "Low", 1: "Middle", 2: "High"}[value]};
            case tuyaLocaldataPoints.alecto_Silence:
                console.log("ALECTO SILENCE " + value + "\n");
                return {silence: {true: "True", false: "False"}[value]};
            default:
                meta.logger.warn(`zigbee-herdsman-converters:tuya_smoke: Unrecognized DP #${ dp} with data ${JSON.stringify(msg.data)}`);
            }
        },
    },
};

const myTz = {
    tuya_alecto_smoke: {
        key: ['self_checking', 'silence'],
        convertSet: async (entity, key, value, meta) => {
          switch(key) {
            case 'self_checking':
            await tuya.sendDataPointBool(entity, tuyaLocaldataPoints.alecto_Self_checking, value);
            console.log("ALECTO SEND " + value + "\n");
            break;
            case 'silence':
            await tuya.sendDataPointBool(entity, tuyaLocaldataPoints.alecto_Silence, value);
            console.log("ALECTO SEND " + value + "\n");  
            break;
      			default: // Unknown key
				        throw new Error(`zigbee-herdsman-converters:tuya_alecto_smoke: Unhandled key ${key}`);
          }
        },
    },
};
	      
const definition = {
    fingerprint: [
        {
            modelID: 'tbrwrfv\u0000',
            manufacturerName: '_TYST11_qtbrwrfv'
        },
    ],
    model: 'SMART-SMOKE10',
    vendor: 'Alecto',
    description: 'Alecto Smoke Detector SMART-SMOKE 10',
    fromZigbee: [
      myFz.tuya_alecto_smoke],
    toZigbee: [myTz.tuya_alecto_smoke],
    meta: {},
    exposes: [exposes.text("smoke_state", ea.STATE, ["Alarm", "Normal"]), exposes.text('battery_state', ea.STATE, ["Low", "Middle", "High"]), exposes.text("checking_result", ea.STATE, ["checking", "Check success", "Check failure", "Others"]), exposes.numeric('smoke_value', ea.STATE), exposes.numeric('battery', ea.STATE), exposes.binary('lifecycle', ea.STATE), exposes.enum('self_checking', ea.STATE_SET, ["True", "False"]), exposes.enum('silence', ea.STATE_SET, ["True", "False"])]};

module.exports = definition;
```

Here's how it's looking right now:

[image]

TweakIT, 2022-02-06:
> I made some small changes to your code in regards to the use of spaces and capital letters and now it seems to be working for me. I changed "Smoke state" to smoke_state and now more things seem to report :).

Nice! I changed it to "Smoke state" because I preferred how it looked. Since you didn't change battery_state and it's also working now, I think you just had to wait for the first time the device reported these values. Something I believe the device reports when pairing.

MickazNL, 2022-03-05:
I think this device can be added to a new release of zigbee2mqtt so it works out of the box. 👍 

As it's your work @TweakIT do you want to take those steps or should I do that?

TweakIT, 2022-03-07:
Thank you for the reminder. I will add it later this week. I was a bit busy.

Beantwoorden - Aan: Koenkk/zigbee2mqtt ***@***.***>
Datum: zaterdag 5 maart 2022 om 16:01
Aan: Koenkk/zigbee2mqtt ***@***.***>
CC: TweakIT ***@***.***>, Mention ***@***.***>

closed 2022-04-15 by github-actions[bot]

TweakIT, 2022-04-20:
Added the support:

https://github.com/TweakIT/zigbee-herdsman-converters/commit/f909bf43749135d2109bf890e768e6e73926fd2f
https://github.com/TweakIT/zigbee2mqtt.io/commit/0965d0bc0d51c0a479d4a2d767f75a4e357232bd

MickazNL, 2022-06-13:
Hi!

It doesn't seems to be added yet. Is that correct?

TweakIT, 2022-06-13:
Correct. Thank you for reminding. I did a new PR: https://github.com/Koenkk/zigbee-herdsman-converters/pull/4345

If you can test it after it is merged, it would be great. I changed two values to binary, but I am not sure that will work. I can remember some problems with it when I was developing the converter.

TweakIT, 2022-07-04:
@MickazNL The smoke detector is now included in the official release. Can you test it?

MickazNL, 2022-07-05:
Hi @TweakIT 

Nice work, I think it looks good. Icon is also looking great.

Only thing I could think off is giving the smoke_state value an initial state of Normal instead of N/A but that's just my ocd speaking.

TweakIT, 2022-07-05:
Great to hear! 

> Only thing I could think off is giving the smoke_state value an initial state of Normal instead of N/A but that's just my ocd speaking.

I believe this is the default behavior of Zigbee2MQTT if it didn't receive a status yet. I don't believe I can do anything about it. 

@dicktump If you didn't notice, yours should be working now as well :)

dicktump, 2022-07-31:
Mine is defective unfortunately and isn't coming back from repair :( So I bought a ZSDR-850, which is also supported since a month or so :)

wjansenw, 2023-01-18:
strange, just re-paired the devices, and it is showing correct values now. Interesting.