Koenkk/zigbee2mqtt #10089: IKEA Styrbar E2001/E2002

bartplessers, 2021-12-08:
I just joined my IKEA Styrbar remote.
https://www.zigbee2mqtt.io/devices/E2001_E2002.html

When clicking on the device, the action is published 3 times, i.e.:
```
Client (null) received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/my/nxx/pxx/remote/R25/IKEA_Styrbar', ... (70 bytes))
{"action":"on","battery":100,"linkquality":51,"update_available":null}
Client (null) received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/my/nxx/pxx/remote/R25/IKEA_Styrbar', ... (68 bytes))
{"action":"","battery":100,"linkquality":51,"update_available":null}
Client (null) received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/my/nxx/pxx/remote/R25/IKEA_Styrbar/action', ... (2 bytes))
on

Client (null) received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/my/nxx/pxx/remote/R25/IKEA_Styrbar', ... (89 bytes))
{"action":"on","action_group":901,"battery":100,"linkquality":51,"update_available":null}
Client (null) received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/my/nxx/pxx/remote/R25/IKEA_Styrbar', ... (68 bytes))
{"action":"","battery":100,"linkquality":51,"update_available":null}
Client (null) received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/my/nxx/pxx/remote/R25/IKEA_Styrbar/action', ... (2 bytes))
on

Client (null) received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/my/nxx/pxx/remote/R25/IKEA_Styrbar', ... (70 bytes))
{"action":"on","battery":100,"linkquality":48,"update_available":null}
Client (null) received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/my/nxx/pxx/remote/R25/IKEA_Styrbar', ... (68 bytes))
{"action":"","battery":100,"linkquality":48,"update_available":null}
Client (null) received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/my/nxx/pxx/remote/R25/IKEA_Styrbar/action', ... (2 bytes))
on
```

for my other devices, this is i.e.
```
Client (null) received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/my/n99/p99/switch/S08/testschakelaar', ... (71 bytes))
{"action":"single_left","battery":100,"linkquality":108,"voltage":3075}
Client (null) received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/my/n99/p99/switch/S08/testschakelaar', ... (60 bytes))
{"action":"","battery":100,"linkquality":108,"voltage":3075}
Client (null) received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/my/n99/p99/switch/S08/testschakelaar/action', ... (11 bytes))
single_left
```

a small bug?

grtz

debug info:
```
Zigbee2MQTT version: 1.22.1
Adapter hardware: zzh (CC2652R Stick) - Electrolama
Adapter firmware version: 
{"commit":"unknown","coordinator":{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20210708,"transportrev":2},"type":"zStack3x0"},"log_level":"info","network":{"channel":11,"extendedPanID":"0xdddddddddddddddd","panID":6758},"permit_join":true,"version":"1.22.1"}

hardware: RPI4
hass, Zigbee2MQTT + MQTT broker running in docker
```

Koenkk, 2021-12-08:
May already be fixed in the latest dev branch, upgrade to the latest-dev and repair the device afterwards.

bartplessers, 2021-12-09:
Yup. 
Solved in latest dev.
Thanks. Again.

closed 2021-12-09 by Koenkk

Koenkk, 2022-04-22:
@jmgutierrezm expected behavior in your case: https://www.zigbee2mqtt.io/guide/faq/#why-is-the-action-property-always-empty