Koenkk/zigbee2mqtt #10095: Multiple Hue Dimmers not in sync when controlling a group (genGroups needed?)

877dev, 2021-12-08:
## What happened
Hue bulbs are in a group, I then bound 2 x Hue Dimmers to that group one at a time.

[Hue Dimmer v2](https://www.zigbee2mqtt.io/devices/929002398602.html)
[Hue bulbs in group](https://www.zigbee2mqtt.io/devices/929002294203.html)

## What did you expect to happen
Dimmer 1 and 2 should each control the group, i.e. 

Press dimmer 1 ON, lights come on.
Press dimmer 2 OFF, lights go off.

## How to reproduce it (minimal and precise)
Press dimmer 1 ON, lights come on.
Press dimmer 1 OFF, lights go off.
Press dimmer 2 ON, nothing happens.
Press dimmer 2 ON (again), lights come on.
and so on.....

The state of the last command is not sync'd across.....

## Debug info
Zigbee2MQTT version 1.22.0 (commit #0a1fccf)
Zigbee-herdsman (0.13.164)

## Thoughts
Reading about the [application layer](https://www.zigbee2mqtt.io/advanced/zigbee/01_zigbee_network.html#the-application-layer) it mentions group endpoints:
```
There are also some special clusters. The ‘group’ cluster allows the definition of ‘groups’ - each defined by a small integer. If an endpoint implements the group cluster then it can be configured to be part of some number of groups. The device remembers which groups it is a member of. A group can then be treated like a virtual endpoint. Messages sent to a particular group ID are broadcast over the network and all devices that are part of that group will respond to the message. Similarly, a client can be bound to a group rather than another endpoint, so that, for example, a single switch can control a whole group of lights.
```

Should I be binding the dimmers using `genGroups` as well?

[image]

877dev, 2022-01-08:
Anyone?

Kirbo, 2022-02-19:
This seems to be an issue also with one remote:
- Press ON -> expect the light to turn on -> light turns on
- Turn the light off from somewhere else, be it Home Assistant, HomeKit, motion triggered or something
- Press ON -> expect the light to turn on -> nothing happens
- Press ON -> expect the light to turn on -> light turns on
- Press ON -> expect the light to turn on -> light turns off
- Turn the light on from somewhere else, be it Home Assistant, HomeKit, motion triggered or something
- Press ON -> expect the light to turn off -> nothing happens
- Press ON -> expect the light to turn off -> light turns off

I'm not sure whether the remote stores the last state it previously sent, as the ON button in Hue Dimmer Switch v2 is a Toggle ON/OFF, or does Z2M have something to do with this, but the remote doesn't behave as I'd expect it to behave, UNLESS the light is turned on and off ONLY via the remote.

I'd expect this to be the same issue as with using two remotes, since if the remote (or Z2M) stores the last state that was sent while pressing ON, e.g.:

```
Initial state:
- Lights are off
- Remote 1 has the state of "Lights off"
- Remote 2 has the state of "Lights off"

Now press Power On on Remote 1
- On Remote 1 Press Power On -> lights turn on (lights are on)
- Remote 1 has the state of "Lights on"
- Remote 2 has the state of "Lights off"

Now press Power On on Remote 2
- On Remote 2 Press Power On -> nothing happens (lights are on)
- Remote 1 has the state of "Lights on"
- Remote 2 has the state of "Lights on"

- On Remote 1 Press Power On -> lights turn off (lights are off)
- Remote 1 has the state of "Lights on"
- Remote 2 has the state of "Lights off"

Now press Power On on Remote 2
- On Remote 2 Press Power On -> nothing happens (lights are off)
- Remote 1 has the state of "Lights off"
- Remote 2 has the state of "Lights off"

Now press Power On on Remote 1
- On Remote 1 Press Power On -> lights turn on (lights are on)
- Remote 1 has the state of "Lights on"
- Remote 2 has the state of "Lights off"

Now turn off the lights somewhere else (Home Assistant, HomeKit...)
- Lights turn off (lights are off)
- Remote 1 has the state of "Lights on"
- Remote 2 has the state of "Lights off"

Now press Power On on Remote 1
- On Remote 1 Press Power On -> nothing happens (lights are off)
- Remote 1 has the state of "Lights off"
- Remote 2 has the state of "Lights off"

Now press Power On on Remote 1
- On Remote 1 Press Power On -> lights turn on (lights are on)
- Remote 1 has the state of "Lights on"
- Remote 2 has the state of "Lights off"
```
...not completely sure about this though, but this is what I currently think would happen in a scenario as above.

Either way, I personally find it problematic that the Hue Dimmer Switch V2 "Power ON" button sends every time `on_press` event, no matter does it turn the lights on or off. This makes creating automations on Home Assistant a bit more cumbersome, as I need to check the state of the device in the automation every time this event is triggered.

Kirbo, 2022-02-19:
Perhaps not directly related to this issue, but I'll write it down here anyway:
As an user, I'd like to have an option in Zigbee 2 MQTT for Philips Hue Dimmer Switch V2 that would allow me to define whether I want the "Power ON" to behave as merely "Power ON" or to behave as "Toggle Power ON/OFF"

In more details:
- A new checkbox/boolean switch that would be e.g. `Power ON behaviour: "V1 / V2"`

V1:
- Power ON button = Power ON
- UP button = Brightness UP
- DOWN button = Brightness DOWN
- HUE button = Power OFF

V2:
- Power ON button = Toggle Power ON/OFF
- UP button = Brightness UP
- DOWN button = Brightness DOWN
- HUE button = "Scene button"

Kirbo, 2022-03-22:
Nope

closed 2022-05-01 by github-actions[bot]