Koenkk/zigbee2mqtt #10098: Essentials  Premium Radiator Thermostat

BinGeradeOnline, 2021-12-08:
### Discussed in https://github.com/Koenkk/zigbee2mqtt/discussions/10091

<div type='discussions-op-text'>

<sup>Originally posted by **BinGeradeOnline** December  8, 2021</sup>
I have bought this radiator, but when i wanna connect it it just shows me that the device is not supported.
I'm pretty new to all of this and don't really know what to do.

I would be happy for any suggestions.</div>

BinGeradeOnline, 2021-12-08:
also i tried to make something with the device.js but i just find a device.yaml

m0wlheld, 2021-12-12:
The device is supported, i use one by myself and pairing was exactly as described on the device page: https://www.zigbee2mqtt.io/devices/GS361A-H04.html

antst, 2021-12-13:
In my case, it got connected, and seems to be working, but I get "null" for position and for the running_state.

m0wlheld, 2021-12-14:
I agree, at least for position. But- does position reflect the valve‘s position or could it be the window position and is therefor null, because i did not bind any window sensor to the thermostat?

d-two, 2021-12-22:
In my case, it got connected, and seems to be working, but I get "null" for battery.

jufo84, 2022-01-23:
Same here both values shows null

m0wlheld, 2022-01-24:
Additionally, when trying to read for example "running_state" using MQTT command from docs (To read send a message to zigbee2mqtt/FRIENDLY_NAME/get with payload {"running_state": ""}), the Z2M logs show _No converter available for 'running_state' ("")_

m0wlheld, 2022-03-01:
With 1.24.0 battery is now reported in %.

MattWestb, 2022-03-02:
@m0wlheld I also having some Siterwell clones branded from pearl and they dont have valve position and running state implanted in the firmware. I dont knowing if its getting firmware versions that have that imparted but i have never seen one user reporting it getting anything on the documented (false) DPs.

If you is removing the battery and putting it back in is all tuya DP devices MCU sending all current status of the DP its have implanted to its host system.

closed 2022-04-09 by github-actions[bot]