Koenkk/zigbee2mqtt #10099: filtered_attributes publishes attributes with 'null' values

silvanverschuur, 2021-12-08:
## What happened
I'm trying to use the `filtered_attributes` option to remove attributes from the MQTT payload. The filtered attributes are not removed from the payload as described in the documentation but the values are replaced with `null`.

## What did you expect to happen
I would like to have the attributes removed from the JSON payload instead of `null` values.

## How to reproduce it (minimal and precise)

This is the configuration of one of my devices (devices.yaml):
```
'0x60a423fffe61d0f3':
  friendly_name: blitzwolf_bw_shp13_1
  filtered_attributes: ["power_outage_memory", "current", "voltage"]
  debounce: 1
```

MQTT log
```
zigbee2mqtt/blitzwolf_bw_shp13_1
{"current":null,"energy":1.91,"linkquality":69,"power":34,"power_outage_memory":null,"state":"ON","voltage":null}
```
zigbee2mqtt/z2m_router_1 {"led":null,"linkquality":72}

## Debug info
Zigbee2MQTT version: 1.22.1

sjorge, 2021-12-09:
@Koenkk not really sure how this can happen, as we unconditionally delete the attributes listed from the message?

https://github.com/Koenkk/zigbee2mqtt/blob/c2b5229f762358055bce4760c33212b89a04e6d7/lib/controller.ts#L258-L260

I've never seen attributes with `null` value myself, they are either present and set or absent.

Koenkk, 2021-12-09:
Should be fixed now. 
Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

@sjorge the filtered attributes were put back in: https://github.com/Koenkk/zigbee2mqtt/blob/48b790cbf3cb2dada56855cad6c6f0792f051aa2/lib/extension/homeassistant.ts#L1297 , fixed the ordering now.

sjorge, 2021-12-09:
Aah that makes sense, as I don’t run the homeassistent extension I never ran into that.

silvanverschuur, 2022-01-09:
I can confirm it works fine in stable version `v1.22.2`.

closed 2022-01-09 by silvanverschuur