Koenkk/zigbee2mqtt #10107: Failed to start zigbee - getExtAddr  after 6000ms

glasody, 2021-12-09:
## What happened
unable to start zigbee2mqtt in haos 6.6. I was able to get it working last night and pair some devices.
But after changing some settings on the zigbee2mqtt frontend and restarting the addon, it stopped working.
The xiaomi buttons that i paired also stopped working as well.

I can see the error with getExtAddr just keep popping up.
I've tried reflashin the device using TI Flash Programmer 2 and the coordinator firmware linked in the debug info but no success.

HA terminal shows the device mapped to `usb-1a86_USB_Serial-if00-port0 -> ../../ttyUSB0`
Not sure what else I can do, couldnt find anything related to the getExtAddr function error anywhere
please help

## What did you expect to happen
Connecting device to be smooth and keep connection

## How to reproduce it (minimal and precise)
Install HAOS 6.6 in Virtualbox VM
pass through usb device of Vision CC2652 dongle
install mosquito
install zigbee2mqtt
add ZHA integration
configure  serial port: with /dev/serial/by-id address, permit_join: true

## Debug info
Zigbee2MQTT version: 1.22.1-1
Adapter hardware: Vision CC2652
Adapter firmware version: https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_3.x.0/bin/CC2652RB_coordinator_20210708.zip
Other info: running VM on Asustor as5202t NAS

```
2021-12-09T03:40:46.362Z zigbee-herdsman:adapter Failed to validate path: 'Error: spawn udevadm ENOENT'
2021-12-09T03:40:46.366Z zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"baudRate":115200,"rtscts":false,"path":"/dev/serial/by-id/usb-1a86_USB_Serial-if00-port0"},"databasePath":"/config/zigbee2mqtt/database.db","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","backupPath":"/config/zigbee2mqtt/coordinator_backup.json","adapter":{"disableLED":false,"concurrent":null,"delay":null}}'
2021-12-09T03:40:46.368Z zigbee-herdsman:adapter:zStack:znp:log Opening SerialPort with /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0 and {"baudRate":115200,"rtscts":false,"autoOpen":false}
2021-12-09T03:40:46.387Z zigbee-herdsman:adapter:zStack:znp:log Serialport opened
2021-12-09T03:40:46.390Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1}
2021-12-09T03:40:46.392Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32]
2021-12-09T03:40:46.640Z zigbee-herdsman:adapter:zStack:znp:log Writing CC2530/CC2531 skip bootloader payload
2021-12-09T03:40:46.641Z zigbee-herdsman:adapter:zStack:unpi:writer --> buffer [239]
2021-12-09T03:40:47.643Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1}
2021-12-09T03:40:47.646Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32]
2021-12-09T03:40:47.657Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,2,97,1,89,6,61]
2021-12-09T03:40:47.658Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,2,97,1,89,6,61]
2021-12-09T03:40:47.659Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 2 - 3 - 1 - 1 - [89,6] - 61
2021-12-09T03:40:47.660Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - ping - {"capabilities":1625}
2021-12-09T03:40:47.662Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-12-09T03:40:47.662Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1}
2021-12-09T03:40:47.663Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32]
2021-12-09T03:40:53.664Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1}
2021-12-09T03:40:53.665Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32]
2021-12-09T03:40:53.676Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,2,97,1,89,6,61]
2021-12-09T03:40:53.678Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,2,97,1,89,6,61]
2021-12-09T03:40:53.678Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 2 - 3 - 1 - 1 - [89,6] - 61
2021-12-09T03:40:53.679Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - ping - {"capabilities":1625}
2021-12-09T03:40:53.679Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-12-09T03:40:53.680Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - version - {}
2021-12-09T03:40:53.680Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,2,35]
2021-12-09T03:40:59.680Z zigbee-herdsman:adapter:zStack:adapter Failed to get zStack version, assuming 1.2
2021-12-09T03:40:59.681Z zigbee-herdsman:adapter:zStack:adapter Adapter concurrent: 2
2021-12-09T03:40:59.682Z zigbee-herdsman:adapter:zStack:adapter Detected znp version 'zStack12' ({"transportrev":2,"product":0,"majorrel":2,"minorrel":0,"maintrel":0,"revision":""})
2021-12-09T03:40:59.683Z zigbee-herdsman:adapter:zStack:startup beginning znp startup
2021-12-09T03:40:59.684Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - getExtAddr - {}
2021-12-09T03:40:59.685Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,4,37]
Zigbee2MQTT:error 2021-12-09 16:41:05: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2021-12-09 16:41:05: Failed to start zigbee
Zigbee2MQTT:error 2021-12-09 16:41:05: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2021-12-09 16:41:05: Exiting...
Zigbee2MQTT:error 2021-12-09 16:41:05: Error: SRSP - SYS - getExtAddr after 6000ms
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (internal/timers.js:557:17)
    at processTimers (internal/timers.js:500:7)
[... 24 lines cut ...]
2021-12-09T03:41:17.000Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,2,97,1,89,6,61]
2021-12-09T03:41:17.001Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 2 - 3 - 1 - 1 - [89,6] - 61
2021-12-09T03:41:17.001Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - ping - {"capabilities":1625}
2021-12-09T03:41:17.002Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-12-09T03:41:17.002Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - version - {}
2021-12-09T03:41:17.003Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,2,35]
2021-12-09T03:41:23.002Z zigbee-herdsman:adapter:zStack:adapter Failed to get zStack version, assuming 1.2
2021-12-09T03:41:23.003Z zigbee-herdsman:adapter:zStack:adapter Adapter concurrent: 2
2021-12-09T03:41:23.003Z zigbee-herdsman:adapter:zStack:adapter Detected znp version 'zStack12' ({"transportrev":2,"product":0,"majorrel":2,"minorrel":0,"maintrel":0,"revision":""})
2021-12-09T03:41:23.005Z zigbee-herdsman:adapter:zStack:startup beginning znp startup
2021-12-09T03:41:23.006Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - getExtAddr - {}
2021-12-09T03:41:23.006Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,4,37]
Zigbee2MQTT:error 2021-12-09 16:41:29: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2021-12-09 16:41:29: Failed to start zigbee
Zigbee2MQTT:error 2021-12-09 16:41:29: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2021-12-09 16:41:29: Exiting...
Zigbee2MQTT:error 2021-12-09 16:41:29: Error: SRSP - SYS - getExtAddr after 6000ms
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (internal/timers.js:557:17)
    at processTimers (internal/timers.js:500:7)
[16:41:29] INFO: Zigbee Herdsman debug logging enabled
[16:41:29] INFO: Handing over control to Zigbee2mqtt Core ...
> zigbee2mqtt@1.22.1 start
> node index.js
```

Koenkk, 2021-12-09:
> install zigbee2mqtt
add ZHA integration

Either install zigbee2mqtt or ZHA, those cannot be ran at the same time.

glasody, 2021-12-09:
@Koenkk oh wow, all this time I thought i needed both. its working properly now. Thank you!

closed 2021-12-09 by glasody

cmk1989, 2022-05-12:
@Koenkk jesus....I spent 4 days on this one thing... thanks you!!!!!!