Koenkk/zigbee2mqtt #10108: Device skips interview on remove + join unless z2m is restarted

sjorge, 2021-12-09:
## What happened
After a dev update I had to re-add a router as it went missing (sometimes happens, very annoying but not been able to reproduce at will). One of my lumi.sen_ill.mgl01 was connected through the router, so it stopped sending updates.

The usual remove battery, insert battery didn't seem to get it back to reporting it just showed up as a reconnected device. Also, not abnormal, those seem to 'forget' everything if their parent routers go missing.

However they can't be re-paired! z2m always has the standard reconnected device message where it tries to retrieve the state.

1. remove device
2. click button to wake it up -> timeout removing
3. repeat 1-2 a few times until it does somehow succeed!
4. enable joining
5. device join and announce message come in, then z2m tries to retrieve the state
6. notice the device shows up as interviewed without actually having the interview messages (???)

At this point, the device still doesn't work or do anything.

1. force remove device
2. enable joining
3. device join and announce message come in, then z2m tries to retrieve the state
4. notice the device shows up as interviewed without actually having the interview messages (???)

??? it looks like z2m is skipping the interview process still

1. remove device
2. click button to wake it up (a few times until it's gone)
3. restart z2m
4. enable joining
5. device join, announce, start of interview (yay)
6. check device, shows up as interviewing (yay)
7. periodically click button on device, eventually it completes the interview
8. device working and shows up as interviewed

I'm guessing there is somehow still a reference to the device kept in memory that is triggering the skipping of the interview.

## What did you expect to happen

Device (force) remove to remove all references to the device (also looks like at least the /availability topic was not cleared or updates to offline when removing the device).

## How to reproduce it (minimal and precise)
n/a

## Debug info
Zigbee2MQTT version: 1.22.1-dev commit: d0195d0
Adapter hardware: CC1352P-2
Adapter firmware version: 20210708

sjorge, 2021-12-09:
It's sadly repeatable.

1. remove sensor (leaves network)
2. enable joining
3. devices joins but z2m uses cached info and does not interview

```
 Device '0x54ef4410000190fb' joined
info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x54ef4410000190fb","ieee_address":"0x54ef4410000190fb"},"type":"device_joined"}'
info Configuring '0x54ef4410000190fb'
debug Device '0x54ef4410000190fb' announced itself
info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x54ef4410000190fb","ieee_address":"0x54ef4410000190fb"},"type":"device_announce"}'
debug Retrieving state of '0x54ef4410000190fb' after reconnect
```

Koenkk, 2021-12-09:
When a device is removed it will be marked as deleted (https://github.com/Koenkk/zigbee-herdsman/blob/4294218cc398fbdb399f1b19512f5836de0a54df/src/controller/model/device.ts#L728), if it joins again it is undeleted https://github.com/Koenkk/zigbee-herdsman/blob/4294218cc398fbdb399f1b19512f5836de0a54df/src/controller/controller.ts#L501 . This is needed for some TuYa devices, they require pairing twice sometimes: https://github.com/Koenkk/zigbee-herdsman/blob/4294218cc398fbdb399f1b19512f5836de0a54df/src/controller/model/device.ts#L449

I think that some things should indeed be cleared on delete, as most of the times this will also cause the device to factory reset itself:
- `meta`: so that the device is reconfigured again
- ias enrolling

Q: why do you want the device to be re-interviewed?

sjorge, 2021-12-09:
I think so, just reconfiguring it after it rejoined without interviewing didn’t seem to make it functional again.

Although I don’t see a technical reason why just reconfigure would not be enough though ¯\_(ツ)_/¯ 

Same with one of my SONOFF sensors that was also disconnected which I noticed later, that one definitely needs a full cycle.

Koenkk, 2021-12-09:
Is it fixed after adding the following to [`removeFromData`](https://github.com/Koenkk/zigbee-herdsman/blob/4294218cc398fbdb399f1b19512f5836de0a54df/src/controller/model/device.ts#L717):

```js
this._interviewCompleted = false;
this.meta = {}
```

sjorge, 2021-12-09:
I did the testing with one of my spare SONOFF devices:

# before the change
## initial pairing
```
 Device '0x00124b0022fe62df' joined
info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00124b0022fe62df","ieee_address":"0x00124b0022fe62df"},"type":"device_joined"}'
info MQTT publish: topic 'zigbee2mqtt/0x00124b0022fe62df/availability', payload 'online'
info Starting interview of '0x00124b0022fe62df'
info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00124b0022fe62df","ieee_address":"0x00124b0022fe62df","status":"started"},"type":"device_interview"}'
debug Device '0x00124b0022fe62df' announced itself
info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00124b0022fe62df","ieee_address":"0x00124b0022fe62df"},"type":"device_announce"}'
debug Received Zigbee message from '0x00124b0022fe62df', type 'attributeReport', cluster 'msRelativeHumidity', data '{"measuredValue":8763}' from endpoint 1 with groupID 0
debug Received Zigbee message from '0x00124b0022fe62df', type 'attributeReport', cluster 'msTemperatureMeasurement', data '{"measuredValue":2218}' from endpoint 1 with groupID 0
debug Received Zigbee message from '0x00124b0022fe62df', type 'attributeReport', cluster 'msRelativeHumidity', data '{"measuredValue":9684}' from endpoint 1 with groupID 0
debug Received Zigbee message from '0x00124b0022fe62df', type 'attributeReport', cluster 'msTemperatureMeasurement', data '{"measuredValue":2311}' from endpoint 1 with groupID 0
debug Received Zigbee message from '0x00124b0022fe62df', type 'attributeReport', cluster 'msRelativeHumidity', data '{"measuredValue":9556}' from endpoint 1 with groupID 0
debug Received Zigbee message from '0x00124b0022fe62df', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"eWeLink","modelId":"TH01"}' from endpoint 1 with groupID 0
debug Received Zigbee message from '0x00124b0022fe62df', type 'readResponse', cluster 'genBasic', data '{"powerSource":3}' from endpoint 1 with groupID 0
debug Received Zigbee message from '0x00124b0022fe62df', type 'readResponse', cluster 'genBasic', data '{"zclVersion":1}' from endpoint 1 with groupID 0
debug Received Zigbee message from '0x00124b0022fe62df', type 'readResponse', cluster 'genBasic', data '{"appVersion":5}' from endpoint 1 with groupID 0
debug Received Zigbee message from '0x00124b0022fe62df', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
debug Received Zigbee message from '0x00124b0022fe62df', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1}' from endpoint 1 with groupID 0
debug Received Zigbee message from '0x00124b0022fe62df', type 'readResponse', cluster 'genBasic', data '{"dateCode":"20201026"}' from endpoint 1 with groupID 0
debug Received Zigbee message from '0x00124b0022fe62df', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
info Successfully interviewed '0x00124b0022fe62df', device has successfully been paired
info Device '0x00124b0022fe62df' is supported, identified as: SONOFF Temperature and humidity sensor (SNZB-02)
info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Temperature and humidity sensor","exposes":[{"access":1,"description":"Remaining battery in %","name":"battery","property":"battery","type":"numeric","unit":"%","value_max":100,"value_min":0},{"access":1,"description":"Measured temperature value","name":"temperature","property":"temperature","type":"numeric","unit":"°C"},{"access":1,"description":"Measured relative humidity","name":"humidity","property":"humidity","type":"numeric","unit":"%"},{"access":1,"description":"Voltage of the battery in millivolts","name":"voltage","property":"voltage","type":"numeric","unit":"mV"},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"SNZB-02","options":[{"access":2,"description":"Number of digits after decimal point for temperature, takes into effect on next report of device.","name":"temperature_precision","property":"temperature_precision","type":"numeric","value_max":3,"value_min":0},{"access":2,"description":"Calibrates the temperature value (absolute offset), takes into effect on next report of device.","name":"temperature_calibration","property":"temperature_calibration","type":"numeric"},{"access":2,"description":"Number of digits after decimal point for humidity, takes into effect on next report of device.","name":"humidity_precision","property":"humidity_precision","type":"numeric","value_max":3,"value_min":0},{"access":2,"description":"Calibrates the humidity value (absolute offset), takes into effect on next report of device.","name":"humidity_calibration","property":"humidity_calibration","type":"numeric"}],"supports_ota":false,"vendor":"SONOFF"},"friendly_name":"0x00124b0022fe62df","ieee_address":"0x00124b0022fe62df","status":"successful","supported":true},"type":"device_interview"}'
info Configuring '0x00124b0022fe62df'
debug Received Zigbee message from '0x00124b0022fe62df', type 'readResponse', cluster 'genPowerCfg', data '{"batteryVoltage":28}' from endpoint 1 with groupID 0
info MQTT publish: topic 'zigbee2mqtt/0x00124b0022fe62df', payload '{"linkquality":96,"voltage":2800}'
[... cut ...] me":"humidity_calibration","property":"humidity_calibration","type":"numeric"}],"supports_ota":false,"vendor":"SONOFF"},"friendly_name":"0x00124b0022fe62df","ieee_address":"0x00124b0022fe62df","status":"successful","supported":true},"type":"device_interview"}'
error Configure failed: Error: Bind 0x00124b0022fe62df/1 msTemperatureMeasurement from '0x00124b00228120b5/1' failed (AREQ - ZDO - bindRsp after 10000ms)
info Successfully configured '0x00124b0022fe62df'
```

So that seems to do the trick!

Koenkk, 2021-12-10:
looks good, I think some other things need to be cleared to (like bindings and configured reportings), will check it next week since I will go on a short trip without having access to a pc.

sjorge, 2021-12-10:
> looks good, I think some other things need to be cleared to (like bindings and configured reportings), will check it next week since I will go on a short trip without having access to a pc.

Enjoy the trip, no rush. If I need to rejoin another device and it's having issues I know how to deal with it until it gets fixed.

Clearing bindings and reporting might be a good idea too yeah, if they survive on the device, doing them again shouldn't hurt, if not, we need to redo them anyway.

Krystm, 2021-12-14:
I'm seeing the same issue with my new ZZH setup. I have a few devices that just wont join and then out of no where will join but doesn't seem to complete as full exposes aren't working. This may also be related to the routers not allowing / working to join devices. I know this isn't constructive but never had these problems on hubitat /sigh. I'm staying the course just may have had a bad time to try to move over as most everything else is working.

Koenkk, 2021-12-14:
Should be fixed now, @sjorge can you confirm?

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

sjorge, 2021-12-14:
Can I just npm install the new version of herdsman inside the zigbee2mqtt path?

Koenkk, 2021-12-14:
I will update zh in z2m in 10 mins from now.

Krystm, 2021-12-14:
So I upgraded my ZZH to the Dev 112021 version, all the routers fell off, I re-paired them all. Everything seems good to go, there was one device kinda had a hard time coming back online but I brought it a bit closer in the same room (this was a plug / router behind a cabinet), verified it worked and then moved to final place and seems good to go. 

I did however, notice that I manually force deleted the plugs, when I went to re-pair it would show device left network, and now I have one sengled bulb throwing an error, but Ill try removing it and re adding see if the error is there (I have 2 other of exact same bulb no issue). 

2021-12-14 14:46:01: Failed to read state of 'Front Door Light 2' after reconnect (Read 0xb0ce181400170796/1 lightingColorCtrl(["colorMode","currentX","currentY","enhancedCurrentHue","currentSaturation","colorTemperature"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE'))

Koenkk, 2021-12-15:
@sjorge also fixed, can this be closed?

sjorge, 2021-12-15:
Yep!

closed 2021-12-15 by sjorge