Koenkk/zigbee2mqtt #10111: E1525/E1745 Motion sensor occupancy does not go to "false"

atmel14, 2021-12-09:
## What happened
E1525/E1745 Motion sensor is joind sucessfully.
Than an motion occur and get this message:
`Zigbee2MQTT:info  2021-12-09 14:31:51: MQTT publish: topic 'zigbee2mqtt/0x8cf681fffe51e282', payload '{"battery":100,"elapsed":144598,"illuminance_above_threshold":false,"linkquality":92,"occupancy":true}'`
"occupancy":true is set to true.

Wait now without moving, but "occupancy" is not set to "false".
When after some time moving occurs again a new message with "occupancy":true is received

## What did you expect to happen
Without moving in a timeframe "occupancy" goes to "false"

## How to reproduce it (minimal and precise)
Every time

## Debug info
Zigbee2MQTT version: 1.22.1
Adapter hardware: CC2531
Adapter firmware version: 20211115

atmel14, 2022-01-03:
Test with Version 1.22.2 => "occupancy" goes to "false" after 3 min!
How this 3min could be change?

GSzabados, 2022-01-28:
> Wait now without moving, but "occupancy" is not set to "false".
> When after some time moving occurs again a new message with "occupancy":true is received

Have you changed the `occupancy_timeout` under the Settings (specific) tab of the device?

Actually this setting should not be applicable for the IKEA motion sensors. They have a hard codes 3 minutes cool down time for preserving battery. It means that they will not detect any motion for 3 minutes after motion was detected. It is a hard coded value from IKEA and it cannot be changed as far as I know. So the `occupancy_timeout` setting should not be available for these devices.

closed 2022-03-07 by github-actions[bot]