Koenkk/zigbee2mqtt #10113: Slow response on AQARA B1 after hours of inoperation

ChengyiP, 2021-12-09:
## What happened
Aqara B1 would take approx. 30 secs to react on after hours in activity. Example,  if my curtain are left untouched during the day. When i am home at night, it would move approx. 30sec after my trigger. However, after that initial movement, the next movement would be almost instant.
## What did you expect to happen
Faster reaction time. 
## How to reproduce it (minimal and precise)
As long as i left it untouched for couple of hours, this symptoms would occur
## Debug info
Zigbee2MQTT version: 1.22.0-2
Adapter hardware: CC2652
Adapter firmware version:

closed 2022-01-16 by github-actions[bot]

jeroenvdd, 2022-01-20:
do you still have this? cause it is showing the same behavior here

ChengyiP, 2022-01-22:
It seems to have gone away after i publish option to manual open to zigbee2mqtt/FRIENDLY_NAME/set

https://www.zigbee2mqtt.io/devices/ZNCLDJ12LM.html#configuration-of-device-attributes