Koenkk/zigbee2mqtt #10123: Philips 5045131P7 (hue centura) with different modelid

marcelhey, 2021-12-10:
# Bug Report
Just bought the Philips hue Centura spot, which is supported, but after pairing it comes up with this modelid: 929003047501
This modelid isn’t supported yet. Can it be added?

## Debug Info
Zigbee2MQTT version: 1.22.1
Adapter hardware: Conbee2
Adapter firmware version:

amarce, 2021-12-13:
I have the same problem... seems that the new ones are coming with this id.

Klangen82, 2022-01-06:
I have the same issue here:/ Tried to add it manually but I don't get it to work. Would be great if someone could help:)

ivargj, 2022-01-06:
This sounds a bit similar to my issue here: #10106 
What are the symptoms in your cases, is it similar as mine?

Klangen82, 2022-01-06:
> This sounds a bit similar to my issue here: #10106 What are the symptoms in your cases, is it similar as mine?

I don't believe so, the model name have changed to 929003047501.

Koenkk, 2022-01-06:
If you could fill in the new template I can add it: https://github.com/Koenkk/zigbee2mqtt/issues/new?assignees=&labels=new+device+support&template=new_device_support.yaml&title=%5BNew+device+support%5D%3A+

closed 2022-01-06 by Koenkk

Klangen82, 2022-01-06:
> If you could fill in the new template I can add it: https://github.com/Koenkk/zigbee2mqtt/issues/new?assignees=&labels=new+device+support&template=new_device_support.yaml&title=%5BNew+device+support%5D%3A+

https://github.com/Koenkk/zigbee2mqtt/issues/10617