Koenkk/zigbee2mqtt #10128: Variation of TuYa TS0012 2-gang zigbee switch (Cluster not supported)

psobiech, 2021-12-10:
## Information about the device + link
QA-Zigbee-S04-2C-L
https://www.ifreeq.cn/product-page/qs-zigbee-s04-2c-l

## data/database.db entry of the device
```
{
   "id":13,
   "type":"EndDevice",
   "ieeeAddr":"0xa4c138b96c5996eb",
   "nwkAddr":11646,
   "manufId":4417,
   "manufName":"_TZ3000_llfaquvp",
   "powerSource":"Mains (single phase)",
   "modelId":"TS0012",
   "epList":[
      1,
      2
   ],
   "endpoints":{
      "1":{
         "profId":260,
         "epId":1,
         "devId":256,
         "inClusterList":[
            3,
            4,
            5,
            6,
            57344,
            57345,
            0
         ],
         "outClusterList":[
            25,
            10
         ],
         "clusters":{
            "genOnOff":{
               "attributes":{
                  "onOff":0,
                  "onTime":0,
                  "offWaitTime":0,
                  "moesStartUpOnOff":2
               }
            },
            "manuSpecificTuya2":{
               "attributes":{
                  "53251":"AAAAAgAA"
               }
            },
            "manuSpecificTuya_3":{
               "attributes":{
                  "switchType":0
               }
            },
            "genBasic":{
               "attributes":{
                  "modelId":"TS0012",
                  "manufacturerName":"_TZ3000_llfaquvp",
                  "powerSource":3,
                  "zclVersion":3,
                  "appVersion":64,
                  "stackVersion":0,
                  "hwVersion":1,
                  "dateCode":""
               }
            }
         },
         "binds":[
            {
               "cluster":6,
               "type":"endpoint",
               "deviceIeeeAddress":"0x00124b0021cc44c0",
               "endpointID":1
            }
         ],
         "configuredReportings":[
            
         ],
         "meta":{
            
         }
      },
      "2":{
         "profId":260,
         "epId":2,
         "devId":256,
         "inClusterList":[
            4,
            5,
            6,
            57345
         ],
         "outClusterList":[
            
         ],
         "clusters":{
            "genOnOff":{
               "attributes":{
                  "onOff":0,
                  "onTime":0,
                  "offWaitTime":0
               }
            }
         },
         "binds":[
            {
               "cluster":6,
               "type":"endpoint",
               "deviceIeeeAddress":"0x00124b0021cc44c0",
               "endpointID":1
            }
         ],
         "configuredReportings":[
            
         ],
         "meta":{
            
         }
      }
   },
   "appVersion":64,
   "stackVersion":0,
   "hwVersion":1,
   "dateCode":"",
   "zclVersion":3,
   "interviewCompleted":true,
   "meta":{
      "configured":2
   },
   "lastSeen":1639137567400,
   "useImplicitCheckin":true
}
```

The interview process fails, with the following error:
```
Error: Cluster with name '57344' does not exist
    at Object.getCluster (/app/node_modules/zigbee-herdsman/src/zcl/utils.ts:66:19)
    at Endpoint.saveClusterAttributeKeyValue (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:242:35)
    at Controller.onZclOrRawData (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:644:26)
    at ZStackAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:135:70)
    at ZStackAdapter.emit (events.js:400:28)
    at ZStackAdapter.onZnpRecieved (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:796:30)
    at Znp.emit (events.js:400:28)
    at Znp.onUnpiParsed (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:108:18)
    at Parser.emit (events.js:400:28)
    at Parser.parseNext (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/unpi/parser.ts:45:26)
```

when I add to the cluster definition:
    manuSpecificTuya2: {                                                                                  
        ID: 0xE000,                                                                                       
        attributes: {},                                                                                   
        commands: {},
        commandsResponse: {},
    }

and restart the herdsman, the application is successfully identified as https://www.zigbee2mqtt.io/devices/TS0012.html and seems to be working with no issues.

Image of the frontend: https://i.imgur.com/mgSAaNl.png

I do not know what this new cluster does. 
```
            "manuSpecificTuya2":{
               "attributes":{
                  "53251":"AAAAAgAA"
               }
            },
```

Version:
Zigbee2MQTT version 1.22.1 (commit #c2b5229) - latest docker version
Adapter hardware: slae.sh / CC2652RB
Adapter firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20210708,"transportrev":2},"type":"zStack3x0"}'

psobiech, 2021-12-10:
Nevermind, fount this MR https://github.com/Koenkk/zigbee-herdsman/pull/461 which fixes the issue in zigbee-herdsman

closed 2021-12-10 by psobiech

dkwireless, 2021-12-19:
However there are still two issues:
1. It shows wrong product image. Right one is from this link: https://www.ifreeq.cn/product-page/qs-zigbee-s04-2c-l
2. It doesn't expose switch_type property. I have checked and it is available under: 'manuSpecificTuya_3': {"switchType":0}. I can set it and it works as expected.