Koenkk/zigbee2mqtt #10129: GIRIER JR-ZDS01 Mini Breaker-1CH z2mqtt reload after becoming available again

DrSchmurge, 2021-12-10:
## What happened
power cycling the device crashes and restarts z2mqtt with error:
`Error: Cluster with name '57344' does not exist`
the device is named "Girier Реле 1" in my system
availability is swiched to true in config. Restarting z2mqtt sets states of all devices to 'unavailable', triggering related automations. Huge pain :(

## What did you expect to happen
z2mqtt is expected not to crash

## How to reproduce it (minimal and precise)
1) pair device
2) power cycle the device
3) spectate z2mqtt to crash

## Debug info
Zigbee2MQTT version: 1.22.1-1
Adapter hardware: CC2531
Adapter firmware version: 20201127

logs:
```
Zigbee2MQTT:info  2021-12-10 19:59:26: MQTT publish: topic 'zigbee2mqtt/Girier Реле 1', payload '{"last_seen":"2021-12-10T16:59:26.964Z","linkquality":76,"power_on_behavior":"previous","state":"OFF","switch_type":"toggle"}'
Zigbee2MQTT:info  2021-12-10 20:00:58: MQTT publish: topic 'zigbee2mqtt/Girier Реле 1', payload '{"last_seen":"2021-12-10T17:00:58.771Z","linkquality":73,"power_on_behavior":"previous","state":"OFF","switch_type":"toggle"}'
Error: Cluster with name '57344' does not exist
    at Object.getCluster (/app/node_modules/zigbee-herdsman/src/zcl/utils.ts:66:19)
    at Endpoint.saveClusterAttributeKeyValue (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:242:35)
    at Controller.onZclOrRawData (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:644:26)
    at ZStackAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:135:70)
    at ZStackAdapter.emit (events.js:400:28)
    at ZStackAdapter.onZnpRecieved (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:796:30)
    at Znp.emit (events.js:400:28)
    at Znp.onUnpiParsed (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:108:18)
    at Parser.emit (events.js:400:28)
    at Parser.parseNext (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/unpi/parser.ts:45:26)
[20:01:05] INFO: Handing over control to Zigbee2mqtt Core ...
> zigbee2mqtt@1.22.1 start
> node index.js
Zigbee2MQTT:info  2021-12-10 20:01:14: Logging to console and directory: '/config/zigbee2mqtt/log/2021-12-10.20-01-10' filename: log.txt
```

Koenkk, 2021-12-14:
Could you check if this has already been fixed in the dev branch? (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

DrSchmurge, 2021-12-23:
> Could you check if this has already been fixed in the dev branch? (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

I will try to check in several days, thanks. Sorry for long reply, shortage of time now =(

DrSchmurge, 2021-12-26:
> Could you check if this has already been fixed in the dev branch? (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Hi, checked on the dev (edge) branch, the problem seems to be solved there! Thanks!

closed 2021-12-27 by Koenkk