Koenkk/zigbee2mqtt #10131: "No update available" when clearly an update *should be* available

SHxKM, 2021-12-10:
## What happened

I have 3 [Philips 5633030P6](https://www.zigbee2mqtt.io/devices/5633030P6.html#philips-5633030p6) (Pillar) lights. All of them paired with Z2M a while ago. When I first paired them all were on firmware `1.65.11_hB798F2B`. When I went to the OTA tab, 2 of them reported an update is available, and a firmware update to `1.88.2` was successful. However, the other one did not. When I "Check for new updates", it always reports "No updates are available". 

See image 1:

[image]

I have also noticed its "power-source" indicator is a "?" instead of "". See image 2:

[image]

## What did you expect to happen

For a device with exact same model as other devices, to report an update needed if it's on an earlier firmware version.

## How to reproduce it (minimal and precise)

In this case the "What happened" section should be sufficient.

## Debug info
Zigbee2MQTT version: 1.22.1-1
Adapter hardware: Conbee II

irekzielinski, 2021-12-12:
Maybe related/maybe not.
I have 2 Philips Hue bulbs. Both where out of date so I triggered OTA via zigbee2mqtt. One completed OK and it's shown as having version 1.88.1.
Second one got stuck at 99% and zigbee2mqtt still shows an old version. When I check for the update - it find no update.

But then when I connected this build to hue hub it was showing that it's already on version 1.88.1. So it looks like zigbee2mqtt has an invalid/old firmare information for this one build. I tried re-pairing and it still think it's on old firmware where in fact it's already up to date.

[image]

SHxKM, 2021-12-12:
Very interesting @irekzielinski. Thanks. 

Is there any way we can get the raw firmware directly from the device? If so, what @irekzielinski can be validated and maybe some logic can be implemented to “sync” between the values.

irekzielinski, 2021-12-12:
> Is there any way we can get the raw firmware directly from the device? 
If you have a Philips Hue Hub - pair one of the builbs to it temporarly. That would allow you to verify if firmware is up-to-date or not. But maybe indeed there is a way to do it in zigbee2mqtt.

SHxKM, 2021-12-12:
> > Is there any way we can get the raw firmware directly from the device?
> 
> If you have a Philips Hue Hub - pair one of the builbs to it temporarly. That would allow you to verify if firmware is up-to-date or not. But maybe indeed there is a way to do it in zigbee2mqtt.

I was referring to such option *without* the need to pair to another bridge, so we can maybe write a fix for Z2M to identify it has the wrong firmware showing.

Koenkk, 2021-12-19:
Maybe the reading of the new software version failed after updating, you can retrieve it again by:
- remove bulb
- stop z2m
- start z2m
- pair bulb

SHxKM, 2021-12-19:
> Maybe the reading of the new software version failed after updating, you can retrieve it again by:
> 
> * remove bulb
> * stop z2m
> * start z2m
> * pair bulb

Is there any way to do this[^1] without going through a re-pair process? I have some adjacent lights and things can get messy..

[^1]: maybe you have a pointer to the responsible code?

Koenkk, 2021-12-19:
A read in the frontend might also do it:

[image]

SHxKM, 2021-12-19:
@Koenkk - very interesting, when I do that, I get:

```
Read result of 'genBasic': {"swBuildId":"1.88.2"}
```

Which **_is_** the most up-to-date version. But the frontend does not seem to update the value anywhere, not on the device page nor on the OTA page. Isn't it possible to "refresh" this value when its read from the cluster like this?

Koenkk, 2021-12-19:
@SHxKM added an improvement to zigbee-herdsman to update these properties when read. Regarding the `?` a reread of `powerSource` should also fix it (you might need to restart z2m afterwards).

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

SHxKM, 2021-12-19:
> Regarding the ? a reread of powerSource should also fix it (you might need to restart z2m afterwards).

That worked perfectly. Had to restart Z2M as you said. Thanks.

> @SHxKM added an improvement to zigbee-herdsman to update these properties when read. 

Awesome, thank you. I'm a bit wary of going the dev branch route (already waiting for #10120) simply because I don't know how can I do that in a safe, reversible way while on HassOS.

SHxKM, 2021-12-19:
@Koenkk  Now I'm wondering about something else. I have 3 units of this, and the "problematic" one has an entry in "Bind" which I have never seen before on any of my router devices. I also cannot remove this binding. See pics.

[image]

[image]

Koenkk, 2021-12-20:
> Awesome, thank you. I'm a bit wary of going the dev branch route (already waiting for #10120) simply because I don't know how can I do that in a safe, reversible way while on HassOS.

Just uninstall the normal addon and install the edge should do it (similarly you can switch back)

Regarding the bind, it's unclear to me where this comes from, I cannot remember any place in the z2m which does this.

SHxKM, 2021-12-20:
> Just uninstall the normal addon and install the edge should do it (similarly you can switch back)

I thought I saw something about backups needed for config files because they are not auto-restored/are override, or something along those lines..thanks again!

SHxKM, 2022-01-01:
@Koenkk After the update to Z2M `v1.22.2` almost all my devices have the red "update device firmware", where previously they didn't. I don't know if it's a coincidence or not. Before the update there was no indication that these devices needed updating, neither in the UI nor in their "State" tab.

fisch55, 2022-01-01:
my hue goe bulbs  (Z2M v1.22.2) shows me updates...but ends in "invalid" image...

[
[image]
](url)

SHxKM, 2022-01-01:
Yep, mine is stuck. This seems to be some kind of regression introduced with 1.22.2. I’m not sure if it has to do with the change mentioned by @Koenkk some replies above.

fisch55, 2022-01-01:
I had it with 1.22.1 , too!

SHxKM, 2022-01-01:
> I had it with 1.22.1 , too!
> This FW of hue go are old, so that updates defiantly available.

Are you saying `1.88.1/2` is old? 

Just so we don’t conflate two issues together: this issue is about one specific light’s firmware not getting reflected or updated in Z2M. 

By coincidence or not, after updating to 1.22.2, I’ve seen 99% of my lights (previously there was only one or two) report through the UI that an update is available. So I’m not sure what’s going on: either with the previous version of Z2M most devices weren’t correctly reported, or that is the situation now.

Koenkk, 2022-01-01:
@SHxKM yesterday new hue firmwares were added: https://github.com/Koenkk/zigbee-OTA/commit/5acb22ab5e123ff678ed007f91e44a83bbba1744

fisch55, 2022-01-01:
ok, sorry -for mixed two issues....

no i ment, the fw of my bulbs ist old...

SHxKM, 2022-01-01:
> @SHxKM yesterday new hue firmwares were added: [Koenkk/zigbee-OTA@5acb22a](https://github.com/Koenkk/zigbee-OTA/commit/5acb22ab5e123ff678ed007f91e44a83bbba1744)

Thanks, that explains it.

I'm gonna mark this issue as resolved for now.

"Unfortunately", since this update coincided with a Philips firmware update, I could not really test the fix. All my devices were showing an update needed, including the problematic device. When I went to update it, of course it updated, because there *was* a new firmware now available for it, and for all other devices. 

@fisch55 - I saw you opened a new issue, I hope you get it sorted out soon!

SHxKM, 2022-01-02:
Gonna have to reopen this as there seems to be another issue now. 

1. For a _different device_ which I updated from `1.88.1` to `1.93.3`, I noticed the firmware in the UI stayed at `1.88.1`. I then tried to read `swBuildId`, and it showed `1.93.3`. But **the UI did not update**. I had to restart Z2M for the new firmware to show up.

2. For devices which are on `1.93.3`, I get this error this error when I click on "Check for new updates":

> Failed to check if update available for 'outdb_pantry_window_impress_l' (No image available for imageType '285')

I'm gonna open a different issue about the second one in order to see if more people are experiencing this.

Koenkk, 2022-01-02:
> Regarding the wrong firmware version displayed, as I said it appears that the UI does not update accordingly for some devices, and a restart is required in order for the new version to be reflected.

The rereading should only be needed when you updated the bulb pre z2 1.22.2, since I believe updating with 1.22.2 will give you the correct version immediately. Assuming this can be closed now.

SHxKM, 2022-01-02:
> The rereading should only be needed when you updated the bulb pre z2 1.22.2, since I believe updating with 1.22.2 will give you the correct version immediately. Assuming this can be closed now.

Actually, this was while I’m on 1.22.2. And with a different device, too. It updated Philips to 1.93.3, the UI continued showing 1.88.1 until both a rereading and a restart occurred. Maybe I’m missing something?

Koenkk, 2022-01-02:
@SHxKM interesting, do you still have bulbs left to update? If yes please provide me the debug log when doing that.

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

SHxKM, 2022-01-02:
> @SHxKM interesting, do you still have bulbs left to update? If yes please provide me the debug log when doing that.
> 
> See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

Unfortunately not. Thanks anyway!

SHxKM, 2022-01-04:
> @SHxKM interesting, do you still have bulbs left to update? If yes please provide me the debug log when doing that.
> 
> See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

I think I just found at least two more cases, when exactly do you want me to turn on debugging? While updating? (that's already done) While reading the cluster?

SHxKM, 2022-01-04:
@Koenkk Here's an example.  Can this be reopened?

The update for this device clearly succeeded (the newer `1.93.3` firmware requests an imageType currently not supported by Z2M), but Z2M shows the old version (even after reading the cluster manually). I bet (or hope) now if I restart Z2M the new version will show. I'm not doing so I can help debug it.

[image]

[image]

Koenkk, 2022-01-04:
@SHxKM do you have the debug log for me? https://github.com/Koenkk/zigbee2mqtt/issues/10131#issuecomment-1003768735

SHxKM, 2022-01-04:
> @SHxKM do you have the debug log for me? [#10131 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/10131#issuecomment-1003768735)

If you mean *during* and after the update, then no. Sorry.

Koenkk, 2022-01-04:
@SHxKM yes during indeed, that is a pitty, I will need a debug log from starting the update till finish to further work on this.

SHxKM, 2022-01-04:
> @SHxKM yes during indeed, that is a pitty, I will need a debug log from starting the update till finish to further work on this.

Ill try to dig if there are files from the time of update.

SHxKM, 2022-01-04:
@Koenkk I have the `log.txt` from when before the update started and after it had finished. Is there any way I can pass it to you privately?

SHxKM, 2022-01-05:
> @Koenkk I have the `log.txt` from when before the update started and after it had finished. Is there any way I can pass it to you privately?

The issue is more widespread than I thought - I've got at least 5 lights that show an older version on the UI than what their cluster reads. Only a cluster read AND a restart update the value. It's no biggie but @Koenkk  let me know how I can ship the logs privately.

Koenkk, 2022-01-05:
@SHxKM you can send it to me on telegram (@koenkk)

SHxKM, 2022-01-05:
> @SHxKM you can send it to me on telegram (@Koenkk)

Done. Can we re-open this ticket for better tracking?

Koenkk, 2022-01-06:
@SHxKM in your log I see the following:

```js
info  2022-01-04 15:56:52: Finished update of 'outub_econic_balcony_family', from '{"dateCode":"20190529","softwareBuildID":"1.56.7_r29434"}' to '{"dateCode":"20211021","softwareBuildID":"1.93.3"}'
```

Can you confirm that `{"dateCode":"20211021","softwareBuildID":"1.93.3"}` is the correct new version?

SHxKM, 2022-01-06:
> @SHxKM in your log I see the following:
>
> ```js
> info  2022-01-04 15:56:52: Finished update of 'outub_econic_balcony_family', from '{"dateCode":"20190529","softwareBuildID":"1.56.7_r29434"}' to [...]

Yes, but that’s not getting reflected in Z2M UI without a read from the cluster (like above) AND a restart.

SHxKM, 2022-01-07:
I don’t think outub_econic_balcony_family is the right device to look at. I’m AFK at the moment so I can’t scan the logs - but I’m sure it’s one of the devices there.

Koenkk, 2022-01-07:
I've tried to reproduce this with one of my Philips bulbs which still had an update available but failed to do so.

During the update:
[image]

Once the update was finished:
[image]

To debug this further I need some more information; in case you still have an update available:
- Make sure you are on z2m 1.22.2
- Make a copy of the data/database.db (call it database.before.db)
- Start z2m with herdsman debug log enabled
- Update bulb
- Stop z2m
- Make a copy of the data/database.db (call it database.after.db)

Now send me a zip with the herdsman debug logging + database before and after.

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

Paladin1404, 2022-01-09:
I have a Philips lamp which is not updated.
After the update it shows again that update is available with same version.

I have the same issue with Philips Hue dimmer switch version 1.

The logs and databases are uploaded here:
https://drive.google.com/drive/folders/1YWwu5RJ3q3rjR-JBhhdJXpW5XKlanhyY?usp=sharing

@Koenkk Please let me know once you have downloaded the files. I want to remove the files from Google Drive.

Koenkk, 2022-01-09:
@Paladin1404 it seems in your case the dimmer refuses the update for some reason, it might be that the dimmer first requires another (older) update to be loaded, could you try the following. Compliments for the logs files, exactly what I need.

- update to the latest dev branch (required because we will use a feature which is not in a release yet). (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)
- stop z2m
- create a file next to the z2m `configuration.yaml` called `my_index.json` with the following contents
```json
[
    {
        "fileVersion": 16783620,
        "fileSize": 271050,
        "manufacturerCode": 4107,
        "imageType": 270,
        "sha512": "5843552ab361d2d063e36be24785afcb8af34491ae721c2426da6afec94967acd4005d5e2abfcca5ebd10f3c9e39656524775b50cef115f67c3f636b9609d3c2",
        "url": "http://fds.dc1.philips.com/firmware/ZGB_100B_010E/16783620/100B-010E-01001904-ConfLight-ModuLum_0012.zigbee"
    }
]
```
- start z2m
- capture logs again
- try to update the hue dimmer switch
- provide database.db after the update

For my reference, relevant information:
```
[34mZigbee2MQTT:debug[39m 2022-01-08 23:32:25: Got upgrade end request for '0x0017880106519268': {"status":0,"manufacturerCode":4107,"imageType":270,"fileVersion":16785152}

[34mZigbee2MQTT:debug[39m 2022-01-08 23:37:07: Is new image available for '0x0017880106519268', current '{"fieldControl":0,"manufacturerCode":4107,"imageType":270,"fileVersion":16779030}', latest meta '{"fileVersion":16785152,"fileSize":271114,"url":"http://fds.dc1.philips.com/firmware/ZGB_100B_010E/16785152/100B-010E-01001F00-ConfLight-ModuLum_0012.zigbee","sha512":"9639556342d4c7210bc57a4a71655a6f16804b207542fe4558cd40b0b26758488a6adbbd4fd7d4d6390d2a153ae1fc5c45b9f3cc07bd98ce6aeb66f7be228b6f"}'

# db before any updates
{"id":24,"type":"Router","ieeeAddr":"0x0017880106519268","nwkAddr":64155,"manufId":4107,"manufName":"Philips","powerSource":"Mains (single phase)","modelId":"LTC015","epList":[11,242],"endpoints":{"11":{"profId":49246,"epId":11,"devId":544,"inClusterList":[0,3,4,5,6,8,768,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"LTC015","manufacturerName":"Philips","powerSource":1,"zclVersion":1,"appVersion":2,"stackVersion":1,"hwVersion":2,"dateCode":"20160907","swBuildId":"1.16.1_r19181"}},"lightingColorCtrl":{"attributes":{"colorCapabilities":16,"colorTempPhysicalMin":153,"colorTempPhysicalMax":454,"colorMode":2,"colorTemperature":366}},"genOnOff":{"attributes":{"onOff":1}},"genLevelCtrl":{"attributes":{"currentLevel":254}}},"binds":[],"configuredReportings":[],"meta":{"scenes":{"0_3":{"name":"AllLampsOff","state":{"color_mode":"color_temp","state":"OFF","brightness":254,"color_temp":366}},"1_3":{"name":"AllLampsOn","state":{"color_mode":"color_temp","state":"ON","brightness":254,"color_temp":366}}}}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":1,"hwVersion":2,"dateCode":"20160907","swBuildId":"1.16.1_r19181","zclVersion":1,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1641680329689,"useImplicitCheckin":true}
```

SHxKM, 2022-01-09:
@Koenkk could the above be related to issue #10139, and apply there also?

Koenkk, 2022-01-09:
@SHxKM devices not responding to ota requests are a different issues (seems to be on the device side)

Paladin1404, 2022-01-09:
@Koenkk I did the downgrade to version 1.50.2_r30933 using latest dev branch.

On branch dev commit cd60eb5069a91f26524d24b898894c6fbca36eac

The log and database files are here:
https://drive.google.com/drive/folders/1vArcY8OXcY7QZOcHH1XC-fRE3Wekf13g?usp=sharing

Unfortunately the log contains only the last ~9% of the update process. Let me know if you need the full from beginning. I can do then the update to newer version and redo the update to old version.

BTW: I did the update of the Philips light (https://www.zigbee2mqtt.io/devices/3216331P5.html#philips-3216331p5) and not the dimmer switch as you wrote in your post above.

I will keep the current configuration. Let me know once I can do more tests/verifications.

Paladin1404, 2022-01-09:
> @SHxKM devices not responding to ota requests are a different issues (seems to be on the device side)

For the dimmer switch I pressed them before try to check for ota updates and before start to execute the update. During the update process I pressed several times one of the buttons of the switches.
The hue dimmer switch goes in some kind of deep sleep state where they don't response to requests. To wake them up and "keep alive" pressing the button helped for me.
At least the update process was completely executed. The issue that the update is not taken is another issue that still happens for me.

SHxKM, 2022-01-09:
> At least the update process was completely executed. The issue that the update is not taken is another issue that still happens for me.

That's exactly what happened to me as I describe in my latest post in #10139, which I believe is the correct issue to track for the dimmers. Instead of pressing the buttons (which didn't work for me) I continuously read from the clusters to not let the device sleep. But as it happened to you, once it got to 100% and everything "succeeded", the version is still the same.

Koenkk, 2022-01-17:
I found the issue and fixed this now in the latest-dev
 (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Successfully updated an battery powered IKEA remote, after updating it immediately shows the new buildID and dateCode.

Paladin1404, 2022-01-18:
I can confirm that with yesterdays dev version my device "TRADFRI on/off switch" could be updated. It took some time, initial estimate was 10h but it was less. The update finished over night, I don't know how long it finally took.

@Koenkk I still have issues to update Hue dimmer switch. I guess that this issue is not yet fixed.
I did not try to update "Hue white ambiance Aurelle rectangle panel light".
Please let me know once you have something to test.

Thanks for your great work.

Koenkk, 2022-01-19:
@Paladin1404 have you tried to to update the hue dimmer switch with the latest-dev? Just updated mine successfully:

Before:
[image]

After:
[image]

SHxKM, 2022-01-19:
> @Paladin1404 have you tried to to update the hue dimmer switch with the latest-dev? Just updated mine successfully:
> 
> Before: [image]
> 
> After: [image]

This looks nice, if it works it should solve #10139 as well.

Paladin1404, 2022-01-19:
Yes, I updated after your post. Today I updated again to latest state.

I was able to update the hue ceiling lamp. This works for me now.

But still the hue dimmer switch stuck at version 5.45.1.17846.
I tried to update 2 switches.
I did also a remove from network. Reset switch and restart switch.
Then joining network and execute update.

Koenkk, 2022-01-20:
@Paladin1404 can you provide the debug log when updating those dimmer switches?

Paladin1404, 2022-01-20:
@Koenkk 

Please find the log here:
https://drive.google.com/drive/folders/1bVOPE1Ejy4EdBXwZW_z-RIzhq2SaAsQN?usp=sharing

Update of Hue dimmer switch firmware 5.45.1.17846

Friendly name and IEEE Address: 0x00178801096ad8ff

Koenkk, 2022-01-20:
@Paladin1404 I checked the log, for some reason the dimmer doesn't update indeed. This is especially strange since I had exactly the same fw on it and managed to update it successfully (I have no clue what the difference between our setups is, couldn't find anything strange in the log). I've checked the history of https://github.com/Koenkk/zigbee-OTA/blob/master/index.json but couldn't find other fws for this device. So I'm afraid I cannot solve this issue.

SHxKM, 2022-01-20:
> @Paladin1404 I checked the log, for some reason the dimmer doesn't update indeed. This is especially strange since I had exactly the same fw on it and managed to update it successfully (I have no [...]

I’m pretty sure this is also documented in the issue I opened #10139, maybe these require intermediary firmware?

Koenkk, 2022-01-20:
- It doesn't explain why the update worked on my device
- The swbuildid and datecode in https://github.com/Koenkk/zigbee2mqtt/issues/10139#issuecomment-1008349082 cannot be trusted. This was before the fix and as you can see the read fails: `Error Publish 'set' 'read' to 'buckram_quad_remote' failed: 'Error: Read 0x00XXXX0108f1d611/1 genBasic(["swBuildId"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'`

Paladin1404, 2022-01-21:
Hmm,

Do you have a link to old firmware? Maybe I can try to do first downgrade then upgrade to latest firmware.
This helped for Hue ceiling lamp.

Koenkk, 2022-01-21:
@Paladin1404 I don't

closed 2022-03-01 by github-actions[bot]

Luminger, 2023-02-07:
I've been struggling for a while to get a bunch of my [9290024406](https://www.zigbee2mqtt.io/devices/9290024406.html#philips-9290024406) (LWU001) updated. They won't update either with the official Hue bridge, the Bluetooth part of the official Hue Android app nor z2m.

When I got them brand new, it looked like they all requested new firmware images once (in my z2m setup). Once that happened, they're stuck on either 1.55.7_r28193 (20190322) or 1.76.10 (20201012).

When I watch for the `commandQueryNextImageRequest` dumps from z2m, I can see something strange though. All of them return `{"fieldControl":0,"fileVersion":16784918,"imageType":276,"manufacturerCode":4107}` which basically means they (believe?) they're already running the latest version, 16784918 (this should be https://github.com/Koenkk/zigbee-OTA/pull/241).

My current plan is to wait for the next round of OTAs for those bulbs and have debug logging enabled to see what's happening on the network.

Either the OTA process for those bulbs is somewhat broken or the `swBuildId` and `dateCode` attributes are unchanged in new images, which I don't think should be the case.

Alternatively another theory is that the bulbs never switched over to the new software version but have them stored in memory but were never told to boot into?

I shiuld have one last LWU001 which hasn't been connected to z2m before and should be in factory state. I'll see that I pair it and have both z2m and z2m herdsman with debug logging enabled running while trying to update it.

[1 more reply]