Koenkk/zigbee2mqtt #10136: Philips Hue Play - 440400982842

valfr, 2021-12-11:
## Information about the device + link
Philips HUE PLAY
New model ID = 440400982842
Same as = 915005733701
https://www.zigbee2mqtt.io/devices/915005733701.html#philips-915005733701

## data/database.db entry of the device
Link to support Herdsman => https://github.com/Koenkk/zigbee-herdsman-converters/actions/runs/1555052205

Thanks :)

HB25, 2021-12-11:
Hi,
  quick question: I have the same device and want to add support for it.  Do I have to do it the same way as described for new devices, i.e. create an external converter?   Since it has already been integrated in zigbee-herdsman-converters I was hoping to just update this and I'm fine but it did not work.

I tried to  npm update zigbee-herdsman-converters and I also patched the philips.cs with what I found on Github. This did not help.  

Heiko

valfr, 2021-12-27:
Hello,
Any idea to get it worked ?

Thx

Koenkk, 2021-12-27:
This device is already supported in the dev branch (and next release on 1 January). (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2021-12-27 by Koenkk

valfr, 2021-12-27:
many thanks :)

HB25, 2022-01-02:
Hello,

I just updated to the latest release but I still have no luck to have it added to the network.

Touchlink scan finds it: `{"data":{"found":[{"channel":11,"ieee_address":"0x001788010b438b17"}]},"status":"ok"}`
It also appears in the network map below `data/links` as follows but does not appear in `data/nodes` or in my HA network map. 

Neither touchlink reset nor the 5s power reset seems to work (although the device flashes after the reset).

I have the feeling that the device is already paired from release 1.22.1. It is already present in configuration.yaml with a friendly name.  Uncommenting it did not help but it also does not appear in HA with this name.

Has anybody hints where to look at or what to try to get this running?

```
             {
                    "depth": 15,
                    "linkquality": 132,
                    "lqi": 132,
                    "relationship": 2,
                    "routes": [],
                    "source": {
                        "ieeeAddr": "0x001788010b438b17",
                        "networkAddress": 42410
                    },
                    "sourceIeeeAddr": "0x001788010b438b17",
                    "sourceNwkAddr": 42410,
                    "target": {
                        "ieeeAddr": "0x0017880108fbd5d3",
                        "networkAddress": 5611
                    },
                    "targetIeeeAddr": "0x0017880108fbd5d3"
                }
```

Thanks in advance and  Happy new year everyone!

anthonycvella, 2022-04-09:
@Koenkk Was this released with 1.18.1 because it shows as unsupported for me still.

Koenkk, 2022-04-09:
@anthonycvella 
If you are still on 1.18.1 check first point of https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.19.0

anthonycvella, 2022-04-09:
@Koenkk I realized that when I was looking at the version :P, been re-interviewing all my devices today since I couldn't migrate them, but this seems to be working great now!