Koenkk/zigbee2mqtt #10137: Web UI memory leak in Firefox?

artur3, 2021-12-11:
After update to 1.22.1 web UI loads very slowly in Firefox and if tab Devices is left open for longer time (just few minutes) all RAM memory will be taken and system (openSUSE) becomes too slow to take any action.

This happens every time When web UI is opened in Firefox on all my computers.

For some reason web UI won't fully load in Chromium, only top menus and "loading please wait".

zigbee2mqtt works fine all the time.

KDE process monitor:

<h1 id="heading" style="color: rgb(35, 38, 39); font-family: &quot;Roboto Slab&quot;; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;">Process 2101 - Web</h1><span id="errorMessage" class="errorMessage" style="color: red; font-family: &quot;Roboto Slab&quot;; font-size: medium; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;"></span><span style="color: rgb(35, 38, 39); font-family: &quot;Roboto Slab&quot;; font-size: medium; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; background-color: rgb(239, 240, 241); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial; display: inline !important; float: none;"></span><h2 id="SummaryHeading" style="color: rgb(35, 38, 39); font-family: &quot;Roboto Slab&quot;; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;">Summary</h2><span id="processsummary" style="color: rgb(35, 38, 39); font-family: &quot;Roboto Slab&quot;; font-size: medium; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;">The process<span> </span><b>Web</b><span> </span>(with pid 2101) is using approximately<span> </span><span class="memory" title="5155098 KB" style="color: blue; white-space: nowrap; cursor: default;">4.9 GB</span><span> </span>of memory.<br>It is using<span> </span><span class="memory" title="5152660 KB" style="color: blue; white-space: nowrap; cursor: default;">4.9 GB</span><span> </span>privately, and a further<span> </span><span class="memory" title="11552 KB" style="color: blue; white-space: nowrap; cursor: default;">11.3 MB</span><span> </span>that is, or could be, shared with other programs.<br>Dividing up the shared memory between all the processes sharing that memory we get a reduced shared memory usage of<span> </span><span class="memory" title="2438 KB" style="color: blue; white-space: nowrap; cursor: default;">2.4 MB</span>. Adding that to the private usage, we get the above mentioned total memory footprint of<span> </span><span class="memory" title="5155098 KB" style="color: blue; white-space: nowrap; cursor: default;">4.9 GB</span>.<br><span class="memory" title="10432 KB" style="color: blue; white-space: nowrap; cursor: default;">10.2 MB</span><span> </span>is swapped out to disk, probably due to a low amount of available memory left.<br></span><span style="color: rgb(35, 38, 39); font-family: &quot;Roboto Slab&quot;; font-size: medium; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; background-color: rgb(239, 240, 241); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial; display: inline !important; float: none;"></span><h2 id="LibraryUsageHeading" style="color: rgb(35, 38, 39); font-family: &quot;Roboto Slab&quot;; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;">Library Usage</h2><span id="libraryusageintro" style="color: rgb(35, 38, 39); font-family: &quot;Roboto Slab&quot;; font-size: medium; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;">The memory usage of a process is found by adding up the memory usage of each of its libraries, plus the process's own heap, stack and any other mappings, plus the stack of its 21 threads.</span><span style="color: rgb(35, 38, 39); font-family: &quot;Roboto Slab&quot;; font-size: medium; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; background-color: rgb(239, 240, 241); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial; display: inline !important; float: none;"></span><br style="color: rgb(35, 38, 39); font-family: &quot;Roboto Slab&quot;; font-size: medium; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;"><div class="summaryTable" style="float: left; padding: 0px; min-width: 48%; color: rgb(35, 38, 39); font-family: &quot;Roboto Slab&quot;; font-size: medium; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;">

Private
--
hide
5149560 KB | [heap]
2720 KB | /usr/lib64/firefox/libxul.so
136 KB | [stack]
32 KB | /usr/lib64/libfreetype.so.6.17.1
32 KB | /usr/lib64/libstdc++.so.6.0.29
28 KB | /memfd:mozilla-ipc (deleted)
28 KB | /usr/lib64/libsqlite3.so.0.8.6
16 KB | /lib64/libc-2.31.so
12 KB | /usr/lib64/libfreeblpriv3.so
12 KB | /usr/lib64/libsoftokn3.so
12 KB | /usr/lib64/libfontconfig.so.1.10.1
8 KB | /usr/lib64/firefox/firefox
8 KB | /usr/lib64/libglib-2.0.so.0.6200.6
8 KB | /usr/lib64/libnspr4.so
8 KB | /lib64/libdl-2.31.so
8 KB | /lib64/libpthread-2.31.so
8 KB | /lib64/ld-2.31.so
4 KB | /usr/lib64/libplc4.so
4 KB | /usr/lib64/libGLdispatch.so.0.0.0
4 KB | /lib64/libz.so.1.2.11
4 KB | /usr/lib64/libcairo.so.2.11600.0
4 KB | /lib64/libm-2.31.so
4 KB | /usr/lib64/firefox/libmozsandbox.so

</div></div><div class="fullDetails" style="clear: both; float: left; color: rgb(35, 38, 39); font-family: &quot;Roboto Slab&quot;; font-size: medium; font-style: normal; font-variant-ligatures: normal; font-variant-caps: n [... cut ...] ble columns. An empty filename means that it is an<span> </span><span title="This is a fast way to allocate and clear a block of space.   See the mmap(2) man page under MAP_ANONYMOUS" class="definedWord" style="border-bottom: 1px dashed blue; cursor: help;">anonymous mapping</span>.<br>Both the<span> </span><span class="definedWord" title="Memory Management Unit" style="border-bottom: 1px dashed blue; cursor: help;">MMU</span><span> </span>page size and the kernel page size are 4 KB.</div></div>

tabnul, 2021-12-11:
same here, weg gui is VERY slow, especially initially it takes ages to load the online/offline status of devices.

CSchlipp, 2021-12-13:
Same on Windows using latest Chrome. 
With >100 devices, the UI sometimes does not even load at all, but only shows a spinning tab icon.

nurikk, 2021-12-14:
Hi, try to switch to the latest dev branch

https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html#switch-to-the-dev-branch

nurikk, 2021-12-14:
> ld be, shared with other programs.
> Dividing up the shared memory between all the processes sha

Hi!
Can you attach complete state.json file from `Settings -> Tools -> Download state` ?

artur3, 2021-12-14:
Firefox:
[state.json.zip](https://github.com/Koenkk/zigbee2mqtt/files/7713044/state.json.zip)

Chromium (after few minutes of waiting):
[state.json.zip](https://github.com/Koenkk/zigbee2mqtt/files/7713117/state.json.zip)

nurikk, 2021-12-14:
Can you test on latest dev branch?

https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html#switch-to-the-dev-branch

artur3, 2021-12-14:
After update to dev, devices page loaded almost instantly in Firefox. Chromium shows empty page.

pvi25, 2021-12-14:
I installed the 1.22.1-1 version couple of days ago as home assistant addon and also got the 'Loading, please wait' screen (from Chrome). Just installed the dev version and now I do get a proper page but no devices are shown either, like @artur3. Just started with zigbee2mqtt, never had any devices before in there, so I dont know if the problem of the empty page is because the only zigbee device that I have connected to my usb stick (motion detector) is not recognized by zigbee2mqtt or that it is another problem (the device does work with zha)

nurikk, 2021-12-15:
> After update to dev, devices page loaded almost instantly in Firefox. Chromium shows empty page.

Can you check chromium dev tools logs?

artur3, 2021-12-15:
runtime.3296ac9a5caef629c7d6.js:1 Failed to load resource: the server responded with a status of 404 (Not Found)
192.168.x.x/:1 Refused to apply style from 'http://192.168.x.x:8081/main.7379448ff31d4e2a65b8.css' because its MIME type ('text/html') is not a supported stylesheet MIME type, and strict MIME checking is enabled.
vendor.49fee1be1555be7ec8f9.js:1 Failed to load resource: the server responded with a status of 404 (Not Found)
main.03d2b8de8a4207f907b0.js:1 Failed to load resource: the server responded with a status of 404 (Not Found)
192.168.x.x/:1 Refused to apply style from 'http://192.168.x.x:8081/main.7379448ff31d4e2a65b8.css' because its MIME type ('text/html') is not a supported stylesheet MIME type, and strict MIME checking is enabled.
DevTools failed to load source map: Could not load content for http://192.168.x.x:8081/scripts/vendor.109be7bda8112bf971cc.css.map: HTTP error: status code 404, net::ERR_HTTP_RESPONSE_CODE_FAILURE

I tried on another computer and it loaded in Chromium. Visibly slower than in Firefox, first only text and then rearranged it with graphic. Devices status loaded quickly.

artur3, 2021-12-15:
Report from tool called Lighthouse:
[192.zigbee2mqttUI_8081-20211215T165519.zip](https://github.com/Koenkk/zigbee2mqtt/files/7720899/192.zigbee2mqttUI_8081-20211215T165519.zip)

Now page loads but without devices.

adorobis, 2021-12-21:
I have the same issue "loading please wait". I have latest version installed on TrueNAS Core (FreeBSD) jail. After restarting the jail, sometimes need to do it twice, it would work for a while before again failing to load "devices" and "dashboard" pages.

pvi25, 2021-12-31:
Any news on this? Tried it today again with another zigbee device (power plug), but same issue, normal version gives 'loading, please wait', edge version gives empty device list. Both devices work in zha.

nurikk, 2021-12-31:
This issue already resolved in dev branch, tomorrow happens scheduled release.
Happy new year

closed 2021-12-31 by nurikk