Koenkk/zigbee2mqtt #10139: Cannot OTA update Philips end-devices

SHxKM, 2021-12-11:
## What happened
I am completely unable to OTA update _any_ devices like the [Philips motion sensor  - 9290012607](https://www.zigbee2mqtt.io/devices/9290012607.html#philips-9290012607) (currently 4 paired) and the [Hue dimmer switch - 324131092621](https://www.zigbee2mqtt.io/devices/324131092621.html#philips-324131092621) (currently 8 paired) for quite a while now.

## What did you expect to happen

To be able to update these devices via Zigbee2MQTT.

## How to reproduce it (minimal and precise)

1. Pair a device like `9290012607` (firmware I have: `6.1.1.27575`) or `324131092621` (firmware I have: `5.45.1.17846`)
2. Go to OTA screen in Z2M UI
3. Click  "update device firmware"
4. Wait a while while inspecting the logs

Logs, each appear at a different stage:

```
Updating 'buckram_quad_remote' to latest firmware
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Updating 'buckram_quad_remote' to latest firmware","meta":{"device":"buckram_quad_remote","status":"update_in_progress"},"type":"ota_update"}'
```

```
Debug Updating to latest '0x0017880108f1d611' (RWL021)
Debug Using endpoint '2'
```

```
Error Update of 'buckram_quad_remote' failed (Device didn't respond to OTA request)
Debug Error: Device didn't respond to OTA request at requestOTA (/app/node_modules/zigbee-herdsman-converters/lib/ota/common.js:196:15) at Object.updateToLatest (/app/node_modules/zigbee-herdsman-converters/lib/ota/common.js:278:21) at OTAUpdate.onMQTTMessage (/app/lib/extension/otaUpdate.ts:234:21)
```

## Debug info
Zigbee2MQTT version: 1.22.1-1
Adapter hardware: Conbee II

Bascht74, 2021-12-12:
@SHxKM 
Did you press any button on the dimmers after you click update? 
I had to wake up the devices for OTA...

I don't know if there is anything in Z2M, like -> "after pressing the OTA button wait for the device to wake up by itself (e.g. report battery) and then send an OTA-command". 
For me it usually times out after some time...

Seb

SHxKM, 2021-12-12:
@Bascht74,

> Did you press any button on the dimmers after you click update?

I have, it rarely helps. Even if it does, the operation will time out after little progress.

> Did you press any button on the dimmers after you click update?

> For me it usually times out after some time...

So does this mean clicking a button works for you, or does not?

Bascht74, 2021-12-12:
> So does this mean clicking a button works for you, or does not?

Yes, that worked for me for all devices…

SHxKM, 2021-12-12:
> > So does this mean clicking a button works for you, or does not?
> 
> Yes, that worked for me for all hue devices…

Unfortunately does not work for me. Clicking any button before/after and both before and after the update command does not work.

SHxKM, 2021-12-28:
**Update (Dec 28 2021):**

- For motion sensors (9290012607) I can sometimes start the process after trying a few times, but most of the time it won't finish the OTA update.
- For the remotes (324131092621) it's still quite impossible to OTA update.

@Koenkk any idea why this might be happening? is there any way to "force" the device to respond to OTA request?

jeroenhoebe, 2021-12-29:
After also getting the same Error. I was finally able to update my remote 324131092621 to (6.1.1.28573) through the OTA update. 

It didn't go smoothly though. Needed to press a button every few minutes so it wouldn't lose connection. Then after it finished nothing really changed. But in the logs the status said no update available anymore.

Then I needed to get it to update in zigbee2mqtt because it didn't show the correct firmware yet. So I tried a few things ( remove remote, re-insert battery, holding setup till it flashed, re-pairing) and then after two tries I saw that is was updated. (maybe not exactly in that order, I tried a few things).

SHxKM, 2021-12-29:
> It didn't go smoothly though. Needed to press a button every few minutes so it wouldn't lose connection.

I have tried this exactly numerous times, without success.

Tonio16, 2022-01-01:
Hello

For 324131092621 I can update it but though z2m write it is successfull, the firmware version is the same and thedevice is stillavaillble for a new update. Realy strange behavior.
For philipps or ikea bulbs no issue.

Antoine

SHxKM, 2022-01-01:
@Koenkk - do you recognize these bindings? you told me Z2M shouldn't create certain bindings in another issue, could these bindings be the cause of trouble here?

[image]

SHxKM, 2022-01-02:
Well this is super weird.

I decided to try following @jeroenhoebe's advice and try once again with a  `324131092621` (first-gen remote). After several failed attempts to even start the update process, here's what happened:

1. The device started the process, got to ~35%. Then Failed.
2. The device started the process, got to ~35-40%. Then Failed. 
3. During all those attempts I held the device's hand: click off every few minutes, and read `swBuildId` every few minutes.
4. Third attempt was successful, but at the end of the process the firmware was the same version it was before the "successful" update: `5.45.1.17846`.
5. Reading `swBuildId` from the dev console also shows `5.45.1.17846`.
6. BUT, the device now shows "Check for new updates" and reports no new firmware version in the UI. Super weird because before the "successful" update, it was on the same version, and it wanted a new one. It doesn't seem to have had one installed, neither from the UI nor by reading `swBuildId`.

Bascht74, 2022-01-02:
> 6\. BUT, the device now shows "Check for new updates" and reports no new firmware version in the UI. Super weird because before the "successful" update, it was on the same version, and it wanted a new one. It doesn't seem to have had one installed, neither from the UI nor by reading `swBuildId`.

Remove the device from Z2M, restart Z2M (e.g. restart docker / adding) and join the device again. This is probably a known bug: doesn’t show the right firmware version after OTA.

Koenkk, 2022-01-02:
@SHxKM I don't expect binding to create any troubles

SHxKM, 2022-01-02:
> Remove the device from Z2M, restart Z2M (e.g. restart docker / adding) and join the device again. This is probably a known bug: doesn’t show the right firmware version after OTA.

Please read my update. I will try this anyway.

puijken, 2022-01-03:
Was just able to update a HUE DIMMER (324131092621) via zigbee2mqtt.
Steps followed:

1. Force remove device from z2m
2. Reset device with paper clip
3. Rejoin device to z2m
4. OTA update
5. Press button every minute
6. Wait and see updated version :)

Next steps:
- Update another dimmer switch regular way (press button once before update, click OTA update, wait)
- Restart z2m to see if first updated dimmer switch still reports correct version

Tried this multiple times before but failed every time .. only things that have changed in the meantime are newer z2m version (1.22.2) and newer coordinator firmware installed today (20211217). 

Will update here:
Current status (15h03 - dimmer updating 28%)

SHxKM, 2022-01-03:
> Was just able to update a HUE DIMMER (324131092621) via zigbee2mqtt. Steps followed:
> 
> 1. Force remove device from z2m
> 2. Reset device with paper clip
> 3. Rejoin device to z2m (keep default name)
> 4. OTA update
> 5. Press button every minute
> 6. Wait and see updated version :)
> 
> Next steps:
> 
> * Update another dimmer switch regular way (press button once before update, click OTA update, wait)
> * Restart z2m to see if first updated dimmer switch still reports correct version
> 
> Tried this multiple times before but failed every time .. only things that have changed in the meantime are newer z2m version (1.22.2) and newer coordinator firmware installed today (20211217).
> 
> **Will update here:** _Current status second dimmer - updating regular way: 15h03 - dimmer updating 28% 15h31 - still updating, no more buttons pressed - 88% 15h43 - update done, logs showing update from version 5.45.1.17846 to 5.45.1.17846 (for the previous dimmer it showed 5.45.1.17846 to 6.1.1.28573) So update failed - will retry with procedure 1._
> 
> _Updating second dimmer - removed/rejoin device + regular button press: 15h51 - dimmer updating 3%_

Unrealistic to me because I have > 10 of those remotes and it would take days, and I would have to remap every automation, group, etc...that these entities belong to. And not that it worked for me anyway.

puijken, 2022-01-03:
> Unrealistic to me because I have > 10 of those remotes and it would take days, and I would have to remap every automation, group, etc...that these entities belong to. And not that it worked for me anyway.

Yeah I understand, I have now done 2 and took up an entire afternoon pressing the buttons .. one more to go after this if it works. Luckily all logic/automation is in nodered, so after rejoining and updating with the default name, I can very easily rename the device again to get all up and running again.

SHxKM, 2022-01-03:
> Yeah I understand, I have now done 2 and took up an entire afternoon pressing the buttons .. one more to go after this if it works. Luckily all logic/automation is in nodered, so after rejoining and [...]

Yeah using the same name is a good strategy, I can do the same. But again, too many devices in my case :)

Did method 1 work for the second remote?

puijken, 2022-01-03:
No .. did not work .. same error again:

MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"id":"0x0017880108f1bcce","transaction":"21m3s-4"},"meta":{"device":"0x0017880108f1bcce","from":{"dateCode":"20160302","softwareBuildID":"5.45.1.17846"},"status":"update_succeeded","to":{"dateCode":"20160302","softwareBuildID":"5.45.1.17846"}},"type":"ota_update"}'

Guess I was lucky the first time.

oscfor, 2022-01-04:
I have exactly the same problem with my 324131092621. I have updated it numerous times but only ends up with the same version. Removed, repaired, reset, new batteries, still firmware 5.45.1.17846.

Paladin1404, 2022-01-05:
I discovered the same issue when updating my Philips HUE devices.

Hue dimmer switch with firmware 5.45.1.17846: https://www.zigbee2mqtt.io/devices/324131092621.html#philips-324131092621

Hue white ambiance Aurelle rectangle panel light with firmware 1.16.1_r19181: https://www.zigbee2mqtt.io/devices/3216331P5.html#philips-3216331p5

The update process was according to the log successful. Remove, reset and reconnect device does not help.

SHxKM, 2022-01-05:
> I discovered the same issue when updating my Philips HUE devices.
>
> Hue dimmer switch with firmware 5.45.1.17846: https://www.zigbee2mqtt.io/devices/324131092621.html#philips-324131092621
>
> Hue white [...]

I think your issue (at least with the Aurelle) is different and has more to do with #10131, this topic is strictly about the OTA devices (and probably only the remote at this point). I'm sure if you follow the steps I outlined in #10131 (read cluster, restart) you'll see that the Aurelle update did work, it's just that Z2M is unaware of it for some reason.

Paladin1404, 2022-01-05:
> > I discovered the same issue when updating my Philips HUE devices.
> > Hue dimmer switch with firmware 5.45.1.17846: https://www.zigbee2mqtt.io/devices/324131092621.html#philips-324131092621
> > Hue white ambiance Aurelle rectangle panel light with firmware 1.16.1_r19181: https://www.zigbee2mqtt.io/devices/3216331P5.html#philips-3216331p5
> > The update process was according to the log successful. Remove, reset and reconnect device does not help.
> 
> I think your issue (at least with the Aurelle) is different and has more to do with #10131, this topic is strictly about the end devices (and probably only the remote at this point). I'm sure if you follow the steps I outlined in #10131 (read cluster, restart) you'll see that the Aurelle update did work, it's just that Z2M is unaware of it for some reason.

I followed the instructions from the other ticket but still the same issue for me. Both devices are not updated.

SHxKM, 2022-01-09:
So I just tried updating a remote and have some interesting logs. I managed to keep the OTA update going until 100%, but it seems that it fails and reverts.

Here are two snippets I see when it goes near the end (the device I'm updating is called `buckram_quad_remote`):

```
Debug Got upgrade end request for '0x00XXXX0108f1d611': {"status":0,"manufacturerCode":4107,"imageType":265,"fileVersion":1107324829}
Debug Update succeeded, waiting for device to restart
Debug Device 'buckram_quad_remote' announced itself
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"buckram_quad_remote","ieee_address":"0x00XXXX0108f1d611"},"type":"device_announce"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"buckram_quad_remote"},"type":"device_announced"}'
Debug Received Zigbee message from 'buckram_quad_remote', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":200}' from endpoint 2 with groupID null
Info MQTT publish: topic 'zigbee2mqtt/buckram_quad_remote', payload '{"action":null,"action_duration":null,"battery":100,"brightness":255,"counter":1,"linkquality":183,"update":{"progress":99.53,"remaining":9,"state":"updating"},"update_available":false}'
Debug Retrieving state of 'buckram_quad_remote' after reconnect
```

and after that

```
Info Configuring 'buckram_quad_remote'
Error Publish 'set' 'read' to 'buckram_quad_remote' failed: 'Error: Read 0x00XXXX0108f1d611/1 genBasic(["swBuildId"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'
Debug Error: Read 0x00XXXX0108f1d611/1 genBasic(["swBuildId"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received) at DeconzAdapter.sendZclFrameToEndpoint (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:641:23) at Endpoint.sendRequest (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:291:20) at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:422:28) at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/converters/toZigbee.js:39:28) at Publish.onMQTTMessage (/app/lib/extension/publish.ts:246:36)
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'read' to 'buckram_quad_remote' failed: 'Error: Read 0x00XXXX0108f1d611/1 genBasic([\"swBuildId\"], {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (no response received)'","meta":{"friendly_name":"buckram_quad_remote"},"type":"zigbee_publish_error"}'
Debug Received Zigbee message from 'buckram_quad_remote', type 'readResponse', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":200}' from endpoint 2 with groupID null
Info MQTT publish: topic 'zigbee2mqtt/buckram_quad_remote', payload '{"action":null,"action_duration":null,"battery":100,"brightness":255,"counter":1,"linkquality":183,"update":{"state":"idle"},"update_available":false}'
Info Successfully configured 'buckram_quad_remote'
Error Publish 'set' 'read' to 'buckram_quad_remote' failed: 'Error: Read 0x00XXXX0108f1d611/1 genBasic(["swBuildId"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'
Debug Error: Read 0x00XXXX0108f1d611/1 genBasic(["swBuildId"],
```

I don't know if this means anything.

Tonio16, 2022-02-03:
@Koenkk  The 1.23  has still this issue. I troed to update my two 324131092621 but no success.

Antoine

SHxKM, 2022-02-03:
@Koenkk Just tried with `1.23` and it did not succeed for me. After holding its hand with reading the `swBuildId` throughout the update, these are the logs. Goes from 0% to 100% and then just goes back to report the same verison.

```
info  2022-02-03 20:14:34: Update of 'centris_remote' at 97.67%, ≈ 1 minutes remaining
info  2022-02-03 20:14:34: MQTT publish: topic 'zigbee2mqtt/centris_remote', payload '{"action":null,"action_duration":null,"battery":100,"brightness":159,"counter":1,"linkquality":255,"update":{"progress":97.67,"remaining":60,"state":"updating"},"update_available":false}'
info  2022-02-03 20:14:34: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Update of 'centris_remote' at 97.67%, ≈ 1 minutes remaining","meta":{"device":"centris_remote","progress":97.67,"status":"update_progress"},"type":"ota_update"}'
info  2022-02-03 20:15:05: Update of 'centris_remote' at 98.86%, ≈ 0 minutes remaining
info  2022-02-03 20:15:05: MQTT publish: topic 'zigbee2mqtt/centris_remote', payload '{"action":null,"action_duration":null,"battery":100,"brightness":159,"counter":1,"linkquality":255,"update":{"progress":98.86,"remaining":30,"state":"updating"},"update_available":false}'
info  2022-02-03 20:15:05: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Update of 'centris_remote' at 98.86%, ≈ 0 minutes remaining","meta":{"device":"centris_remote","progress":98.86,"status":"update_progress"},"type":"ota_update"}'
info  2022-02-03 20:15:35: Update of 'centris_remote' at 100.00%
info  2022-02-03 20:15:35: MQTT publish: topic 'zigbee2mqtt/centris_remote', payload '{"action":null,"action_duration":null,"battery":100,"brightness":159,"counter":1,"linkquality":255,"update":{"progress":100,"remaining":30,"state":"updating"},"update_available":false}'
info  2022-02-03 20:15:35: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Update of 'centris_remote' at 100.00%","meta":{"device":"centris_remote","progress":100,"status":"update_progress"},"type":"ota_update"}'
debug 2022-02-03 20:15:37: Device 'centris_remote' announced itself
info  2022-02-03 20:15:37: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"centris_remote","ieee_address":"0x0017880108f77dd2"},"type":"device_announce"}'
info  2022-02-03 20:15:37: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"centris_remote"},"type":"device_announced"}'
info  2022-02-03 20:15:37: Finished update of 'centris_remote'
info  2022-02-03 20:15:37: MQTT publish: topic 'zigbee2mqtt/centris_remote', payload '{"action":null,"action_duration":null,"battery":100,"brightness":159,"counter":1,"linkquality":255,"update":{"state":"idle"},"update_available":false}'
debug 2022-02-03 20:15:37: Received Zigbee message from 'centris_remote', type 'readResponse', cluster 'genBasic', data '{"dateCode":"20160302","swBuildId":"5.45.1.17846"}' from endpoint 1 with groupID null
info  2022-02-03 20:15:37: Configuring 'centris_remote'
info  2022-02-03 20:15:37: Device 'centris_remote' was updated from 'null' to '{"dateCode":"20160302","softwareBuildID":"5.45.1.17846"}'
info  2022-02-03 20:15:38: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"id":"centris_remote","transaction":"6daf2-1"},"meta":{"device":"centris_remote","from":null,"status":"update_succeeded","to":{"dateCode":"20160302","softwareBuildID":"5.45.1.17846"}},"type":"ota_update"}'
info  2022-02-03 20:15:38: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"from":null,"id":"centris_remote","to":{"date_code":"20160302","software_build_id":"5.45.1.17846"}},"status":"ok","transaction":"6daf2-1"}'
debug 2022-02-03 20:15:38: Received Zigbee message from 'centris_remote', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":200}' from endpoint 2 with groupID null
info  2022-02-03 20:15:38: MQTT publish: topic 'zigbee2mqtt/centris_remote', payload '{"action":null,"action_duration":null,"battery":100,"brightness":159,"counter":1,"linkquality":255,"update":{"state":"idle"},"update_available":false}'
debug 2022-02-03 20:15:39: Retrieving state of 'centris_remote' after reconnect
debug 2022-02-03 20:15:41: Received Zigbee message from 'centris_remote', type 'readResponse', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":200}' from endpoint 2 with groupID null
info  2022-02-03 20:15:41: MQTT publish: topic 'zigbee2mqtt/centris_remote', payload '{"action":null,"action_duration":null,"battery":100,"brightness":159,"counter":1,"linkquality":255,"update":{"state":"idle"},"update_available":false}'
info  2022-02-03 20:15:41: Successfully configured 'centris_remote'
debug 2022-02-03 20:16:50: Received Zigbee message from 'centris_remote', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":1107314102,"imageType":265,"manufacturerCode":4107}' from endpoint 2 with groupID null
debug 2022-02-03 20:16:50: Device 'centris_remote' requested OTA
info  2022-02-03 20:16:51: MQTT publish: topic 'zigbee2mqtt/centris_remote', payload '{"action":null,"action_duration":null,"battery":100,"brightness":159,"counter":1,"linkquality":255,"update":{"state":"available"},"update_available":true}'
info  2022-02-03 20:16:51: Update available for 'centris_remote'
[... cut ...]
debug 2022-02-03 20:17:20: Received Zigbee message from 'centris_remote', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":1107314102,"imageType":265,"manufacturerCode":4107}' from endpoint 2 with groupID null
debug 2022-02-03 20:17:20: Device 'centris_remote' requested OTA
debug 2022-02-03 20:17:30: Received Zigbee message from 'centris_remote', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":1107314102,"imageType":265,"manufacturerCode":4107}' from endpoint 2 with groupID null
debug 2022-02-03 20:17:30: Device 'centris_remote' requested OTA

hughb8on, 2022-02-04:
Just adding this comment so I see any updates on this issue, and to let you know I have the same problem with my Hue v1 dimmers.

domosgabor, 2022-02-04:
I have the same problem, could update only 4 of my 12 devices. Firmware version is 5.45.1.17846.

Koenkk, 2022-02-05:
I think the hue dimmer switch requires a 2 step upgrade in some cases, but I don't have any other firmwares (https://github.com/Koenkk/zigbee2mqtt/issues/10131#issuecomment-1017841050)

TheJulianJES, 2022-02-05:
Mhm, I also can't find any images between file version 1107314102 and 1107324829 (that update is already older than 1,5 years) for image type 265. (To me, it doesn't look like an intermediate version is required.)
Are there any other OTAs I should look for?

(Perhaps trying -- with another stick -- and ZHA's OTA process can "solve it"? Not sure what it would do do differently though, as the image seems to be successfully sent with Z2M(?))

poonab, 2022-02-06:
Same problem here, two V1 dimmers on 5.45.1.17846 that won't update

Koenkk, 2022-02-06:
@TheJulianJES are you able to update it with ZHA? If trying, please make a sniff when doing this: https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html#with-cc2531

TheJulianJES, 2022-02-06:
@Koenkk I've already updated my two EU-style RWL021 dimmers months ago with no issues with ZHA. I don't have Hue dimmers with older firmware at the moment. (Perhaps it's possible to downgrade them though?)
I probably won't be able to provide a sniff in the near future.

deviantintegral, 2022-03-07:
I took a look at this with my RWL020. I originally tried a normal update, which "failed" in that the firmware version is reported as the same old version, but I can also no longer apply updates as it says it's up to date.

- Firmware Build Date: 20160302
- Firmware version: 5.45.1.17846

Looking at the OTA update request:

```json
{
  "frame": {
    "Header": {
      "frameControl": {
        "frameType": 1,
        "manufacturerSpecific": false,
        "direction": 0,
        "disableDefaultResponse": false,
        "reservedBits": 0
      },
      "transactionSequenceNumber": 116,
      "manufacturerCode": null,
      "commandIdentifier": 1
    },
    "Payload": {
      "fieldControl": 0,
      "manufacturerCode": 4107,
      "imageType": 265,
      "fileVersion": 1107324829
    },
    "Command": {
      "ID": 1,
      "response": 2,
      "parameters": [
        {
          "name": "fieldControl",
          "type": 32
        },
        {
          "name": "manufacturerCode",
          "type": 33
        },
        {
          "name": "imageType",
          "type": 33
        },
        {
          "name": "fileVersion",
          "type": 35
        }
      ],
      "name": "queryNextImageRequest"
    }
  },
  "address": 33790,
  "endpoint": 2,
  "linkquality": 98,
  "groupID": 0,
  "wasBroadcast": false,
  "destinationEndpoint": 1
}
```

What confuses me is that `"fileVersion": 1107324829` matches the file version in the index:

https://github.com/Koenkk/zigbee-OTA/blob/126f99b167c2334f104099dc92a9e052fa2a9e9d/index.json#L125

```json
    {
        "fileVersion": 1107324829,
        "fileSize": 240760,
        "manufacturerCode": 4107,
        "imageType": 265,
        "sha512": "d20057d6c4e61d4ccdd3947b8a8c9bf9e126a26c080b9ce3e6839a0615b912c6f93246a60c55a44f2c8b8ec49f5041f46ac38b8a3c0a1c6ab44f062cc62089cf",
        "url": "http://fds.dc1.philips.com/firmware/ZGB_100B_0109/1107324829/Switch-ATmega_6.1.1.28573_0012.sbl-ota"
    }
```

What it seems like to me is somehow the file version got incremented, but the firmware update didn't apply? Or the firmware _is_ up to date, but reporting the wrong version?

The remote is not working to turn off three non-Philips bulbs, which is why I want to update the firmware in case it's been fixed.

https://github.com/Koenkk/zigbee2mqtt/discussions/11543

As for next steps, my thinking is to set up a local index with `force` set to `true`, to re-apply the same firmware image. Is there anything else I can do to help gather information before doing that?

deviantintegral, 2022-04-07:
I tried the above by forcing the firmware upgrade. When I forced the upgrade, the firmware version reflected in the UI immediately updated to the latest version, and the firmware upgrade was not actually executed. This makes me think the issue is not that the OTA is failing, but that the version reporting after the OTA is complete is somehow being lost.

zandadoum, 2022-04-25:
> I tried the above by forcing the firmware upgrade. When I forced the upgrade, the firmware version reflected in the UI immediately updated to the latest version, and the firmware upgrade was not [...]

Hello,
so, is it solved for you tho?
1) upgrade normally. goes to 100% but then reports old version
2) force upgrade, doesn't do anything but now reports correct version
is this correct? if so, because I did (1) could you show me how to do (2)?

WatskeBart, 2022-05-20:
> I placed the following in `/opt/zigbee2mqtt/data/my_index.json`:
> 
> ```json
> [
>     {
>         "url": "Switch-ATmega_6.1.1.28573_0012.sbl-ota",
> 	"fileVersion": 1107324829,
>         "fileSize": 240760,
>         "manufacturerCode": 4107,
>         "imageType": 265,
>         "sha512": "d20057d6c4e61d4ccdd3947b8a8c9bf9e126a26c080b9ce3e6839a0615b912c6f93246a60c55a44f2c8b8ec49f5041f46ac38b8a3c0a1c6ab44f062cc62089cf",
> 	"force": true
>     }
> ]
> ```
> 
> And set the index location as shown in https://www.zigbee2mqtt.io/guide/configuration/ota-device-updates.html#ota-index-override-file
> 
> Then I triggered the upgrade in the UI.

Unfortunately it didn't work for me.
Z2M debug output was:
`ZigbeeOTA: Loading override index /config/zigbee2mqtt/my_index.json
Is new image available for '0x0017880108f1a5c7', current '{"fieldControl":0,"manufacturerCode":4107,"imageType":265,"fileVersion":1107314102}', latest meta '{"fileVersion":1107324829,"fileSize":240760,"url":"Switch-ATmega_6.1.1.28573_0012.sbl-ota","sha512":"d20057d6c4e61d4ccdd3947b8a8c9bf9e126a26c080b9ce3e6839a0615b912c6f93246a60c55a44f2c8b8ec49f5041f46ac38b8a3c0a1c6ab44f062cc62089cf","force":true}'`

But still the same version after the update finished.

MrSpoocy, 2022-06-02:
@Koenkk, I also have the issue with my Philips LST002 (915005106701), Firmware: 1.24.0_r20239 (20170323), it say 100% but still old firmware.  I still have a Philips Hue Bridge here and I have a switch to mirror the network traffic. I use a Sonoff Zigbee 3.0 dongle on a Raspberry CM4 with Docker. If you tell me (maybe live via discord screen sharing) what exactly you need, I can make the update via Hue Bridge and maybe you see wich steps needed to work. If you want to contact me via Discord: MrSpoocy#8445.

MrSpoocy, 2022-06-02:
@TheJulianJES http://fds.dc1.philips.com/firmware/ZGB_100B_010F/16781312/100B-010F-01001000-ConfLight-LedStrips_0012.zigbee

TheJulianJES, 2022-06-02:
http://fds.dc1.philips.com/firmware/ZGB_100B_010F/16779778/100B-010F-01000A02-ConfLight-LedStrips_0012.zigbee should be the correct one for an intermediate update (and needs to be added to the repo with a min/max firmware version for both images)

MrSpoocy, 2022-06-02:
Update works, I'm now at 1.50.2_r30933. But the next update not work anymore ... console say `Update of '0x00178801049b25d2' failed (Timeout: device did not request any image blocks)` after I pressed the update button :/

deviantintegral, 2022-07-07:
I recently purchased 2 additional remotes that ship with firmware `5.45.1.17846`. I also have two bridges that are new-in-box. And, I have a conbee II I believe I can use for zigbee traffic sniffing if required. If someone could post what information we need to move this along, I can try to grab the zigbee traffic or mitm the update process from a bridge.

puijken, 2022-07-11:
I currently have 4 dimmers with the following status:
[image]

All have been updated or tried updating by Zigbee2MQTT only:
- 2 after multiple retries, reconnects, battery removals, etc.. are successful and running the latest version
- another one has "updated" but is consistently reporting the older version - however it seems there is no new version available when checking - also tried reconnecting, battery removal, etc..
- the last one has gone trough the process multiple times (updates, reconnects, battery removed) and keeps reporting the older version

Strange stuff ... luckily they all just work :)

WatskeBart, 2022-07-14:
> * 2 after multiple retries, reconnects, battery removals, etc.. are successful and running the latest version

What kind of procedure would you recommend to get them updated eventually?

hughb8on, 2022-08-17:
unstale

umrath, 2022-08-26:
Interesting topic as I fail to update 4 Hue Dimmer devices as well, even after trying pretty much any kind of combination/clicking marathon I could come up with and retrying literally dozens of times.

deviantintegral, 2022-09-01:
I confirmed this is still an issue with the latest release. Updating gets to 100%, but then the remote reports the old version even after removing and re-pairing.

bjornarfjelldal, 2022-09-28:
Can confirm to be an issue for me to

mogim0gi, 2022-10-09:
issue confirmed too

inertia666, 2022-10-23:
Been having problems with my some of my Philips 929002398602 dimmers.

Had to force remove and take out the battery and repair to kick the start the process. Sometimes takes a few attempts to get a successful update but is ball-achingly slow.

spicerackk, 2022-11-13:
Also having the same issue with the gen 1 dimmer switch, cannot get it to update at all, no amount of clicking or resetting changes that.

ChrisThomas42, 2022-11-18:
Same issue here.  I've tried removing from Z2M and connecting to Hue hub and trying the update.  The dimmer does update through the Hue hub but when I connect back to Z2M the old firmware is still reported!?

Is the update actually happening just that the version is somehow misreported?

blackie333, 2022-12-02:
The same problem with my 2 324131092621 controllers. Both are reporting version 5.45.1.17846 from 20160302.
1st was updated multiple times and stopped requesting new update. The 2nd was also updated multiple times but this one is still requesting new updates. The problem is that latest Z2MQTT versions generate notification message in HA for this new updates.
Would like to solve it for good but not sure how.

PSTB, 2022-12-25:
> The same problem with my 2 324131092621 controllers. Both are reporting version 5.45.1.17846 from 20160302.

Exactly the same for me today. 
It has been one year with no solution. What is needed to get it fixed ?
I _can_ do the update using my Hue hub but that defeats my objective of using Z2M...

Thanks for any ideas ?!

deviantintegral, 2023-01-04:
Well, this is fun.

I got around to setting up one of the bridges I mentioned in the summer, and ran through the update process with the official app. I have a (nearly) complete capture of a z2m update (failing), and a capture of the update from the bridge.

However:

1. The official update process seems to split sending the firmware update from actually applying it. You can see it being written to the device, and when done the app prompts to "finish" the update which it says takes 5 minutes or less. If this is part of the way all Zigbee OTAs work, and not a proprietary Philips extension, it would be a _huge_ improvement to see Z2M do this. For example, a user could configure updates to transmit overnight, and then apply and reboot the devices at a good time.
2. The app said the update was complete and showed no further updates. In the app, it showed the same version number though of something like `43.` (sorry, forgot to write it down). I repaired it back to zigbee2mqtt... and it had the old version reporting!
3. I then re-paired with the Hue bridge, and it shows a completely different version number of `1.1.28573` and no updates available.
4. I went back to z2m, and confirmed the version that shows up is `5.45.1.17846`.
5. Since I also obtained an outdoor motion sensor, and it updated firmware just fine. So while this issue says "end devices", I have at least two (another indoor motion sensor) that show that it's not _all_ indoor devices.

A brief scan of the packet captures didn't show any obvious errors or issues. The biggest flag I see seems to be in the version numbers. With Hue bulbs, the version number matches what I see in the Hue app. Perhaps we are looking at the wrong version string in the OTA files? Or Philips made a manufacturing mistake in how they manage firmware on some devices, and are hacking around it in the app?

drifter75, 2023-01-13:
I am having the same issue, update process was successful.

Firmware build date : 20160302
Firmware version    : 5.45.1.17846

puijken, 2023-01-16:
Have been reading and posting in this issue a couple of times in the past. Strangest results… some got updated, some did not and some reported old version but could not update anymore… I eventually gave up and left it for what it is..

Now skip a couple of months and performance /network issues have forced me to recreate my network (changed to channel 25 and all is speedy again.. wohooo). 

And the interesting part.. all my dimmers are now reporting the latest version, even the one that completely failed during updates and the one which was “stuck” on the older version… 

Maybe someone needs to try removing the device from database.db and re-pair the device and see what happen, it might give the same result.. (I actually did this but did not check versions as I was entirely focused on the performance/network stability issue).

t34wrj, 2023-01-21:
I am also having this issue on both of my Philips 324131092621 hue dimmer switches.

If I just press 'update' in Zigbee2MQTT, the update fails with this error in the logs: 'Update of 'Philips Remote 1' failed (Device didn't respond to OTA request)'.

If I try and keep the device alive by pressing a few button after pressing the update button, the error in the logs is 'Update of 'Philips Remote 1' failed (Timeout: device did not request any image blocks').

Luminger, 2023-02-07:
I've been struggling for a while to get a bunch of my [9290024406](https://www.zigbee2mqtt.io/devices/9290024406.html#philips-9290024406) (LWU001) updated. They won't update either with the official Hue bridge, the Bluetooth part of the official Hue Android app nor z2m.

When I got them brand new, it looked like they all requested new firmware images once (in my z2m setup). Once that happened, they're stuck on either 1.55.7_r28193 (20190322) or 1.76.10 (20201012).

When I watch for the `commandQueryNextImageRequest` dumps from z2m, I can see something strange though. All of them return `{"fieldControl":0,"fileVersion":16784918,"imageType":276,"manufacturerCode":4107}` which basically means they (believe?) they're already running the latest version, 16784918 (this should be https://github.com/Koenkk/zigbee-OTA/pull/241).

My current plan is to wait for the next round of OTAs for those bulbs and have debug logging enabled to see what's happening on the network.

Either the OTA process for those bulbs is somewhat broken or the `swBuildId` and `dateCode` attributes are unchanged in new images, which I don't think should be the case.

Alternatively another theory is that the bulbs never switched over to the new software version but have them stored in memory but were never told to boot into?

deviantintegral, 2023-03-11:
Not stale!

D34DC3N73R, 2023-03-15:
Add another to the list. The update completes fine, but it's "updated" to the same firmware.
```
MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"from":{"date_code":"20160302","software_build_id":"5.45.1.17846"},"id":"Dimmer Switch","to":{"date_code":"20160302","software_build_id":"5.45.1.17846"}},"status":"ok","transaction":"9uxht-23"}'
```

[84 more replies]