Koenkk/zigbee2mqtt #10140: Moes BRT-100-TRV does not respond after a faulty OTA update.

vladi1234, 2021-12-11:
## What happened
Moes BRT-100-TRV thermostat doesn't have its own firmware update yet, but Thermostat has received a wrong firmware update from a completely different Tuya thermostat.
Because of this, many thermostats were beefashed out of order

## This fixed the problem and TVR no longer received an incorrect OTA update.
https://github.com/Koenkk/zigbee-OTA/issues/48
https://github.com/Koenkk/zigbee-OTA/pull/50#issuecomment-913736281

## What did you expect to happen
Many users now want to restore broken thermostats.
If someone has ideas, or important information, write here in this topic, because others are already closed.

## Well-known issues that are unfortunately already closed.
https://github.com/Koenkk/zigbee2mqtt/issues/7941
https://github.com/Koenkk/zigbee2mqtt/issues/7941#issuecomment-907148068

https://github.com/Koenkk/zigbee-herdsman-converters/issues/2932
https://github.com/Koenkk/zigbee-herdsman-converters/issues/3317

## Maybe someone can use it.
https://github.com/Koenkk/zigbee-herdsman-converters/issues/2937

########################################################################################
[image]
[image]
[image]

[image]
[image]
[image]

[image]

vladi1234, 2021-12-11:
>@MattWestb commented on 27 Aug
>
>You need one SWD / J-Link probe (if its one EFR32MG21 chip you need one real J-Link) for reading the flash of the Zigbee >module and if can dumping the internal flash pleas also dumping the "user data" that can having device specific setting for the >device.
>
>One working method: https://github.com/MattWestb/IKEA-TRADFRI-ICC-A-1-Module/tree/master/Flashing-MG
>
>I have dumped the tuya ZBGW (ist one EFR32MG1B) and one tuya LED Strip controller with EFR32MG21 and they was not >debug locked so dumping was working OK :-))

Best regards.
vladi1234

MattWestb, 2021-12-11:
Some very brave Z2M and ZHA (and very likels no deCONZ) have rebricking ther IKEA remotes from bad firmware in this thred https://github.com/zigpy/zigpy/discussions/660#discussioncomment-1565541.

If some have one OK TRV then dump the main flash and userdata of the Zigbee module and you can flashing them on other devices with and its working great.

Only need one ESP or STM32 board and loading SWD probe firmware on it or is having one real J-Link probe.

YoannLe, 2021-12-21:
Hi,

Did someone could dump the original firmware of Moes BRT-100-TRV?
If not I'll need to buy a new one with hope to save my 3 TRV with wrong OTA... But I'm not shure to success.

If the original firmware is dumped, do you now if it could bu push by OTA to faulty TRV?
Or will be need to be flashed manualy on each unit?
Does you know if the same full dumped image could reflashed or it will needed to edit some specifics values for each TRV?

I'll try to help ;)

BR,
Yoann

vladi1234, 2021-12-26:
> If the original firmware is dumped, do you now if it could bu push by OTA to faulty TRV?
Or will be need to be flashed manualy on each unit?

Hi,
I think it has to be flashed manually because the defective ones are no longer paired with z2m.

davrodfer, 2022-01-03:
Hi, I have new devices. How can I dump the original firmware and try to recover the brickeds devices?

MattWestb, 2022-01-03:
More working method its depends what SWD / J-Tag you is having and the program you like using https://github.com/MattWestb/IKEA-TRADFRI-ICC-A-1-Module/tree/master/Flashing-MG.

The pads on the module is in the tuya space in the post before.

Normally only needed dumping the main flash but user data can being good to have if tuya have putting somthing in it.

YoannLe, 2022-01-03:
> Hi, I have new devices. How can I dump the original firmware and try to recover the brickeds devices?

Got a new one today too, I'll try to dump the firmware and share my progress

YoannLe, 2022-01-03:
[image]
[image]
Not easy to unglue without breaking the front acrylic glass....

vladi1234, 2022-01-03:
Wiring inserted, see above:

https://github.com/Koenkk/zigbee2mqtt/issues/10140#issue-1077552450

TomaszBich, 2022-01-03:
Fingers crossed!!!

YoannLe, 2022-01-05:
Dump done, need to check them and reflash data/app dump on a break TRV.  Maybe tomorrow.

MattWestb, 2022-01-05:
## Great !!
You need converting the bin file to elf if using GDB or with Silabs commander to s37 files.
The "user data" is also interesting but shall not being needed if restoring one device of the same type.

YoannLe, 2022-01-06:
First, I use:
https://github.com/Jason2866/blackmagic-espidf
https://github.com/MattWestb/IKEA-TRADFRI-ICC-A-1-Module/tree/master/Flashing-MG

Best way to open the TRV: Small holes to get the srews 

Then open, get access to the tuya module:

 
`"c:\Program Files (x86)\GNU Arm Embedded Toolchain\10 2021.10\bin\arm-none-eabi-gdb.exe"
C:\Program Files (x86)\GNU Arm Embedded Toolchain\10 2021.10\bin\arm-none-eabi-gdb.exe: warning: Couldn't determine a path for the index cache directory.
GNU gdb (GNU Arm Embedded Toolchain 10.3-2021.10) 10.2.90.20210621-git
Copyright (C) 2021 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=i686-w64-mingw32 --target=arm-none-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word".
(gdb) target extended-remote 192.168.4.1:2022
Remote debugging using 192.168.4.1:2022
(gdb) mon sw
Target voltage: 3459mV
Available Targets:
No. Att Driver
 1      EFR32MG13P 220 F512 Mighty Gecko M3/M4
(gdb) att 1
Attaching to Remote target
warning: No executable has been specified and target does not support
determining executable automatically.  Try using the "file" command.
0x0002b2b2 in ?? ()
(gdb) mon efm_info
DI version 1 (silabs remix?) base 0x0fe08000

EFR32MG13P 220 F512 = Mighty Gecko 512kiB flash, 64kiB ram
Device says flash page size is 2048 bytes, we're using 2048 bytes

Radio si65535

(gdb) info mem
Using memory regions provided by the target.
Num Enb Low Addr   High Addr  Attrs
0   y   0x00000000 0x00080000 flash blocksize 0x800 nocache
1   y   0x0fe00000 0x0fe00800 flash blocksize 0x800 nocache
2   y   0x0fe10000 0x0fe14000 flash blocksize 0x800 nocache
3   y   0x20000000 0x20010000 rw nocache
(gdb)`

Then dump everything:

`dump memory BRT100_main.bin 0x00000000 0x00080000   
dump memory BRT100_user.bin 0x0fe00000 0x0fe00800  
dump memory BRT100_boot_bis.bin 0x0fe10000 0x0fe14000  
dump memory BRT100_ram.bin 0x20000000 0x20010000 `

Convert bin to elf:

`.\objcopy.exe --input-target binary --output-target elf32-little BRT100_main.bin BRT100_main.elf
.\objcopy.exe --input-target binary --output-target elf32-little BRT100_user.bin BRT100_user.elf`

We need the user config because GPIO are not good on the bricked TRV.

Load elf to a bricked TRV:
`(gdb) target extended-remote 192.168.4.1:2022
Remote debugging using 192.168.4.1:2022
(gdb) mon sw
Target voltage: 3462mV
Available Targets:
No. Att Driver
 1      EFR32MG13P 220 F512 Mighty Gecko M3/M4
(gdb) att 1
Attaching to Remote target
warning: No executable has been specified and target does not support
determining executable automatically.  Try using the "file" command.
0x0002b2b2 in ?? ()
(gdb) mon efm_info
DI version 1 (silabs remix?) base 0x0fe08000

EFR32MG13P 220 F512 = Mighty Gecko 512kiB flash, 64kiB ram
Device says flash page size is 2048 bytes, we're using 2048 bytes

(gdb) info mem
Using memory regions provided by the target.
Num Enb Low Addr   High Addr  Attrs
0   y   0x00000000 0x00080000 flash blocksize 0x800 nocache
1   y   0x0fe00000 0x0fe00800 flash blocksize 0x800 nocache
2   y   0x0fe10000 0x0fe14000 flash blocksize 0x800 nocache
3   y   0x20000000 0x20010000 rw nocache

(gdb) load BRT100_main.elf 0x0
Loading section .data, size 0x80000 lma 0x0
Start address 0x00000000, load size 524288
Transfer rate: 15 KB/sec, 987 bytes/write.
(gdb) load BRT100_user.elf 0x0fe00000
Loading section .data, size 0x800 lma 0xfe00000
Start address 0x00000000, load size 2048
Transfer rate: 11 KB/sec, 682 bytes/write.
`
TRV back to life and working fine now!

I share all dumps, and will add picts.
[Dumps.zip](https://github.com/Koenkk/zigbee2mqtt/files/7824268/Dumps.zip)

vladi1234, 2022-01-06:
Great job!

MattWestb, 2022-01-06:
## Great work done !!
If using Silabs Commander and converting it to s37 file is easier for users flashing it then its having all the offsets inside the file and can being flashed with most flash prebes and programs.
I was doing that with all IKEA remotes that was broken after bad OTA files in there test feed in the summer.

I taking on fast look in the user data file if its some interesting information :-))

TomaszBich, 2022-01-06:
I'm impressed @YoannLe

YoannLe, 2022-01-07:
> I'm impressed @YoannLe

Thanks to @vladi1234 @MattWestb  !
I only fallowed their advices and GitHub 🙂

Styx85, 2022-01-14:
Only one question remains, if someone has a better way to open it than to drill holes into the front glass. @vladi1234  how did you open it? In your pictures, it looks like an intact front without holes.

MattWestb, 2022-01-14:
Heating is normally working on glued plastic things but not too much !!

vladi1234, 2022-01-14:
[image]

I opened with a cutter knife. You have to make a long blade about 5cm and just carefully slide it in between the touch screen and the housing, but don't bend it away! Otherwise breaks. Afterwards you can glue them together again.

Styx85, 2022-01-15:
Ok, thank you very much. It is done and unbricked. I attached some photos, because the cutter was a little bit rough for me. I used a hair dryer to warm up the glue and then a thinner credit card type cardboard (calendar that you get gifted by banks) and a toll I got decades ago to fit GU10 halogen lights. See the photos.
[image]
[image]
[image]
[image]

vladi1234, 2022-01-16:
> What kind of glue @vladi1234 did you use to fix it again?

I took 4 small pieces of transparent Tesafilm 4 x 10 mm and glued them over the edge.
You can hardly see it, and in case you have to open it again, don't break it.

[image]
[image]

Styx85, 2022-01-16:
Well, sounds like a solution, too. I just used some waterproof wood glue and pressurized the RV with some heavy books overnight.  Hope I do not have to open it again.

TomaszBich, 2022-01-19:
Hi Guys,

Where I have to place the YoannLe's dumps? Which windows path?
[image]

MattWestb, 2022-01-19:
If you dont providing one path in the load command its shall being in the same directory you was starting GDB. For making it easier you can doing one copy with name 1234.elf and its going faster writing ;-)

TomaszBich, 2022-01-19:
Thank you @MattWestb - fixed!!!

vladi1234, 2022-02-19:
Hello,
Plastic hooks on my TRV broke off by themselves, so TRV cannot regulate the temperature down.

[image]

closed 2022-03-30 by github-actions[bot]

sgilyin, 2023-10-06:
> First, I use: https://github.com/Jason2866/blackmagic-espidf https://github.com/MattWestb/IKEA-TRADFRI-ICC-A-1-Module/tree/master/Flashing-MG
>
> Best way to open the TRV: Small holes to get the srews [...]

How can I do that with esp32?

phyesix, 2023-10-22:
I'm pretty new at this myself. I have an ESP32 board. I don't know how to flash it. Is there anyone who can prepare a detailed guide? @Styx85 @MattWestb @YoannLe 😇

chuntin, 2023-10-26:
Can someone explain for dummies how to fix the trv's!

paulandrewcrouch, 2023-10-30:
Does anyone have a copy of the original firmware?   I have some on version -1 and some on version 67, with only those on 67 asking for to upgrade to 192.   It's all very puzzling.   I can't seem to control all of them either, but it seems to be a mixture.   Worrying given the money I've spent on them.   v67 appears to work.

jonathan-gatard, 2023-12-09:
Hi everybody,

Solution can be found here [https://github.com/Koenkk/zigbee2mqtt/issues/19173](url)