Koenkk/zigbee2mqtt #10150: Moes TRV generates error when temperature is calibrated

groenmarsmannetje, 2021-12-12:
## What happened
I calibrate the temperature of the Moes TRV with external temperature sensor. As the radiator is build in, the temperature difference can become high when the heating starts.

This TRV allows high calibration differences but you need to calculate them from 128 baseline. So +1 is 129 and -12 is 116. You need to do that for this device when difference is lower/higher then -9 or +9.

With introduction with the number type sensor I get an out of range error in logs. But temperature calibration with use of Mqtt message publish still works if you take the above calculation into account. 

```
Invalid value for number.kinderkamer_thermostaat_local_temperature_calibration: 118.5 (range -20.0 - 20.0)
```

## What did you expect to happen

No errors and also ability to use the number entity in UI for these higher range differences.

## How to reproduce it (minimal and precise)

Set calibration temperature with use of mqtt publish message with higher values (e.g. 118)

## Debug info
Zigbee2MQTT version:
Adapter hardware: CC2531, CC2530, CC26X2R1, CC1352P-2, Conbee II
Adapter firmware version:

blazewicz, 2021-12-12:
related issue: Koenkk/zigbee-herdsman-converters#3488

groenmarsmannetje, 2022-01-12:
Error occurs in version 122.2-1 although range has been changes to -30.0 - 30.0

groenmarsmannetje, 2022-02-12:
Problem still exists

groenmarsmannetje, 2022-04-15:
Issue still exists

groenmarsmannetje, 2022-05-17:
Reopen

Koenkk, 2022-05-17:
Can you provide the data/database.db entry of your device?

groenmarsmannetje, 2022-05-18:
> Can you provide the data/database.db entry of your device?

@Koenkk see entry below

```
{"id":121,"type":"EndDevice","ieeeAddr":"0x847127fffe95ff49","nwkAddr":8262,"manufId":4098,"manufName":"_TZE200_ckud7u2l","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"�9�)\u0011","65506":31,"65508":0,"modelId":"TS0601","manufacturerName":"_TZE200_ckud7u2l","powerSource":3,"zclVersion":3,"appVersion":87,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":87,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1652894401621,"defaultSendRequestWhen":"immediate"}
```

Koenkk, 2022-05-20:
Could you check if the issue is fixed with the following external converter:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const tzLocal = {
    tuya_thermostat_calibration: {
        key: ['local_temperature_calibration'],
        convertSet: async (entity, key, value, meta) => {
            let temp = Math.round(value * 10);
            if (temp < 0) {
                temp = 0xFFFFFFFF + temp + 1;
            }
            // https://github.com/Koenkk/zigbee2mqtt/issues/10150
            if (meta.device.manufacturerName === '_TZE200_ckud7u2l') {
                temp = temp + 128;
            }
            await tuya.sendDataPointValue(entity, tuya.dataPoints.tempCalibration, temp);
        },
    }
}

const definition = {
    zigbeeModel: ['kud7u2l'],
    fingerprint: [
        {modelID: 'TS0601', manufacturerName: '_TZE200_ckud7u2l'},
        {modelID: 'TS0601', manufacturerName: '_TZE200_ywdxldoj'},
        {modelID: 'TS0601', manufacturerName: '_TZE200_cwnjrr72'},
        {modelID: 'TS0601', manufacturerName: '_TZE200_pvvbommb'},
        {modelID: 'TS0601', manufacturerName: '_TZE200_9sfg7gm0'}, // HomeCloud
        {modelID: 'TS0601', manufacturerName: '_TZE200_2atgpdho'}, // HY367
        {modelID: 'TS0601', manufacturerName: '_TZE200_cpmgn2cf'},
        {modelID: 'TS0601', manufacturerName: '_TZE200_4eeyebrt'}, // Immax 07732B
    ],
    model: 'TS0601_thermostat',
    vendor: 'TuYa',
    description: 'Radiator valve with thermostat',
    whiteLabel: [
        {vendor: 'Moes', model: 'HY368'},
        {vendor: 'Moes', model: 'HY369RT'},
        {vendor: 'SHOJZJ', model: '378RT'},
        {vendor: 'Silvercrest', model: 'TVR01'},
        {vendor: 'Immax', model: '07732B'},
    ],
    meta: {tuyaThermostatPreset: tuya.thermostatPresets, tuyaThermostatSystemMode: tuya.thermostatSystemModes3},
    ota: ota.zigbeeOTA,
    onEvent: tuya.onEventSetLocalTime,
    fromZigbee: [fz.tuya_thermostat, fz.ignore_basic_report, fz.ignore_tuya_set_time],
    toZigbee: [tz.tuya_thermostat_child_lock, tz.tuya_thermostat_window_detection, tz.tuya_thermostat_valve_detection,
        tz.tuya_thermostat_current_heating_setpoint, tz.tuya_thermostat_auto_lock,
        tzLocal.tuya_thermostat_calibration, tz.tuya_thermostat_min_temp, tz.tuya_thermostat_max_temp,
        tz.tuya_thermostat_boost_time, tz.tuya_thermostat_comfort_temp, tz.tuya_thermostat_eco_temp,
        tz.tuya_thermostat_force_to_mode, tz.tuya_thermostat_force, tz.tuya_thermostat_preset, tz.tuya_thermostat_away_mode,
        tz.tuya_thermostat_window_detect, tz.tuya_thermostat_schedule, tz.tuya_thermostat_week, tz.tuya_thermostat_away_preset,
        tz.tuya_thermostat_schedule_programming_mode],
    exposes: [
        e.child_lock(), e.window_detection(),
        exposes.binary('window_open', ea.STATE).withDescription('Window open?'),
        e.battery_low(), e.valve_detection(), e.position(),
        exposes.climate().withSetpoint('current_heating_setpoint', 5, 35, 0.5, ea.STATE_SET)
            .withLocalTemperature(ea.STATE).withSystemMode(['heat', 'auto', 'off'], ea.STATE_SET,
                'Mode of this device, in the `heat` mode the TS0601 will remain continuously heating, i.e. it does not regulate ' +
                'to the desired temperature. If you want TRV to properly regulate the temperature you need to use mode `auto` ' +
                'instead setting the desired temperature.')
            .withLocalTemperatureCalibration(-9, 9, 1, ea.STATE_SET)
            .withAwayMode().withPreset(['schedule', 'manual', 'boost', 'complex', 'comfort', 'eco'])
            .withRunningState(['idle', 'heat'], ea.STATE),
        e.auto_lock(), e.away_mode(), e.away_preset_days(), e.boost_time(), e.comfort_temperature(), e.eco_temperature(), e.force(),
        e.max_temperature(), e.min_temperature(), e.away_preset_temperature(),
        exposes.composite('programming_mode').withDescription('Schedule MODE ⏱ - In this mode, ' +
                'the device executes a preset week programming temperature time and temperature.')
            .withFeature(e.week())
            .withFeature(exposes.text('workdays_schedule', ea.STATE_SET))
            .withFeature(exposes.text('holidays_schedule', ea.STATE_SET))],
};

module.exports = definition;

```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

closed 2022-05-28 by github-actions[bot]