Koenkk/zigbee2mqtt #10156: Errors in Home assistant lo

IgorSimic, 2021-12-12:
# Bug Report
## What happened
After home assistant core update to 2021.12.0 from 2021.11.5 lot of error messages in HA log:
Error parsing value: access to attribute 'update' of 'dict' object is unsafe. (value: {"auto_off":null,"consumer_connected":true,"consumption":17.08,"current":0,"energy":17.08,"last_seen":"2021-12-12T15:57:14+01:00","led_disabled_night":null,"linkquality":45,"overload_protection":null,"power":0,"power_outage_memory":null,"state":"ON","temperature":26,"update_available":null,"voltage":210}, template: {{ value_json['update']['state'] == "available" }})
comming from from devices Xiaomi Mi power plug ZigBee EU (ZNCZ04LM.)
Nothing was changed than HA version.

## What did you expect to happen
No messages like before.

## How to reproduce it (minimal and precise)

## Debug Info
Zigbee2MQTT version: 1.21.1
Adapter hardware: CC2652RB
Adapter firmware version: maintrel': 1, 'majorrel': 2, 'minorrel': 7, 'product': 1, 'revision': 20210708

IgorSimic, 2021-12-13:
Sorry, duplicate.
Already discused: https://github.com/Koenkk/zigbee2mqtt/issues/9939

closed 2021-12-13 by IgorSimic

zibous, 2023-02-02:
same Problem here with router-ikea-tradfri-03:

```
Logger: homeassistant.helpers.template
Source: helpers/template.py:627
First occurred: 05:51:40 (36 occurrences)
Last logged: 11:43:21

Error parsing value: access to attribute 'update' of 'dict' object is unsafe. (value: {"device":{"applicationVersion":33,"dateCode":"20190311","friendlyName":"router-ikea-tradfri-03","hardwareVersion":1,"ieeeAddr":"0x5c0272fffe87b6e0","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","model":"E1746","networkAddress":17203,"powerSource":"Mains (single phase)","softwareBuildID":"2.3.086","stackVersion":98,"type":"Router","zclVersion":3},"elapsed":601107,"last_seen":"2023-02-02T11:03:17+01:00","linkquality":207}, template: {{ value_json['update']['state'] == "available" }})
```

[1.30.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.30.0) commit: [8f781db](https://github.com/Koenkk/zigbee2mqtt/commit/8f781db)
Coordinator-Typ: ConBee2/RaspBee2
Coordinator-Version: 0x26660700
Coordinator IEEE Adresse: 0x00212effff060927
Frontend Version: 0.6.124