Koenkk/zigbee2mqtt #10158: Description field per device, without renaming it

P1Rebo, 2021-12-12:
## Is your feature request related to a problem? Please describe
I have about 50 ZigBee devices connected based on a standard naming scheme, e.g. ZX001 = ZigBee (Z) based power plug (X) number 1 (001) which I also use for addressing the device via MQTT. Anyhow, I'd really like having a more clear description of the device without ruining my name scheme :-)

## Describe the solution you'd like
Add a description data field that can be used in order to have a device description. This description shall be entered manually and stored with each device of the gateway. In case a device is deleted, this description will be deleted as well.

## Describe alternatives you've considered
Use an MQTT name just for the MQTT messages of a device, in my case ZX001. The current name won't be used as identifier, just as description.

## Additional context
It would generate a benefit in the device overview table.

P1Rebo, 2022-01-12:
Ok here is my comment :-)

P1Rebo, 2022-02-12:
Ping

Koenkk, 2022-02-12:
Being worked on: https://github.com/Koenkk/zigbee2mqtt/pull/11244

closed 2022-02-20 by github-actions[bot]