Koenkk/zigbee2mqtt #10162: Aqara Opple Switch problems with voltage

Melissen71, 2021-12-13:
## What happened
Aqara Opple Switch gives in this screen VOLTAGE ":null"

Dec 13 10:55:26 raspberrypi npm[453]: Zigbee2MQTT:info  2021-12-13 10:55:26: MQTT publish: topic 'zigbee2mqtt/AqaraDW01', payload '{"battery":100,"contact":true,"linkquality":6,"voltage":3005}'
Dec 13 10:55:27 raspberrypi npm[453]: Zigbee2MQTT:info  2021-12-13 10:55:27: MQTT publish: topic 'zigbee2mqtt/AqaraDW01', payload '{"battery":100,"contact":true,"linkquality":12,"voltage":3005}'
Dec 13 11:24:35 raspberrypi npm[453]: Zigbee2MQTT:info  2021-12-13 11:24:35: MQTT publish: topic 'zigbee2mqtt/Aqara2butWS01', payload '{"battery":0,"linkquality":75,"state":"OFF","temperature":25,"voltage":null}'
Dec 13 11:54:06 raspberrypi npm[453]: Zigbee2MQTT:info  2021-12-13 11:54:06: MQTT publish: topic 'zigbee2mqtt/AqaraDW01', payload '{"battery":100,"contact":true,"linkquality":45,"voltage":3005}'
Dec 13 12:19:35 raspberrypi npm[453]: Zigbee2MQTT:info  2021-12-13 12:19:35: MQTT publish: topic 'zigbee2mqtt/Aqara2butWS01', payload '{"battery":0,"linkquality":81,"state":"OFF","temperature":25,"voltage":null}'
Dec 13 12:19:39 raspberrypi npm[453]: Zigbee2MQTT:info  2021-12-13 12:19:39: MQTT publish: topic 'zigbee2mqtt/AqaraTD01', payload '{"battery":50,"humidity":58.8,"linkquality":18,"temperature":18.1,"voltage":2600}'

So I get some crashes in Domoticz, because the voltage can't be divided bij 1000.
And so these switches doesn't wortk anymore.

## What did you expect to happen
Sometimes the Aqara Opple Switches gives voltage: 3010 or something like that, then the switches do work.

## How to reproduce it (minimal and precise)
Pair device, device works, after a while the device gives errors with voltage, a day later the voltage is back on, later on the voltage is back to :null and problems again.

Remove device and pair again doesn't help.

Problems as well on cc2531 as on cc2652

Is there a possiblity to turn off the voltagecheck for this device?

## Debug info
Zigbee2MQTT version:
Adapter hardware: CC2531, CC2530, CC26X2R1, CC1352P-2, Conbee II
Adapter firmware version:

1igeza, 2021-12-13:
It was fixed in  #9952 
Are you using latest version?

Melissen71, 2021-12-13:
The temperature sensor is working great, But the Aqara Opple Switch gives voltage :null

https://www.zigbee2mqtt.io/devices/WXCJKG12LM.html

Problem is: it worked great for a long time (1,5 years), and now both switches gives :null

(I sended this log to show that not all sensors gives faulty voltages)

z-x-f, 2021-12-13:
Try the dev branch version

https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html

Koenkk, 2021-12-14:
Duplicate of #9952

closed 2021-12-14 by Koenkk