Koenkk/zigbee2mqtt #10172: Firmware version not beeing displayed

T-IOOO, 2021-12-14:
I have an issue with one bulb not displaying firmware version in the frontend.
I have tried to repair the device to see if it changes.
Anyone else having this issue?
@nurikk Im not sure if this is frontend-related or not?

[image]

nurikk, 2021-12-14:
no

groenmarsmannetje, 2021-12-14:
What happens if you remove the device, restart z2m and pair the device again?

T-IOOO, 2021-12-14:
I was hoping that would be the last thing to do. I have three other bulbs in the same armature 😫

sudofly, 2021-12-14:
> I was hoping that would be the last thing to do. I have three other bulbs in the same armature 😫

unscrew the working ones and do the switch dance ?

T-IOOO, 2021-12-15:
I removed the bulb from the network and then preformed a reset. Now it's back to normal.

This can now be closed

closed 2021-12-15 by T-IOOO