Koenkk/zigbee2mqtt #10174: Zigbee2Mqtt Crashes with Sengled RGB Lights

AngellusMortis, 2021-12-14:
## What happened

Zigbee2Mqtt crashes with error `Error: Cluster with name '64531' does not exist`

#### Logs
<details>
```
2021-12-14T13:12:31.741Z zigbee-herdsman:deconz:frameParser write parameter response - parameter id: 33 - status: 0
2021-12-14T13:12:34.359Z zigbee-herdsman:deconz:adapter checking coordinator simple descriptor
2021-12-14T13:12:34.361Z zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0x0 EP:0 SeqNr. 13 request id: 3
2021-12-14T13:12:34.362Z zigbee-herdsman:deconz:frameParser DATA_REQUEST RESPONSE - request id: 3 status: 0
2021-12-14T13:12:34.363Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2021-12-14T13:12:34.364Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101110
2021-12-14T13:12:34.364Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 1 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
2021-12-14T13:12:34.366Z zigbee-herdsman:deconz:driver query aps data confirm
2021-12-14T13:12:34.371Z zigbee-herdsman:deconz:driver DATA_CONFIRM - sending data state request - SeqNr. 14
2021-12-14T13:12:34.371Z zigbee-herdsman:deconz:driver query aps data indication
2021-12-14T13:12:34.372Z zigbee-herdsman:deconz:frameParser DATA_CONFIRM RESPONSE - destAddr: 0x0 request id: 3 confirm status: 0
2021-12-14T13:12:34.372Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
2021-12-14T13:12:34.373Z zigbee-herdsman:deconz:driver query aps data indication
2021-12-14T13:12:34.376Z zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 15
2021-12-14T13:12:34.378Z zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr. 15 srcAddr: 0x0 destAddr: 0x0 profile id: 0x0 cluster id: 0x8004 lqi: 53
2021-12-14T13:12:34.378Z zigbee-herdsman:deconz:frameParser response payload: 3,0,0,0,24,1,4,1,5,0,1,4,0,0,10,0,25,0,1,5,4,1,0,32,0,0,5,2,5
2021-12-14T13:12:34.378Z zigbee-herdsman:deconz:adapter resolve data request without a transSeq Nr.
2021-12-14T13:12:34.378Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2021-12-14T13:12:34.378Z zigbee-herdsman:deconz:adapter RECEIVING SIMPLE_DESCRIPTOR - addr: 0x0 EP:1 inClusters: 0,10,25,1281 outClusters: 1,32,1280,1282
2021-12-14T13:12:34.378Z zigbee-herdsman:deconz:adapter EP: 1
2021-12-14T13:12:34.378Z zigbee-herdsman:deconz:adapter profile ID: 260
2021-12-14T13:12:34.378Z zigbee-herdsman:deconz:adapter device ID: 5
2021-12-14T13:12:34.378Z zigbee-herdsman:deconz:adapter input cluster: 0x0
2021-12-14T13:12:34.378Z zigbee-herdsman:deconz:adapter input cluster: 0xa
2021-12-14T13:12:34.378Z zigbee-herdsman:deconz:adapter input cluster: 0x19
2021-12-14T13:12:34.378Z zigbee-herdsman:deconz:adapter input cluster: 0x501
2021-12-14T13:12:34.378Z zigbee-herdsman:deconz:adapter output cluster: 0x1
2021-12-14T13:12:34.378Z zigbee-herdsman:deconz:adapter output cluster: 0x20
2021-12-14T13:12:34.378Z zigbee-herdsman:deconz:adapter output cluster: 0x500
2021-12-14T13:12:34.378Z zigbee-herdsman:deconz:adapter output cluster: 0x502
2021-12-14T13:12:34.381Z zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 16
2021-12-14T13:12:39.859Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101010
2021-12-14T13:12:39.859Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
2021-12-14T13:12:39.862Z zigbee-herdsman:deconz:driver query aps data indication
2021-12-14T13:12:39.862Z zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 17
2021-12-14T13:12:39.864Z zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr. 17 srcAddr: 0x7d8f destAddr: 0x0 profile id: 0x104 cluster id: 0xc lqi: 255
2021-12-14T13:12:39.865Z zigbee-herdsman:deconz:frameParser response payload: 24,134,10,85,0,57,246,40,166,67
2021-12-14T13:12:39.866Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":134,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":85,"dataType":57,"attrData":332.32000732421875}],"Command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}},"address":32143,"endpoint":2,"linkquality":255,"groupID":null,"wasBroadcast":false,"destinationEndpoint":1}'
2021-12-14T13:12:39.869Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2021-12-14T13:12:41.360Z zigbee-herdsman:deconz:driver send read device state from queue. seqNr: 18
2021-12-14T13:12:41.363Z zigbee-herdsman:deconz:frameParser device state: 10100010
2021-12-14T13:12:41.364Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2021-12-14T13:12:46.184Z zigbee-herdsman:controller:endpoint Command 0x5c0272fffecc1788/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
2021-12-14T13:12:46.186Z zigbee-herdsman:deconz:adapter zclFrame.payload:
2021-12-14T13:12:46.187Z zigbee-herdsman:deconz:adapter {}
2021-12-14T13:12:46.188Z zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0x24e EP:1 SeqNr. 19 request id: 4
2021-12-14T13:12:46.190Z zigbee-herdsman:deconz:frameParser DATA_REQUEST RESPONSE - request id: 4 status: 0
2021-12-14T13:12:46.190Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2021-12-14T13:12:46.279Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10100110
2021-12-14T13:12:46.280Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 1 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2021-12-14T13:12:46.283Z zigbee-herdsman:deconz:driver query aps data confirm
2021-12-14T13:12:46.289Z zigbee-herdsman:deconz:driver DATA_CONFIRM - sending data state request - SeqNr. 20
2021-12-14T13:12:46.291Z zigbee-herdsman:deconz:frameParser DATA_CONFIRM RESPONSE - destAddr: 0x24e request id: 4 confirm status: 0
2021-12-14T13:12:46.291Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2021-12-14T13:12:46.292Z zigbee-herdsman:deconz:adapter sendZclFrameToEndpoint - message send with transSeq Nr.: 3
2021-12-14T13:12:46.408Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101010
2021-12-14T13:12:46.408Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
2021-12-14T13:12:46.411Z zigbee-herdsman:deconz:driver query aps data indication
2021-12-14T13:12:46.411Z zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 21
2021-12-14T13:12:46.413Z zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr. 21 srcAddr: 0x24e destAddr: 0x0 profile id: 0x104 cluster id: 0x6 lqi: 255
2021-12-14T13:12:46.413Z zigbee-herdsman:deconz:frameParser response payload: 8,3,11,0,0
2021-12-14T13:12:46.414Z zigbee-herdsman:deconz:adapter resolve data request with transSeq Nr.: 3
2021-12-14T13:12:46.415Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":3,"manufacturerCode":null,"commandIdentifier":11},"Payload":{"cmdId":0,"statusCode":0},"Command":{"ID":11,"name":"defaultRsp","parameters":[{"name":"cmdId","type":32},{"name":"statusCode","type":32}]}},"address":590,"endpoint":1,"linkquality":255,"groupID":null,"wasBroadcast":false,"destinationEndpoint":1}'
[... 69 lines cut ...]
2021-12-14T13:12:48.583Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
Error: Cluster with name '64531' does not exist
    at Object.getCluster (/app/node_modules/zigbee-herdsman/src/zcl/utils.ts:66:19)
[... cut ...]
    at parseReadReceivedDataResponse (/app/node_modules/zigbee-herdsman/src/adapter/deconz/driver/frameParser.ts:276:23)
    at parseFrame (/app/node_modules/zigbee-herdsman/src/adapter/deconz/driver/frameParser.ts:458:32)
```
</details>

## What did you expect to happen

Some of my Sengled lights cause this error to happen. I have not seems to always be RGB ones. I have had it happen with E12-N1E (RGB BR30, E1G-G8E (RGB strip) and E1G-G8E (RGB Candle)

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version:
Adapter hardware: Conbee II
Adapter firmware version: 0x26660700

Koenkk, 2021-12-14:
Can you check if this has already been fixed in the latest dev?  (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

AngellusMortis, 2021-12-14:
The HA integration for edge is behind stable: 
[image]

Koenkk, 2021-12-14:
It is not, it is rebuild after every push to the z2m dev branch:

[image]

AngellusMortis, 2021-12-14:
Oh cool. There was no clear indication in HA that it does that. Let me try.

AngellusMortis, 2021-12-14:
I can confirm it happens on edge as well.

Koenkk, 2021-12-14:
Can you check if it has been fixed?

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

AngellusMortis, 2021-12-15:
It looks like it.

closed 2021-12-15 by Koenkk