Koenkk/zigbee2mqtt #10177: Occasional z2m crash due to No Route!

sjorge, 2021-12-14:
## What happened
I noticed sometimes z2m will crash and restart, while it shouldn't be. It was usually when a device is unresponsive.

I happend to catch one while looking at the logs for something
```
Zigbee2MQTT:info  2021-12-14 17:18:17: MQTT publish: topic 'zigbee2mqtt/cover/balcony/side1', payload '{"battery":30,"linkquality":45,"position":0}'
Error: Read 0x001788010bb1ed75/11 lightingColorCtrl(["colorTemperature","currentY","currentX","currentSaturation","currentHue","colorMode"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))
    at ZStackAdapter.sendZclFrameToEndpointInternal (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:414:23)
    at runMicrotasks (<anonymous>)
    at processTicksAndRejections (node:internal/process/task_queues:96:5)
    at Queue.executeNext (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)
[ Dec 14 17:18:17 Stopping because service exited with an error. ]
```

Happened again while writing this issue:
```
Zigbee2MQTT:info  2021-12-14 17:22:06: MQTT publish: topic 'zigbee2mqtt/switch/bathroom/light', payload '{"linkquality":99,"power":0,"state":"ON","update":{"state":"idle"}}'
Error: Read 0x001788010bb1ed75/11 genOnOff(["onOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))
    at ZStackAdapter.sendZclFrameToEndpointInternal (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:414:23)
    at processTicksAndRejections (node:internal/process/task_queues:96:5)
    at Queue.executeNext (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)
```

## What did you expect to happen
It not to crash on 'No network route' error, it should definately be logged, but it should not kill zigbee2mqtt.

## How to reproduce it (minimal and precise)
Not sure, it kind of hard to trigger the 'No network route 'error manually.

## Debug info
Zigbee2MQTT version: 1.22.1-dev commit: d0195d0f
Adapter hardware: CC1352P-2
Adapter firmware version: 20210708

Koenkk, 2021-12-14:
Does this still happen after disabling the availability feature?

sjorge, 2021-12-14:
I’ll disable it for the night and check the service start time tomorrow morning.

~ sjorge

sjorge, 2021-12-15:
Didn't happen tonight, might also just be because all devices had a working route of course.

Koenkk, 2021-12-15:
Added a possible fix, can you check if it works?

sjorge, 2021-12-15:
Will update and re-enable availability and let it run for a day or so.

~ sjorge

sjorge, 2021-12-17:
Not seen it again so far, closing

closed 2021-12-17 by sjorge

GSzabados, 2022-06-02:
error 2022-06-02 22:13:32: Failed to configure 'Electric Immersion Water Heater', attempt 1 (Error: Read 0xa4c1381477066fd5/1 genBasic(["manufacturerName","zclVersion","appVersion","modelId","powerSource",65534], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))
    at ZStackAdapter.sendZclFrameToEndpointInternalapp/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:414:23)
    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32

I see this in the log and Z2M keeps restarting every minute without a clear crash log or any.